roleoroleo/yi-hack-Allwinner #269: Share Audio / Alsa with another application (playback).

ricardojlrufino, 2021-03-26:
Would it be possible for another application to use ASLA playback?

Moving question from: https://github.com/roleoroleo/qr_scanner/issues/1

Sample code for testing:
https://github.com/roleoroleo/qr_scanner/issues/1#issuecomment-807998534

ricardojlrufino, 2021-03-26:
Maybe this:
form: https://superuser.com/a/903415

You can use the dsnoop plugin to copy the sample data to multiple virtual capture devices.

Put something like this into your ~/.asoundrc or /etc/asound.conf, and use snooped_digital as device name in both applications:

```
pcm.snooped_digital {
    type dsnoop
    slave {
        pcm "hw:2,1"   # the actual device
        channels 2
        rate 48000
    }
}
```

ricardojlrufino, 2021-03-27:
amixer and aplay compiled 
[amixer-aplay-arm.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/6215066/amixer-aplay-arm.zip)

roleoroleo, 2021-03-27:
dsnoop is used normally for capturing device.
Am I wrong?
We should use dmix but the kernel lacks the support for System V IPC. dmix is based on SYSV semaphore and it doesn't work with this kernel.

roleoroleo, 2021-03-27:
I'm working on it.
I create a new hack of the libalsa.
Maybe it works.

ricardojlrufino, 2021-03-28:
Its amazing ....

roleoroleo, 2021-04-13:
I released the new library in the last version.

closed 2021-05-15 by roleoroleo