roleoroleo/yi-hack-Allwinner #270: RTSP freeze issue with OBS media source with 0.2.4 release

nmenoux, 2021-03-29:
I've multiple cameras (8.2.0.0A_201912270941) with the 0.2.2 hack and everything is working pretty well with rRTSP server to send streams to OBS Studio.
However, I've updated 2 cameras to 0.2.4 last week-end and now the stream is freezing after few minutes on OBS.
When I mean freezing, actually if I reupdate the media source on OBS it works again so I think the rRTSP server is still working and I could see on the top process list on the cameras.
Maybe there are issues on the RTSP frames and something is not understood on OBS side.
On YiYack 0.2.2 it worked very well and since the update it is blocked after few minutes.
I could try to launch manualy rRTSP server to check if I see some warnings...
Thanks for your help.

nmenoux, 2021-03-29:
I've killed the rRTSP process and started it manually with the DEBUG flag and I've catched a segmentation fault each time the stream is broken so the issue is on rRTSP server and not the OBS media source. In my first message as the rRTSP server crashed, a new instance is respawned and then when I refresh my media source on OBS it works again.

So the issue is to understand why rRTSP process crash.  Here is last lines of debug : 

**RRTSP_RES=low RTSP_AUDIO=yes RRTSP_DEBUG=1 ./rRTSPServer**

```
1617023640663: frame_write_index: 39 - cb_current->output_frame_size 42
1617023640663: frame detected - frame_len: 630 - frame_counter: 1404 - frame_counter_last_valid: 1404 - resolution: 360
1617023640690: frame_len: 630 - cb_current->size: 49152
1617023640690: frame_len: 630 - frame_counter: 1404 - resolution: 360
1617023640690: frame_write_index: 12 - cb_current->output_frame_size 42
1617023640690: frame detected - frame_len: 2324 - frame_counter: 1321 - frame_counter_last_valid: 1321 - resolution: 1080
1617023640716: frame_len: 2324 - cb_current->size: 262144
1617023640716: frame_len: 2324 - frame_counter: 1321 - resolution: 1080
1617023640716: frame_write_index: 40 - cb_current->output_frame_size 42
1617023640716: frame detected - frame_len: 584 - frame_counter: 1405 - frame_counter_last_valid: 1405 - resolution: 360
1617023640742: frame_len: 584 - cb_current->size: 49152
1617023640743: frame_len: 584 - frame_counter: 1405 - resolution: 360
1617023640743: frame_write_index: 13 - cb_current->output_frame_size 42
1617023640743: frame detected - frame_len: 2289 - frame_counter: 1322 - frame_counter_last_valid: 1322 - resolution: 1080
1617023640769: frame_len: 2289 - cb_current->size: 262144
1617023640769: frame_len: 2289 - frame_counter: 1322 - resolution: 1080
1617023640769: frame_write_index: 41 - cb_current->output_frame_size 42
1617023640769: frame detected - frame_len: 585 - frame_counter: 1406 - frame_counter_last_valid: 1406 - resolution: 360
1617023640795: frame_len: 585 - cb_current->size: 49152
1617023640795: frame_len: 585 - frame_counter: 1406 - resolution: 360
1617023640795: frame_write_index: 14 - cb_current->output_frame_size 42
1617023640795: frame detected - frame_len: 2258 - frame_counter: 1323 - frame_counter_last_valid: 1323 - resolution: 1080
1617023640796: frame_len: 2258 - cb_current->size: 262144
1617023640796: frame_len: 2258 - frame_counter: 1323 - resolution: 1080
1617023640796: frame_write_index: 0 - cb_current->output_frame_size 42
1617023640796: frame detected - frame_len: 576 - frame_counter: 1407 - frame_counter_last_valid: 1407 - resolution: 360
Segmentation fault (core dumped)
```

Is it a problem with the circular buffer ?
Thanks for your help

roleoroleo, 2021-03-29:
Please try to run it again with debug=3.

nmenoux, 2021-03-29:
Thanks @roleoroleo, here is the new debug lines with level 3 enabled : 

```
1617026289564: frame_len: 695 - frame_counter: 56980 - resolution: 360
1617026289564: frame_write_index: 18 - cb_current->output_frame_size 42
1617026289564: frame detected - frame_len: 2883 - frame_counter: 56888 - frame_counter_last_valid: 56888 - resolution: 1080
1617026289565: frame_len: 2883 - cb_current->size: 262144
1617026289565: frame_len: 2883 - frame_counter: 56888 - resolution: 1080
1617026289565: frame_write_index: 16 - cb_current->output_frame_size 42
1617026289565: frame detected - frame_len: 666 - frame_counter: 56981 - frame_counter_last_valid: 56981 - resolution: 360
RTSP doGetNextFrame() whole frame - fFrameSize 691 - counter 56980 - fMaxSize 262140 - fLimitNumBytesToStream 0
RTSP doGetNextFrame() - whole frame completed
RTSP doGetNextFrame() start - fFrameSize 262140 - fMaxSize 262140 - fLimitNumBytesToStream 0 - fPreferredFrameSize 0
RTSP doGetNextFrame() read_index == write_index
RTSP doGetNextFrame() read_index == write_index
1617026289616: frame_len: 666 - cb_current->size: 49152
1617026289616: frame_len: 666 - frame_counter: 56981 - resolution: 360
1617026289616: frame_write_index: 19 - cb_current->output_frame_size 42
1617026289616: frame detected - frame_len: 2419 - frame_counter: 56889 - frame_counter_last_valid: 56889 - resolution: 1080
1617026289616: frame_len: 2419 - cb_current->size: 262144
1617026289616: frame_len: 2419 - frame_counter: 56889 - resolution: 1080
1617026289616: frame_write_index: 17 - cb_current->output_frame_size 42
1617026289616: frame detected - frame_len: 592 - frame_counter: 56982 - frame_counter_last_valid: 56982 - resolution: 360
RTSP doGetNextFrame() whole frame - fFrameSize 662 - counter 56981 - fMaxSize 262140 - fLimitNumBytesToStream 0
Segmentation fault (core dumped)
```

roleoroleo, 2021-03-29:
Please try this binary:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6222918/rRTSPServer.gz)

I fixed a little little bug.

nmenoux, 2021-03-29:
Thanks @roleoroleo :-) I will test it this evening and keep you informed ;-)

nmenoux, 2021-03-29:
It is working like a charm for 2 hours now... souds great ! Thanks @roleoroleo 👍

nmenoux, 2021-03-29:
As I've another cam (BFUSY44LDMIZ5M200807) with the AllWinner-v2 0.1.5 release, could I use the same fixed rRTSPServer to cope with same issue with that cam @roleoroleo ?
I've seen the post : https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/118  but not sure if I'm concerned by that issue.

roleoroleo, 2021-03-29:
No, the addresses are different.
And it seems a different problem. Ther is no segfault in that case.
Could you confirm?

nmenoux, 2021-03-29:
I will check on my second cam @roleoroleo . Regarding the first one with your patched rRTSPServer, the stream was frozen now() after few hours...as it was launched by the scheduler, I do not have any trace so I will try tonight to relaunch the process manually with debug enabled to see if the pb will occur again.

roleoroleo, 2021-03-30:
I improved the log:
[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6226964/rRTSPServer.gz)

Arkady23, 2021-03-30:
@roleoroleo !!! I also updated my camera to v0. 2. 4. Checked... at first, everything worked. But at the next check the next day, I did not follow, I do not know how long RTSP worked, I found a black screen and a sound broadcast. I, of course, went to the settings, checked.... the sound is disabled :-)

Arkady23, 2021-03-30:
I didn't do anything with the camera. I opened vlc now — an image with sound appeared!!!
@roleoroleo is that what you did? And can our rtsp on old cameras do this, with sound?
Thank you for your work !!!

roleoroleo, 2021-03-30:
The sound is available (thanks to @PieVo) from version 0.1.8.
I don't know it it's possible on hisilicon based cams.

roleoroleo, 2021-03-31:
@nmenoux 
News about the segfault?
Do you have a new log with the latest binary?

nmenoux, 2021-03-31:
> @nmenoux
> News about the segfault?
> Do you have a new log with the latest binary?

on my 0.2.4 realease with your last exe, it is working without any segfault for hours....looks good 👍  Thanks @roleoroleo . On my second cam with AllWinner-v2, I had some power issue, fixed now, so I could carry on my test with your last release.

Arkady23, 2021-04-08:
This is what RTSP has become with firmware 0.2.4.  

[image]

roleoroleo, 2021-04-08:
I know there is a problem in 0.2.4.
Did you try this
https://github.com/roleoroleo/yi-hack-Allwinner/issues/270#issuecomment-809961433
posted above?

Arkady23, 2021-04-08:
I tried it, but it didn't help.

Arkady23, 2021-04-08:
I tried it, but it didn't help. It looks like resource overload.

roleoroleo, 2021-04-08:
Mmmh.
I made these changes to RTSP because it should be lighter in terms of resources.

Arkady23, 2021-04-08:
@roleoroleo It looks like my WiFi module is dead. At first, the camera hung, and then completely stopped connecting to the router. I ordered a V380 insert on Linux. This is the cheapest camera I've found. But there is no telnet or ftp in it. But there is a regular RTSP. Can you take control of this camera?
https://aliexpress.ru/item/1005001387749627.html?spm=a2g0s.9042311.0.0.420c33edl7tS2n&_ga=2.224371931.1304211249.1617718869-910746486.1590901087

roleoroleo, 2021-04-08:
> Can you take control of this camera?

I don't know.

closed 2021-04-10 by roleoroleo