roleoroleo/yi-hack-Allwinner #273: How to rollback / install ?

MisteurJ, 2021-04-01:
Hello all
Since last firmware 0.2.4, I have hundred of disconnections on my Synology.
So I decided to rollback to 0.2.3, as it was working like a charm.
So I extracted content of the .tar file on an formated FAT32 SD card ("Factory" folder, "newhome" folder and "home_y20gam.stage file on root).
Cam rebooted, and...
Nothing changed, 0.2.4 is still on the info page...
What's the matter ?

roleoroleo, 2021-04-01:
Try again.
It must work.

Or, as a last try, unhack the cam https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack
and apply again.

MisteurJ, 2021-04-01:
Once again, doesn't work.
It seems the cam is not loading/reading content on the SD...
I don't need to copy the y20ga_0.2.3.tar on root, right ? It's its content (3 elements above) that I need to copy to root, right ?

If I unhack the cam, do I need the Yi app to set up the wifi again, before hacking again ?

MisteurJ, 2021-04-01:
I think my cam is bricked...
So, I tried to unhack, nothing happens (orange light on for more than 15 mn).
So I unpluged the cam, wrote the 0.2.3 firmware on the SD, inserted the SD, switched on the cam...
And nothing for more than 15 mn, orange light still on !
Tried to reboot several times with or without SD, same thing...
Orange led still on, can't access the cam, don't see it on my router...

roleoroleo, 2021-04-01:
Remove all mp4 saved files from the sd before applyng the hack.

MisteurJ, 2021-04-01:
There were no mp4 files on the SD (I formated it before copying your hack).
But I think it's too late, it doesn't want to switch on anymore, still this fixed orange led.

roleoroleo, 2021-04-01:
Weird, it's very difficult to brick a cam.
Try to run the backup procedure.
If it works, you can recover it.

GoodSoulGermany, 2021-06-03:
I had a similiar issue (cam did not boot anymore, yellow blinking). Cam did not react to "unhack" or trying to flash a new firmware.

What helped in my case was to use the backup script and resetting the cam. That was the first time that the cam was talking again to me (previous resets the cam just rebootet and said nothing). Then I was able to reconnect the cam to the app and then was able to flash it again. 

Hope that helps you somehow.

closed 2024-04-07 by MisteurJ