roleoroleo/yi-hack-Allwinner #274: Night vision - brightness level adjustment somehow ?

SnOOpeRa, 2021-04-02:
Can we have some kind of way to adjust the level when the night vision (black/white view) kicks in , as currently I think is very aggressive imo, it switched to black/white mode very early when there is still much light in the scene and sometime switched normal/night-vision every 5-10 seconds with the clicking sound ( e.g. in the early morning/evening ) , so I was thinking of can we somehow adjust the threshold of when night vision start.

roleoroleo, 2021-04-02:
I don't know how to control this behavior, it's all handled by yi processes.

SnOOpeRa, 2021-04-04:
Thank you for the quick response and the great work to this hack, it's not a big issue, will get used to it :)

closed 2024-05-10 by github-actions[bot]