roleoroleo/yi-hack-Allwinner #275: MQTT "Birth and will messages"

GqSteve, 2021-04-02:
I have the 0.2.4 running on 2 cams - both cams are configured for MQTT with QoS set to 2.

I can see in an MQTT Explorer that both cams are senting the status (Birth/Will Topic Suffix) non stop changing between online & offline.

roleoroleo, 2021-04-02:
Could you explain me better?

GqSteve, 2021-04-02:
Set a a camera to QoS 2 and all the following retain options (like "Retain for birth and will messages", "Retain for motion messages", and so on) to 0. Connect with a MQTT explorer (like http://mqtt-explorer.com/) to the MQTT broker and watch the messages being broadcasted from the camera e.g. yi-hack/status.

In my case, the status being broadcasted is like "online", "offline", "online", "offline", "online", "offline", "online", "offline", "online", ......

You have also an issue opened for the yi-hack-Allwinner-v2 from mcanteri. I have also the same problem with the yi-hack-Allwinner version. The log from MQTT server is e.g. (I changed the IP to XXX.XXX)

2021.04.02 10:38:17 4: Connection accepted from MQTT2_192.168.XXX.XXX_39778
2021.04.02 10:38:17 5: in:  CONNECT: (16)(16)(0)(4)MQTT(4)(2)(0)(15)(0)(4)1705
2021.04.02 10:38:17 3: Login denied via MQTT2_192.168.XXX.XXXX_39778
2021.04.02 10:38:17 5: out: CONNACK:  (2)(0)(4)
2021.04.02 10:38:17 5: in:  PINGREQ: (192)(0)
2021.04.02 10:38:17 4:   MQTT2_192.168.XXX.XXX_51325 mqtt-explorer-fa6a5216 PINGREQ
2021.04.02 10:38:17 5: out: PINGRESP: (208)(0)

roleoroleo, 2021-04-02:
There is no disconnect in the code, so I don't know why the cam (or the broker) sends the offline message.
Setting "Retain for birth and will messages" to 1 probably will fix the problem.

GqSteve, 2021-04-02:
Setting "Retain for birth and will messages" to 1 didn't fix the problem.

Btw, I removed all entries from "Birth and will messages"

[image]

but there still messages broadcasted to MQTT server as yi-hack/**EMPTY_TOPIC**

[image]

On the other hand, as I described above...there are "Login denied..." messages in the MQTT server every 5 seconds from the cams

roleoroleo, 2021-04-02:
If retain doesn't fix the problem, probably you have a network problem and the socket dies.
The will message is sent ONLY when the app terminates
Check your connection with wireshark.

Login denied is related to another process. To avoid log flooding, simply leave empty the box "Topics Prefix for remote configuration".

GqSteve, 2021-04-03:
Regarding the box "Topics Prefix for remote configuration": I upgraded from 0.2.3 to 0.2.4 and the box "Topics Prefix for remote configuration" is neither visible in /index.html?page=mqtt nor in /index.html?page=mqtt_adv.

Is the box "Topics Prefix for remote configuration" actually the parameter MQTT_PREFIX_CONFIG in the /home/yi-hack/etc/mqttv4.conf file? If yes, I set it to "MQTT_PREFIX_CONFIG=" rebootet the cam...but there is still log flooding.

roleoroleo, 2021-04-03:
This is a screenshot of the mqtt page:
[image]

If the box doesn't appear, try to clean the cache of your browser.

GqSteve, 2021-04-03:
I cleared the cache, but this box doesn't appear for both cameras:

[image]

roleoroleo, 2021-04-03:
Sorry, you are right.
I miss a commit.
Overwrite this file in www/pages
[mqtt.html.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6252204/mqtt.html.gz)

GqSteve, 2021-04-04:
Thank you for the file, that solved the problem with displaying the box "Topics Prefix for remote configuration" 👍 

Regarding the advertising of the status (Birth/Will Topic Suffix) I made a couple of tests. IMHO I think that it is a general "issue" with all new yi-hack's firmware versions related to MQTT:

First test with 3 yi cams: 
a. Camera (yi-hack-Allwinner)
Firmware Version | 0.2.4
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20ga

b. Camera (yi-hack-MStar 0.4.0)
Firmware Version | 0.4.0
Base Version | 4.6.0.0A_201908271549
Model Suffix | h201c

c. Camera  (yi-hack-Allwinner)
Firmware Version | 0.2.4
Base Version | 8.2.0.0A_201912270941
Model Suffix | y20ga

If I set MQTT QoS=2 for all these 3 cameras, then all three cameras advertise the status up to 10 times within 1 second like "online", "offline", "online", "offline", "online", "offline", "online", "offline", "online", ...... I was running all these 3 cameras with the previous firmware versions from you and there was not such a problem with QoS=2.

Second test with the same 3 yi cams:
If I set MQTT QoS=1 for all these 3 cameras, then there is no advertise  "flooding" from all three cameras.

Third test with 1 yi cam:
a. Camera (yi-hack-Allwinner-v2)
Firmware Version | 0.1.4
Base Version | 9.0.19.02_202007141512
Model Suffix | y21ga

This camera is running in the same WiFi as the three cameras described above. The camera is set MQTT QoS=2 and no advertise  "flooding" - This camera has the previous firmware (not the 0.1.5).

roleoroleo, 2021-04-04:
I'm trying to test your scenario and I can't find your problem.
yi-hack-allwinner 0.2.4 and yi-hack-allwinner-v2 0.1.5
QoS=2
When the cams start I receive only 1 "online" message.

The difference between the previous version (about mqttv4) is only this commit:
https://github.com/roleoroleo/mqttv4/commit/346536f1dcff28517e3fae2e1df625a7cca850cd
I don't understand.

GqSteve, 2021-04-05:
I am very sorry, the problem was my new MQTT broker. It supoorts QoS 0 & 1 only. I changed to MOSQUITTO again and every works fine. Thank you for your great work and support. Beer is out to you :)

Btw, talking about MQTT...do you consider supporting MQTTS?

roleoroleo, 2021-04-05:
> Btw, talking about MQTT...do you consider supporting MQTTS?

Qhy not...
I will check the size of the binary adding tls.

GqSteve, 2021-04-05:
I can help with testing since I have a mosquitto setup with MQTTS (with certs for the broker and clients) ;)

roleoroleo, 2021-04-05:
Thank you for your donation!

GqSteve, 2021-04-11:
Feel free to ping me for testing MQTTS...if you are going to implement it, which would be great :)

closed 2021-04-11 by GqSteve

roleoroleo, 2021-04-11:
> Feel free to ping me for testing MQTTS...if you are going to implement it, which would be great :)

I'm working on it but it's harder than I thought.

GqSteve, 2021-04-11:
The tasmota project has a feature for using MQTTS (https://tasmota.github.io/docs/TLS/), whereby the "user" has to compile the firmware for using this feature...may it helps?

roleoroleo, 2021-04-11:
I will check it.