roleoroleo/yi-hack-Allwinner #276: GPIO LED Control

ricardojlrufino, 2021-04-05:
Just sharing how to control LED directly via GPIO
```
  int fd;
    if ((fd = open("/dev/cpld_periph", O_RDWR)) < 0){
        printf("Open error on /dev/gpiog\n");
        exit(0);
    }

    printf("LED OFF \n");
    ioctl(fd, _IO(0x70, 0x1b), 0);

    sleep(2);

    printf("LED ON  \n");
    ioctl(fd, _IO(0x70, 0x1c), 0);
```

roleoroleo, 2021-04-05:
Interesting...

roleoroleo, 2021-04-11:
After using 0x1c to switch on the led, you can use:

```
ioctl(fd, _IO(0x70, 0x1), 0);   //blue on
ioctl(fd, _IO(0x70, 0x2), 0);   //blue off
ioctl(fd, _IO(0x70, 0x3), 0);   //blue blink fast
ioctl(fd, _IO(0x70, 0x4), 0);   //blue blink slow

ioctl(fd, _IO(0x70, 0x9), 0);   //orange on
ioctl(fd, _IO(0x70, 0xa), 0);   //orange off
ioctl(fd, _IO(0x70, 0xb), 0);   //orange blink fast
ioctl(fd, _IO(0x70, 0xc), 0);   //orange blink slow
```

Arkady23, 2021-04-11:
Yes, only this algorithm works on CN17, outdoor, etc. with Hi3518ev200.
https://github.com/npetrovski/yi-home-led/tree/master/src

ricardojlrufino, 2022-05-04:
Using GPIOfs

- Pinos do IR_CUT
echo 235 > /sys/class/gpio/export
echo 236 > /sys/class/gpio/export

echo out > /sys/class/gpio/gpio235/direction
echo out > /sys/class/gpio/gpio236/direction 

echo 0  > /sys/class/gpio/gpio235/value
echo 1  > /sys/class/gpio/gpio236/value  (red wire)

- blue
echo 237 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio237/direction
echo 1  > /sys/class/gpio/gpio237/value (active low)

- Orange
echo 145 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio145/direction
echo 1  > /sys/class/gpio/gpio145/value (active low)

closed 2024-05-10 by github-actions[bot]