roleoroleo/yi-hack-Allwinner #282: MIC Record

ricardojlrufino, 2021-04-11:
Hi, can you figure out how to record audio?

./arecord get this error
> ./arecord: line 1: can't create }dataq: Invalid argument
> ./arecord: line 1: RIFFWAVEfmt: not found

ricardojlrufino, 2021-04-11:
Related to the GPIO part, using strace on the various binaries does not "appear" to have any ioctl.
When I send the command BY THE ANDOIRD APP, this log appears:

p2p_tnp.c(myDoIOCtrl-5298)[16:49:10.279]:myDoIOCtrl(..): 0, IOTYPE_USER_IPCAM_AUDIOSTOP
p2p_tnp.c(myDoIOCtrl-5292)[16:49:10.768]:myDoIOCtrl(..): 0, IOTYPE_USER_IPCAM_AUDIOSTART

But no call to ioctl ...

ref:
[image]

roleoroleo, 2021-04-11:
Maybe p2p_tnp doesn't execute directly ioctl.
Probably it sends a socket command to another process.

ricardojlrufino, 2021-04-11:
Yes, I think that's it too
[image]

closed 2024-05-10 by github-actions[bot]