roleoroleo/yi-hack-Allwinner #283: 1080p y20ga Keeps loosing internet connection

jphccfc, 2021-04-11:
Just updated to the latest version and my yi 1080p home y20ga 9fus keeps disconnecting from the internet? I have turned ftp and telnet off in case its a memory issue....Any way to trouble shoot these frequent disconnects? Even saving updates often ends with a error while saving due to these disconnects. running firmware version 8.2.0.0A._201912270941

jphccfc, 2021-04-12:
So the disconnects seem to have gotten better however now it crashes regularly and needs to have the power removed to reconnect.... Also motion detection seems very sensitive.... Even when on low...

roleoroleo, 2021-04-12:
Try to enable swap file.
If it's a memory problem, this should solve it.

jphccfc, 2021-04-12:
Thanks roleo...i did the swap file but have had it non responsive twice today.. Only a reboot sorted it

jphccfc, 2021-04-13:
Bizarrely it was only loading the top header of the web admin panel earlier... Needed a reboot again

roleoroleo, 2021-04-13:
If you can reach a shell, please run ps command.

jphccfc, 2021-04-13:
Here you go.

BusyBox v1.27.2 () built-in shell (ash)

mount: mounting none on /sys/kernel/debug failed: Resource busy

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@yi-4cff:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
  179 root         0 SW   [oom_reaper]
  180 root         0 SW<  [writeback]
  182 root         0 SW   [kcompactd0]
  183 root         0 SW<  [crypto]
  184 root         0 SW<  [bioset]
  186 root         0 SW<  [kblockd]
  224 root         0 SW   [sys_user]
  231 root         0 SW   [kworker/0:1]
  232 root         0 SW<  [cfg80211]
  239 root         0 SW<  [watchdogd]
  253 root         0 SW<  [spi0]
  263 root         0 SW   [kswapd0]
  348 root         0 SW<  [bioset]
  353 root         0 SW<  [bioset]
  358 root         0 SW<  [bioset]
  363 root         0 SW<  [bioset]
  368 root         0 SW<  [bioset]
  373 root         0 SW<  [bioset]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  402 root         0 SW   [irq/302-sunxi-m]
  404 root         0 SW   [irq/166-sdc0 cd]
  405 root         0 SW   [irq/303-sunxi-m]
  464 root         0 SW<  [bioset]
  465 root         0 SW   [mmcqd/0]
  469 root         0 SW<  [kworker/0:1H]
  481 root         0 SWN  [jffs2_gcd_mtd3]
  488 root       980 S    -/bin/sh
  529 root         0 SW   [ksdioirqd/mmc1]
  532 root         0 SW   [RTW_XMIT_THREAD]
  533 root         0 SW   [RTW_CMD_THREAD]
  534 root         0 SW   [RTWHALXT]
  535 root       680 S    ./log_server
  536 root       984 S    ./dispatch
 1029 root     39924 S    ./rmm
 1121 root      1400 S    /home/base/tools/wpa_supplicant -c/tmp/wpa_supplicant.c
 1163 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.script -
 1303 root      2804 S    ./mp4record
 1315 root       848 S    httpd -p 8080 -h /home/yi-hack/www/ -c /tmp/httpd.conf
 1333 root       876 S    dropbear -R -B
 1337 root       668 S    ipc_multiplexer
 1385 root       848 S    ntpd -p pool.ntp.org
 1402 root      4984 S<   rRTSPServer
 1418 root       984 S    {wd_rtsp.sh} /bin/sh /home/yi-hack/script/wd_rtsp.sh
 1449 root      8108 S    onvif_srvd --pid_file /var/run/onvif_srvd.pid --model Y
 1465 root      2072 S    wsdd --pid_file /var/run/wsdd.pid --if_name wlan0 --typ
 1466 root      2420 S    framefinder
 1475 root       852 S    /home/yi-hack/usr/sbin/crond -c /var/spool/cron/crontab
12612 root         0 SW   [kworker/u2:0]
13448 root         0 SW   [kworker/u2:2]
15456 root       900 S    dropbear -R -B
15495 root       980 S    -ash
15939 root       980 S    sleep 10
16006 root       980 R    ps
32508 root         0 SW   [kworker/u2:1]
root@yi-4cff:~#

roleoroleo, 2021-04-14:
The output is ok.
Try a top command.

jphccfc, 2021-04-14:
Mem: 57664K used, 3500K free, 1832K shrd, 4268K buff, 8120K cached
CPU: 29.3% usr  7.6% sys  0.0% nic 62.7% idle  0.0% io  0.0% irq  0.2% sirq
Load average: 1.32 0.93 0.57 1/99 10278
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1029     1 root     S    39884 64.7   0 31.7 ./rmm
 1408     1 root     S<    4984  8.0   0  1.8 rRTSPServer                            529     2 root     SW       0  0.0   0  0.4 [ksdioirqd/mmc1]
 1472     1 root     S     2420  3.9   0  0.2 framefinder
 9994     1 root     S     1400  2.2   0  0.2 /home/base/tools/wpa_supplicant -c/tm10189 10150 root     R      996  1.6   0  0.2 top                                  10145  1341 root     S      900  1.4   0  0.2 dropbear -R -B                           7     2 root     SW       0  0.0   0  0.2 [rcu_preempt]                            3     2 root     SW       0  0.0   0  0.2 [ksoftirqd/0]
   12     2 root     SW       0  0.0   0  0.2 [kworker/u2:1]                        1455     1 root     S     8108 13.1   0  0.0 onvif_srvd --pid_file /var/run/onvif_ 1309     1 root     S     2800  4.5   0  0.0 ./mp4record
 1471     1 root     S     2072  3.3   0  0.0 wsdd --pid_file /var/run/wsdd.pid --i  536     1 root     S      984  1.6   0  0.0 ./dispatch                            1424     1 root     S      984  1.6   0  0.0 {wd_rtsp.sh} /bin/sh /home/yi-hack/sc    1     0 root     S      980  1.5   0  0.0 /sbin/init
  486     1 root     S      980  1.5   0  0.0 -/bin/sh
10150 10145 root     S      980  1.5   0  0.0 -ash
10244  1424 root     S      980  1.5   0  0.0 sleep 10                             10051     1 root     S      980  1.5   0  0.0 udhcpc -i wlan0 -b -s /home/app/scrip 1341     1 root     S      876  1.4   0  0.0 dropbear -R -B
 1391     1 root     S      848  1.3   0  0.0 ntpd -p pool.ntp.org
 1481     1 root     S      848  1.3   0  0.0 /home/yi-hack/usr/sbin/crond -c /var/ 1321     1 root     S      848  1.3   0  0.0 httpd -p 8080 -h /home/yi-hack/www/ -  535     1 root     S      680  1.1   0  0.0 ./log_server
 1343     1 root     S      668  1.0   0  0.0 ipc_multiplexer
  533     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
  418     2 root     SW       0  0.0   0  0.0 [kworker/u2:2]
  481     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd3]
  231     2 root     SW       0  0.0   0  0.0 [kworker/0:1]
  402     2 root     SW       0  0.0   0  0.0 [irq/302-sunxi-m]
  465     2 root     SW       0  0.0   0  0.0 [mmcqd/0]
  532     2 root     SW       0  0.0   0  0.0 [RTW_XMIT_THREAD]

jphccfc, 2021-04-14:
This was directly after a reboot

roleoroleo, 2021-04-14:
Are you sure that it's not a power issue?
Did you change usb cable or power adapter?

jphccfc, 2021-04-14:
Another dump from later on

Mem: 57648K used, 3516K free, 1836K shrd, 948K buff, 10712K cached
CPU: 38.7% usr 12.8% sys  0.0% nic 48.2% idle  0.0% io  0.0% irq  0.1% sirq
Load average: 1.98 1.37 1.14 3/99 8403
  PID  PPID USER     STAT   VSZ %VSZ CPU %CPU COMMAND
 1029     1 root     S    40172 65.2   0 33.1 ./rmm
 1309     1 root     S     2944  4.7   0  3.9 ./mp4record
 1408     1 root     S<    4984  8.0   0  3.1 rRTSPServer                            402     2 root     SW       0  0.0   0  1.1 [irq/302-sunxi-m]
 1472     1 root     S     2420  3.9   0  0.5 framefinder
  529     2 root     SW       0  0.0   0  0.5 [ksdioirqd/mmc1]                         3     2 root     SW       0  0.0   0  0.5 [ksoftirqd/0]                         8322  8207 root     R      996  1.6   0  0.3 top
    7     2 root     SW       0  0.0   0  0.3 [rcu_preempt]
 1471     1 root     S     2072  3.3   0  0.2 wsdd --pid_file /var/run/wsdd.pid --i  536     1 root     S      984  1.6   0  0.2 ./dispatch                            8198  1341 root     S      900  1.4   0  0.2 dropbear -R -B                         465     2 root     SW       0  0.0   0  0.2 [mmcqd/0]
  532     2 root     SW       0  0.0   0  0.2 [RTW_XMIT_THREAD]
 5034     2 root     SW       0  0.0   0  0.2 [kworker/u2:0]
 1455     1 root     S     8108 13.1   0  0.0 onvif_srvd --pid_file /var/run/onvif_29729     1 root     S     1420  2.3   0  0.0 /home/base/tools/wpa_supplicant -c/tm 1424     1 root     S      984  1.6   0  0.0 {wd_rtsp.sh} /bin/sh /home/yi-hack/sc    1     0 root     S      980  1.5   0  0.0 /sbin/init
  486     1 root     S      980  1.5   0  0.0 -/bin/sh
 8207  8198 root     S      980  1.5   0  0.0 -ash
 8334  1424 root     S      980  1.5   0  0.0 sleep 10                             29801     1 root     S      980  1.5   0  0.0 udhcpc -i wlan0 -b -s /home/app/scrip 1341     1 root     S      876  1.4   0  0.0 dropbear -R -B                        1481     1 root     S      852  1.3   0  0.0 /home/yi-hack/usr/sbin/crond -c /var/ 1391     1 root     S      848  1.3   0  0.0 ntpd -p pool.ntp.org
 1321     1 root     S      848  1.3   0  0.0 httpd -p 8080 -h /home/yi-hack/www/ -  535     1 root     S      680  1.1   0  0.0 ./log_server
 1343     1 root     S      668  1.0   0  0.0 ipc_multiplexer
  533     2 root     SW       0  0.0   0  0.0 [RTW_CMD_THREAD]
26928     2 root     SW       0  0.0   0  0.0 [kworker/u2:2]
  481     2 root     SWN      0  0.0   0  0.0 [jffs2_gcd_mtd3]
  263     2 root     SW       0  0.0   0  0.0 [kswapd0]
  534     2 root     SW       0  0.0   0  0.0 [RTWHALXT]
  469     2 root     SW<      0  0.0   0  0.0 [kworker/0:1H]
  464     2 root     SW<      0  0.0   0  0.0 [bioset]
  405     2 root     SW       0  0.0   0  0.0 [irq/303-sunxi-m]
  378     2 root     SW<      0  0.0   0  0.0 [bioset]
  373     2 root     SW<      0  0.0   0  0.0 [bioset]
  383     2 root     SW<      0  0.0   0  0.0 [bioset]
  404     2 root     SW       0  0.0   0  0.0 [irq/166-sdc0 cd]
  353     2 root     SW<      0  0.0   0  0.0 [bioset]
  348     2 root     SW<      0  0.0   0  0.0 [bioset]
  358     2 root     SW<      0  0.0   0  0.0 [bioset]
  253     2 root     SW<      0  0.0   0  0.0 [spi0]
  232     2 root     SW<      0  0.0   0  0.0 [cfg80211]
  224     2 root     SW       0  0.0   0  0.0 [sys_user]
  239     2 root     SW<      0  0.0   0  0.0 [watchdogd]
  363     2 root     SW<      0  0.0   0  0.0 [bioset]
  183     2 root     SW<      0  0.0   0  0.0 [crypto]
  180     2 root     SW<      0  0.0   0  0.0 [writeback]
  179     2 root     SW       0  0.0   0  0.0 [oom_reaper]
  182     2 root     SW       0  0.0   0  0.0 [kcompactd0]
  184     2 root     SW<      0  0.0   0  0.0 [bioset]
    9     2 root     SW       0  0.0   0  0.0 [rcu_bh]
    8     2 root     SW       0  0.0   0  0.0 [rcu_sched]
   10     2 root     SW<      0  0.0   0  0.0 [lru-add-drain]
    5     2 root     SW<      0  0.0   0  0.0 [kworker/0:0H]
    2     0 root     SW       0  0.0   0  0.0 [kthreadd]
   11     2 root     SW       0  0.0   0  0.0 [kdevtmpfs]
  186     2 root     SW<      0  0.0   0  0.0 [kblockd]
  368     2 root     SW<      0  0.0   0  0.0 [bioset]
 8017     2 root     SW       0  0.0   0  0.0 [kworker/u2:1]
29699     2 root     SW       0  0.0   0  0.0 [kworker/0:2]
26437     2 root     SW       0  0.0   0  0.0 [kworker/0:1]

jphccfc, 2021-04-14:
> Are you sure that it's not a power issue?
> Did you change usb cable or power adapter?

I'm using the USB connector connected to a power extension....i will try with the plug that came with it.

jphccfc, 2021-04-14:
I just tried to run the RTSP feed and it crashed it..

roleoroleo, 2021-04-14:
Many users have noticed problems with non original power supplies.

jphccfc, 2021-04-14:
It's now in the original power supply. I will keep you updated

jphccfc, 2021-04-14:
Using the power supply and updated to the latest yihack ... Still crashes... Non responsive after an hour or two

roleoroleo, 2021-04-14:
I don't understand.
ps is ok and top is ok.
When it's not responsive, can you open a ssh session?

Please post your system.conf

jphccfc, 2021-04-14:
Just plugged it back in and now it's saying waiting to connect. How do i pull the system config..

jphccfc, 2021-04-14:
I've reconnected it and can now access yihack web and ssh

roleoroleo, 2021-04-15:
You can save the config from the web page "Maintenance" or copying the files in /home/yi-hack/etc

jphccfc, 2021-04-15:
Will there be any sensitive data in there? How is best to share with you?

roleoroleo, 2021-04-15:
I need only system.conf, you can copy and paste it here.

jphccfc, 2021-04-16:
Here you go

`HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=no
ONVIF=yes
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=
PASSWORD=
TIMEZONE=EST5EDT,M3.2.0,M11.1.0
FREE_SPACE=20
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=
CRONTAB=
SPEAKER_AUDIO=yes
`

jphccfc, 2021-04-16:
It crashed again yesterday at 7.25pm. had been up for probably around 12 hours, the longest it has ever been. I have left it cloud and connected to the yi app this time

roleoroleo, 2021-04-16:
What type of client/protocol are you using?
onvif client?
rtsp?
Do you acquire snapshot continuosly?

jphccfc, 2021-04-16:
I only check the rtsp stream every now and again. I'm only using the defaul yihack web client and yihome app at the moment. My plan was to use home assistant but not until I have a stable cam. I bought two cameras and the other one doesn't have yihack applied yet and it has been rock solid. I dont use the snapshot functionality.

jphccfc, 2021-04-17:
Crashed again at 827pm... Probably 12 hours of uptime only

roleoroleo, 2021-04-17:
Have you already tried to remove the hack and apply it again?

jphccfc, 2021-04-17:
No but i can do

roleoroleo, 2021-04-17:
Read the wiki.
Maybe something wrong during the update.

jphccfc, 2021-04-17:
Ok that had been completed successfully. Confirmed by the fact the camera is working in the yi app but no access to it via yihack web server

jphccfc, 2021-04-17:
Updated based on the y20 version on the git. Web server is working fine so far. It does state it's v0.24 and not v0.25? Should i do The FW update update online to v0.25

tachyon142, 2021-04-18:
Shot in the dark here, but do you have AI Human Detection disabled in Camera Settings?

jphccfc, 2021-04-18:
Hi tachyon i have it enabled. Since the reinstall i have uptime of 24hours

roleoroleo, 2021-04-18:
@jphccfc 
0.2.4?
Which fw did you use for the update?

jphccfc, 2021-04-18:
It say 0.24. I downloaded direct from github

jphccfc, 2021-04-20:
Now been up 2d 7h 54m 6s. Seems very promising

jphccfc, 2021-04-26:
This can be closed. Reinstall fixed it

closed 2021-04-26 by jphccfc