roleoroleo/yi-hack-Allwinner #284: IR light / night vision won't turn off

phoenix1887, 2021-04-13:
This seems to be a bigger problem with all the yi cameras as it doesn't work for any of the 12 cameras I have (different models). But I have several with each of the hacks, allwinner, mstar... You can turn the IR light off for any of them in the yi hack settings or via the yi home cloud app but no matter what it doesn't disable night vision, it continues using it. So there seems to be no way to disable night vision on any of the cameras despite the setting. It seems like all of these hacks break the IR / night vision switch.

Can anyone else confirm this IS indeed the case or that the setting is working properly or their cams? By the way cycling the lights on and off and rebooting the camera doesn't help...

To test it out disable the IR light light setting on the camera then dim the lights in the room until you see the camera switch to night vision... Even though it shouldn't...

Anyone have any ideas how to fix this?

oigan, 2021-09-19:
yep .. its a nice "feature". i just hate this &*^&*& Night Mode and that clank sound. lets hope that someone will find a solution to switch ON/OFF this NightMode  :) @roleoroleo  maybe you can take a look, please?!

roleoroleo, 2021-09-20:
Already tried.
No way at the moment.

tachyon142, 2022-01-17:
Late reply, but this is related to this issue: [https://github.com/roleoroleo/yi-hack-Allwinner/issues/194](url)

It's not an issue of the hacks breaking an IR/Night Vision switch, the issue is that night vision isn't tied to the IR LED Enable switch in that way to begin with. Night vision is controlled by the Yi processes, and a way to change it has not yet been found. Yi's "night vision" consists of three simple components: a camera module sensitive to IR, a lens housing with an IR-cut filter on a solenoid, and a black/white filter in ffmpeg. 

When sufficient external light is present, the Yi process engages the IR-cut filter, so the video doesn't appear pink (the effect of the infraRED waves that we don't usually see). When the Yi process has decided it's dark enough to need night vision, it disables the IR-cut filter so gets more light, and engages the black/white filter in ffmpeg so we can't tell that the video is pink. If the IR LED Enable switch is "on", then it also turns the LEDs on.

In this scheme, the sensor is always sensitive to all colors + IR, and both the IR-cut filter, and the black/white filter really only serve to prevent the video from looking pink. Which one is active depends on the available light. The bad news is that we can't control when Yi decides to switch modes. The good news is that if you don't mind seeing pink in your daytime video, you can eliminate the annoying clunk noise by disabling the IR-cut filter as I described in the linked issue.

closed 2024-05-10 by github-actions[bot]