roleoroleo/yi-hack-Allwinner #285: RTSP Home Assistant and proxy servers

tunerooster, 2021-04-13:
@roleoroleo, I wonder if I could get your opinion on this?  As you know, proxy servers, like live555ProxyServer, open a single connection to an rtsp server and publish it to many clients, thereby avoiding many clients directly opening simultaneous connections with the cameras rtsp.

However, it adds a layer of complexity to the connections (e.g., UDP/TCP from camera to live555 and UDP/TCP from live555 to the client, HA).

But I have 9 dashboards running HA dashboards and when the doorbell rings, all 9 dashboards display the live front door.  Not to mention continuous snapshots via ffmpeg on all 6 cameras.

Because of your experience getting RTSP running on the Yi cameras, and because I read somewhere that you used live555 as a template for rRTSPServer, I expect you are knowledgeable about this.

_My questions is simply, is it good/better to use live555, or is direct to the camera rRTSPServer a better/preferable way to go, given my scenerio above?_

Thanks very much for any advice you can offer!

roleoroleo, 2021-04-13:
The scenario is clear but I don't understand the question.
Please, explain me better.

tunerooster, 2021-04-14:
I am asking if you would recommend using live555ProxyServer, or use 9 open
connections directly to the camera.  Thanks for your advice!

roleoroleo, 2021-04-14:
I think that the cam is not able to stream video to 9 clients.
It uses unicast.
A proxy is needed.

tunerooster, 2021-04-14:
Thanks, that is very useful advice...

closed 2021-04-14 by tunerooster

tunerooster, 2021-08-18:
Here is the screenshot showing the "Microphone on/off" option.  It is under
"Device Settings" in the Yi App.