roleoroleo/yi-hack-Allwinner #286: 0.2.5

robs1234, 2021-04-13:
Installed but not working.
Get as far as the main web page but only the top banner/headers showing, & rtsp not working in VLC.
I'm afraid I know sweetnothing about troubleshooting/logs how to etc...

sermayoral, 2021-04-13:
@robs1234 finally, what version do you have?

robs1234, 2021-04-13:
loaded 0.2.3 but it shows as 0.2.4...

roleoroleo, 2021-04-14:
> loaded 0.2.3 but it shows as 0.2.4...

Maybe there was a problemi during the update procedure.
Please, restart from scratch.
Reformat the sd and apply the hack again.

robs1234, 2021-04-14:
Thanks roleoroleo, tried that again, but same result - still showing as 2.4 on the maintenance page.

No worries, seems to be working fine anyway.

roleoroleo, 2021-04-14:
Last chance: clean the cache of your browser.

robs1234, 2021-04-14:
Tried upgrade to 0.2.5 again and now all is working fine - many thanks roleoroleo!

Arkady23, 2021-04-17:
I did the formatting via the web-interface.
https://github.com/Arkady23/yi-hack-Allwinner-plus

closed 2024-05-10 by github-actions[bot]