roleoroleo/yi-hack-Allwinner #287: 9FUS Y20ga brick

thetux07, 2021-04-14:
hello my camera have a orange led static on boot and nothing is happening
have you a origine firmware or backup or proce for repare my cam ?
thank you

roleoroleo, 2021-04-14:
This is the unbrick procedure: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam
But, before to use this procedure, try to unhack the cam with this one: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Remove-the-hack

thetux07, 2021-04-14:
ok thank you, yes i have already unhack the camera but it don't work to, for the unbrick camera procedure, i have read it but i don't have backup :(, how made without it ?
thank you

roleoroleo, 2021-04-14:
Let me know when you finish to download the file:
[mtdblock3.bin.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6309906/mtdblock3.bin.gz)

thetux07, 2021-04-14:
ok thank you, i've dowload it
i test unbrick procedure and i come back to you
thank you

LZsolt75, 2021-04-19:
I would like to upgrade from 0.2.4 to 0.2.5 through the web gui maintenance menu. After a lot of waiting the camera is unreachble and the orange led is lit continuosly. I couldn't reach the web gui.
What should I do?

roleoroleo, 2021-04-19:
Try to apply again the hack.

LZsolt75, 2021-04-19:
> 
> 
> Try to apply again the hack.

Looks like it solve my problem. I thin the main problem is that the SD card was full so the updater have no space to do its job.

closed 2021-04-27 by roleoroleo