roleoroleo/yi-hack-Allwinner #289: temporary guest wifi

Arkady23, 2021-04-15:
In yi-hack-v3, it is possible to connect to wifi with the data that was on the SD. If the SD card is removed, the connection went with the old wifi data. There was a script on the sd that copied the wpa_supplicant.conf file to /tmp/. It was very convenient and safe, so as not to lose control of the camera. Pulled out the sd and everything went back. It doesn't work that way right now. What is offered via the bin file is somehow not elegant, suitable for changing the wifi to permanently. And I would like to have a guest temporary wifi. While I'm thinking about how to do this, I decided to write.

Arkady23, 2021-04-15:
I think I found a way. In my opinion, you need to put the wpa_supplicant.conf file on the sd in the Factory folder. Tomorrow I'll take the camera to visit and check it out.

Arkady23, 2021-04-16:
Perfectly work.
Don't forget to edit "Your_WiFi" and "Your_password"!!!
[wifi_on_sd.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/6323138/wifi_on_sd.zip)

Arkady23, 2021-05-28:
This feature has nothing to do with the hack. Don't interfere with the hack here. This is an option of the camera manufacturer. So it is provided, if there is such a file on the sd, then it is included in the work. And another such moment. This works on Allwinner models cameras. Is this feature available for allwinnerV2 cameras.... I don't know.

Pulsar981, 2021-05-28:
I understand, haven't considered that this might not be available in the allwinnerV2. Seems like I have to keep trying other ways. Thanks

WalkingDot, 2021-05-30:
It's working on Allwinner-v2, too.
However, I would recommend using the app for the initial setup.

Pulsar981, 2021-05-31:
@WalkingDot have you tried that succuessfully? If so, could you please describe your process of doing that?
I've also tried with the yi-app to initialize first, then done the allwinnerV2 hack, then copied the modified wpa_supplicant.conf (with another ssid and pw) to the "Factory.done" folder (also tried copy to the SD root folder), but it keeps using the initialized wifi data ...

WalkingDot, 2021-05-31:
@Pulsar981 
Remove all files from the "Factory.done" folder and rename it to "Factory".
Then copy the "wpa_supplicant.conf" into the "Factory" folder.

Pulsar981, 2021-06-04:
That worked for me! Thanks @WalkingDot
I've tried two ways succesfully:
a) When initialized with the yi-app before the hack, I had also to delete the "log"-folder (since there are the old wifi-settings stored). Then proceed how you advised - worked.
b) When reset the camera and do the hack without initializing with the yi-app before, it also worked with your advise. Camera connects on startup automatically to the wifi set up in the wpa_supplicant.conf.

closed 2024-05-10 by github-actions[bot]