roleoroleo/yi-hack-Allwinner #290: No sound detection MQTT

Mugen4, 2021-04-17:
I have 0.2.5 and 2020 cam firm with MQTT to HA configured.

Motion_detection with ai_human_detection off works ok, but sound_detection have never worked to me.

With MQTT explorer see no sound_detection topic on mqtt, but I see baby_crying as always crying.

I tried to reset, enable/disable options, change sound sensitivity... without success. I have sound detection on and RTSP Audio to pcm.

roleoroleo, 2021-04-17:
If ai_human_detection is off it should work.
I retested it now and the event is triggered.

When you receive the stream, is audio ok?

Mugen4, 2021-04-17:
Well I dont know how, maybe after power off and not reboot, but now I'm seeing sound_detection topic and detect sound. 

The problem I have now is when a sound is detected, the "rearm time" is too high, 5min. Can I change this timer?

Thanks for your work and support.

roleoroleo, 2021-04-17:
> The problem I have now is when a sound is detected, the "rearm time" is too high, 5min. Can I change this timer?

I don't know. I have to check if the rearm depends on the yi firmware or on my hack.

closed 2024-05-10 by github-actions[bot]