roleoroleo/yi-hack-Allwinner #291: HA integration

roleoroleo, 2021-04-21:
Please, take a look at this project:

https://github.com/roleoroleo/yi-hack_ha_integration

pbanj, 2021-05-03:
this is amazing. i had just asked a buddy if he could try and make one as the current way of doing it was a pain in the ass. keep up the good work

pbanj, 2021-05-03:
hmmm seems motion detection isnt working. their status states never seem to change

roleoroleo, 2021-05-03:
Check if motion detection is on in your yi app and if you configured mqtt parameters.
The integration requires mqtt.

pbanj, 2021-05-03:
> 
> 
> Check if motion detection is on in your yi app and if you configured mqtt parameters.
> The integration requires mqtt.

ah didnt know it needed mqtt. just turned it on and it's working now. one thing i noticed. turning on the ai detection makes it not send anything

roleoroleo, 2021-05-03:
There is a know issue with ai.
And it's not related to the hack.

rickytugapt, 2021-05-19:
Hi, 
before my question, thanks for your amazing work, this integration is aswome for noobs like me.
Now my question if you can help me. My camera has ptz support (r30gb with allwinner-v2 0.1.7), but i can't find where to "set" it inside home assistant. Can you give me some help?
Thanks.

roleoroleo, 2021-05-20:
ptz is a service.
You can run it for testing purpose in the "Developer tools" page.
If you want to use from lovelace, you have to add a button or something else.
For example I use a picture-glance (after creating 4 switches):
```
camera_image: camera.kitchen
entities:
  - entity: switch.switch_cam_kitchen_left
    icon: 'mdi:arrow-left-bold-circle-outline'
  - entity: switch.switch_cam_kitchen_up
    icon: 'mdi:arrow-up-bold-circle-outline'
  - entity: switch.switch_cam_kitchen_down
    icon: 'mdi:arrow-down-bold-circle-outline'
  - entity: switch.switch_cam_kitchen_right
    icon: 'mdi:arrow-right-bold-circle-outline'
tap_action:
  action: url
  url_path: 'rtsp://192.168.1.200'
title: Kitchen
type: picture-glance
```

rickytugapt, 2021-05-20:
> ptz is a service.
> You can run it for testing purpose in the "Developer tools" page.
> If you want to use from lovelace, you have to add a button or something else.
> For example I use a picture-glance [...]

That's it :) i didn't realise it was a service. Thank you very much.

rickytugapt, 2021-05-20:
Hi again roleoroleo,
I was testing it in Developer tools and realised that i can't use a travel time bigger then 1, or it won't move.
Do you have any sugestion to implement some kind of presets? Or the only way is completely move to one side then start "counting" the time until the position i want?

roleoroleo, 2021-05-20:
I don't know why you can't use a number bigger than 1.
The max value is set to 5.
Try to send a command from the shell.
```
ipc_cmd -m right
ipc_cmd -m stop
```

If I'm not wrong you can create presets using the original app.
After you save them, you can call from the hack.

rickytugapt, 2021-05-20:
I'm really a noob, i don't know how to send 

ipc_cmd -m right
ipc_cmd -m stop

I figured how to make a lovelace button to press and make the camera move, but if i set a travel_time bigger then 1 i get the following:
[image]
If i use 1 or less works great.
But anyway, this isn't really important to me, i can live with that. Can you just point me how to create a switch to use with the picture glance?

Thanks one more time for your patience to a noob like me.

roleoroleo, 2021-05-21:
I will check the problem about travel_time.

Try to create a switch with this syntax:
```
  - platform: template
    switches:
      switch_cam_kitchen_left:
        value_template: "off"
        turn_on:
          service: yi_hack.ptz
          data:
            entity_id: camera.kitchen
            movement: left
            travel_time: 0.3
        turn_off:
```

roleoroleo, 2021-05-21:
Fixed error when travel_time > 1

rickytugapt, 2021-05-21:
Updated but I'm getting the same error if travel_time >1. 
But finally i created the switchs and already have a nice picture glance with ptz. 

Thanks a lot for your help :)

closed 2024-05-10 by github-actions[bot]