roleoroleo/yi-hack-Allwinner #294: [y20ga] Baby crying detection not working?

vin-w, 2021-04-26:
In mqtt, I always get crying status for "yicam/baby_crying" since started the cam.
In the app, it could detect crying event. So I am sure that the crying detection is working in the camera and should not be always "crying"

```
Firmware Version | 0.2.4
Base Version | 8.2.0.0A_202012041528
Model Suffix | y20ga
```

App setting:
|Config|Value|
|---|---|
|Motion detection|yes|
|Human detection|no|
|Baby crying detecion|yes|

```
$  mosquitto_sub -d -t yicam/baby_crying
Client mosqsub|7203-raspberryp sending CONNECT
Client mosqsub|7203-raspberryp received CONNACK (0)
Client mosqsub|7203-raspberryp sending SUBSCRIBE (Mid: 1, Topic: yicam/baby_crying, QoS: 0)
Client mosqsub|7203-raspberryp received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|7203-raspberryp received PUBLISH (d0, q0, r1, m0, 'yicam/baby_crying', ... (6 bytes))
crying
```

roleoroleo, 2021-04-27:
There is a problem with baby crying detection.
Because some month ago the setting in the app disappeared and changed to sound detection.
I don't know if I able to receive baby crying messages from the system.

When you use the app, do you receive "baby crying" messages or in general "abnormal sound" messages?

vin-w, 2021-04-28:
I am not sure if it is related to camera region problem, but for my yi cam y20ga with "serial_number":"9FCN......." 
there is baby crying setting and no abnormal sound.

With yi app from China ios app store with the latest version installed,
there is baby crying setting and I could receive baby crying notification (see all the screenshots)

There isn't a setting about abnormal sound which is shown in http://my_ip:8080/?page=camera_settings

If you want me to check some internal commands from camera ssh, I could help.

Thank you!

[image]

[image]

[image]

[image]

roleoroleo, 2021-04-28:
Because I'm not sure about this setting, I didn't remove baby crying from the hack.
So, if the system send me the message, I convert it to a mqtt message.
Please check if the system sends the IPC message using this app:
[ipc_sniff.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6393595/ipc_sniff.tar.gz)

- decompress it to /tmp/sd
- kill mqttv4 app
- run ipc_sniff from shell
- capture the standard output while the cam detects a baby cryng event
- post the output

vin-w, 2021-04-29:
Hi roleoroleo,

After some investigation and reading the last commit in https://github.com/roleoroleo/yi-hack_ha_integration/commit/26ef0964d52ff60fd4ef276b897b7853089dfd18#diff-c7cee1ec54e93d5cea9fe44ae92f8524b8ea1947a90e253c54fa352b8646f300

### baby cry detection event 

Correct me if I am wrong.
If I subscribe the mqtt event "yicam/baby_crying", the value is always "crying".
So in order to detect a new crying start, it means receiving a new yicam/baby_crying publishing  "crying"
But it does not have a clear event of crying is end.

I messed up because for motion detection, we have "motion_start", "motion_stop" value
and so I thought new crying start = "crying" and crying stop = "" empty or sth else.

In the yi hack HASS intergration, there is a _off_delay of 60 sec to complement that (not sure if it is good to do similar in the yi-hack firmware by empty the crying after 60 sec too)

```
 $ mosquitto_sub -d -t yicam/baby_crying
Client mosqsub|11445-raspberry sending CONNECT
Client mosqsub|11445-raspberry received CONNACK (0)
Client mosqsub|11445-raspberry sending SUBSCRIBE (Mid: 1, Topic: yicam/baby_crying, QoS: 0)
Client mosqsub|11445-raspberry received SUBACK
Subscribed (mid: 1): 0
Client mosqsub|11445-raspberry received PUBLISH (d0, q0, r1, m0, 'yicam/baby_crying', ... (6 bytes))
crying ######## this is the value I received when subscribed
Client mosqsub|11445-raspberry sending PINGREQ
Client mosqsub|11445-raspberry received PINGRESP
Client mosqsub|11445-raspberry sending PINGREQ
Client mosqsub|11445-raspberry received PINGRESP
Client mosqsub|11445-raspberry received PUBLISH (d0, q0, r0, m0, 'yicam/baby_crying', ... (6 bytes))
crying  ######## this is a new baby crying event
Client mosqsub|11445-raspberry sending PINGREQ
Client mosqsub|11445-raspberry received PINGRESP
Client mosqsub|11445-raspberry sending PINGREQ
```

### baby cry detection setting 

I used ipc_sniff and I could confirm that the baby cry setting on/off is still valid in the source code
https://github.com/roleoroleo/yi-hack-Allwinner/blob/master/src/ipc_cmd/ipc_cmd/ipc_cmd.h

I hope both sound detection and baby crying setting could still keep in the yi-hack config web. It would be good to have all the settings w/o thinking about the camera model and region.

```
#define IPC_BABYCRYING_OFF "\x02\x00\x00\x00\x08\x00\x00\x00\x29\x10\x01\x00\x04\x00\x00\x00\x01\x00\x00\x00"
#define IPC_BABYCRYING_ON  "\x02\x00\x00\x00\x08\x00\x00\x00\x29\x10\x01\x00\x04\x00\x00\x00\x02\x00\x00\x00"
```

Capture
```
# baby cry setting off
Parsing message.
02 00 00 00 08 00 00 00 29 10 01 00 04 00 00 00 01 00 00 00 
GOT UNRECOGNIZED MESSAGE

# baby cry setting on
IPC message. Len: 20. Status: No error information!
Parsing message.
02 00 00 00 08 00 00 00 29 10 01 00 04 00 00 00 02 00 00 00 
GOT UNRECOGNIZED MESSAGE
```

roleoroleo, 2021-04-29:
> If I subscribe the mqtt event "yicam/baby_crying", the value is always "crying".
> So in order to detect a new crying start, it means receiving a new yicam/baby_crying publishing "crying"
> But it does not have a clear event of crying is end.

Correct.

> I messed up because for motion detection, we have "motion_start", "motion_stop" value
> and so I thought new crying start = "crying" and crying stop = "" empty or sth else.
>
> In the yi hack HASS [...]

Correct. The yi system doesn't send a message to close a baby crying event or a sound detection event.
This is the reason why I added the off_delay parameter in the wiki: https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Home-Assistant-integration

I will not remove the baby crying settings.