roleoroleo/yi-hack-Allwinner #295: nanotts not working

vin-w, 2021-04-29:
hello!

with version 0.25 y20ga
I was unable to execute the nanotts binary.  Any idea?

p.s. I have updated the speak.sh script to latest version from git

```
root@babyyi:/# ls -al /tmp/sd/yi-hack/bin/nanotts
-rwxr-xr-x    1 root     root        193440 Mar 31 19:34 /tmp/sd/yi-hack/bin/nanotts
root@babyyi:/# /tmp/sd/yi-hack/bin/nanotts
-ash: /tmp/sd/yi-hack/bin/nanotts: not found

```

```
mylinuxbox:~ $ wget -v http://my_ip:8080/cgi-bin/speak.sh?lang=en-US --user=XXX --password=XXX --post-data="morning"
--2021-04-30 03:17:44--  http://my_ip:8080/cgi-bin/speak.sh?lang=en-US
Connecting to my_ip:8080... connected.
HTTP request sent, awaiting response... 401 Unauthorized
Authentication selected: Basic realm="Web Server Authentication"
Connecting to my_ip:8080... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/json]
Saving to: 'speak.sh?lang=en-US.10'

speak.sh?lang=en-US.10            [  <=>                                           ]      44  41.0 B/s    in 1.1s    

2021-04-30 03:17:46 (41.0 B/s) - 'speak.sh?lang=en-US.10' saved [44]

pi@raspberrypi:~ $ cat speak.sh?lang=en-US.10
{
"error":"false",
"description":"morning"
```

```
root@babyyi:/tmp/sd/yi-hack# ls -alR
.:
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 .
drwxr-xr-x   12 root     root         32768 Apr 29 19:17 ..
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 bin
-rwxr-xr-x    1 root     root       4805155 Apr 28 18:56 nanotts.tgz
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 usr

./bin:
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 ..
-rwxr-xr-x    1 root     root        193440 Mar 31 19:34 nanotts

./usr:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 share

./usr/share:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 pico

./usr/share/pico:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 lang

./usr/share/pico/lang:
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
-rwxr-xr-x    1 root     root        634996 Mar 31 19:05 de-DE_gl0_sg.bin
-rwxr-xr-x    1 root     root        440732 Mar 31 19:05 de-DE_ta.bin
-rwxr-xr-x    1 root     root        584436 Mar 31 19:05 en-GB_kh0_sg.bin
-rwxr-xr-x    1 root     root        412248 Mar 31 19:05 en-GB_ta.bin
-rwxr-xr-x    1 root     root        777396 Mar 31 19:05 en-US_lh0_sg.bin
-rwxr-xr-x    1 root     root        650668 Mar 31 19:05 en-US_ta.bin
-rwxr-xr-x    1 root     root        256744 Mar 31 19:05 es-ES_ta.bin
-rwxr-xr-x    1 root     root        605280 Mar 31 19:05 es-ES_zl0_sg.bin
-rwxr-xr-x    1 root     root        833236 Mar 31 19:05 fr-FR_nk0_sg.bin
-rwxr-xr-x    1 root     root        381936 Mar 31 19:05 fr-FR_ta.bin
-rwxr-xr-x    1 root     root        628268 Mar 31 19:05 it-IT_cm0_sg.bin
-rwxr-xr-x    1 root     root        252044 Mar 31 19:05 it-IT_ta.bin
```

roleoroleo, 2021-04-29:
Please, show me the content of speak.sh
-------- Messaggio originale --------Da: Vin W ***@***.***> Data: 29/04/21  21:25  (GMT+01:00) A: roleoroleo/yi-hack-Allwinner ***@***.***> Cc: Subscribed ***@***.***> Oggetto: [roleoroleo/yi-hack-Allwinner] nanotts not working (#295) 
hello!
with version 0.25 y20ga
I was unable to execute the nanotts binary.  Any idea?
p.s. I have updated the speak.sh script to latest version from git
***@***.***:/# ls -al /tmp/sd/yi-hack/bin/nanotts
-rwxr-xr-x    1 root     root        193440 Mar 31 19:34 /tmp/sd/yi-hack/bin/nanotts
***@***.***:/# /tmp/sd/yi-hack/bin/nanotts
-ash: /tmp/sd/yi-hack/bin/nanotts: not found

mylinuxbox:~ $ wget -v http://my_ip:8080/cgi-bin/speak.sh?lang=en-US --user=XXX --password=XXX --post-data="morning"
--2021-04-30 03:17:44--  http://my_ip:8080/cgi-bin/speak.sh?lang=en-US
Connecting to my_ip:8080... connected.
HTTP request sent, awaiting response... 401 Unauthorized
Authentication selected: Basic realm="Web Server Authentication"
Connecting to my_ip:8080... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/json]
Saving to: 'speak.sh?lang=en-US.10'

speak.sh?lang=en-US.10            [  <=>                                           ]      44  41.0 B/s    in 1.1s    

2021-04-30 03:17:46 (41.0 B/s) - 'speak.sh?lang=en-US.10' saved [44]

***@***.***:~ $ cat speak.sh?lang=en-US.10
{
"error":"false",
"description":"morning"

***@***.***:/tmp/sd/yi-hack# ls -alR
.:
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 .
drwxr-xr-x   12 root     root         32768 Apr 29 19:17 ..
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 bin
-rwxr-xr-x    1 root     root       4805155 Apr 28 18:56 nanotts.tgz
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 usr

./bin:
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 ..
-rwxr-xr-x    1 root     root        193440 Mar 31 19:34 nanotts

./usr:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    4 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 share

./usr/share:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 pico

./usr/share/pico:
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 lang

./usr/share/pico/lang:
drwxr-xr-x    2 root     root         32768 Apr 29 18:01 .
drwxr-xr-x    3 root     root         32768 Apr 29 18:01 ..
-rwxr-xr-x    1 root     root        634996 Mar 31 19:05 de-DE_gl0_sg.bin
-rwxr-xr-x    1 root     root        440732 Mar 31 19:05 de-DE_ta.bin
-rwxr-xr-x    1 root     root        584436 Mar 31 19:05 en-GB_kh0_sg.bin
-rwxr-xr-x    1 root     root        412248 Mar 31 19:05 en-GB_ta.bin
-rwxr-xr-x    1 root     root        777396 Mar 31 19:05 en-US_lh0_sg.bin
-rwxr-xr-x    1 root     root        650668 Mar 31 19:05 en-US_ta.bin
-rwxr-xr-x    1 root     root        256744 Mar 31 19:05 es-ES_ta.bin
-rwxr-xr-x    1 root     root        605280 Mar 31 19:05 es-ES_zl0_sg.bin
-rwxr-xr-x    1 root     root        833236 Mar 31 19:05 fr-FR_nk0_sg.bin
-rwxr-xr-x    1 root     root        381936 Mar 31 19:05 fr-FR_ta.bin
-rwxr-xr-x    1 root     root        628268 Mar 31 19:05 it-IT_cm0_sg.bin
-rwxr-xr-x    1 root     root        252044 Mar 31 19:05 it-IT_ta.bin

vin-w, 2021-04-29:
```
root@yicam:/home/yi-hack/www/cgi-bin# ls
camera_settings.sh       hostname.js              save.sh
eventsdir.sh             links.sh                 service.sh
eventsdirdel.sh          load.sh                  set_configs.sh
eventsfile.sh            preset.sh                snapshot.sh
eventsfiledel.sh         ptz.sh                   speak.sh
fw_upgrade.sh            reboot.sh                status.json
get_configs.sh           record.sh                wifi.sh
getlastrecordedvideo.sh  reset.sh
root@yicam:/home/yi-hack/www/cgi-bin# cat speak.sh 
#!/bin/sh

LANG="en-US"

validateLang()
{
    RES=$(echo ${1} | sed -E 's/[a-zA-Z-]//g')
    if [ -z "$RES" ]; then
        LANG=$1
    else
        LANG="invalid"
    fi
}

PARAM="$(echo $QUERY_STRING | cut -d'&' -f1 | cut -d'=' -f1)"
VALUE="$(echo $QUERY_STRING | cut -d'&' -f1 | cut -d'=' -f2)"

if [ "$PARAM" == "lang" ] ; then
    LANG="$VALUE"
fi

validateLang $LANG
if [ "$LANG" == "invalid" ]; then
    printf "{\n"
    printf "\"%s\":\"%s\",\\n" "error" "true"
    printf "\"%s\":\"%s\"\\n" "description" "Invalid language"
    printf "}"
    exit
fi

read -r POST_DATA

printf "Content-type: application/json\r\n\r\n"

if [ -f /tmp/sd/yi-hack/bin/nanotts ] && [ -e /tmp/audio_in_fifo ]; then
    TMP_FILE="/tmp/sd/speak.pcm"
    if [ ! -f $TMP_FILE ]; then
        echo "$POST_DATA" | /tmp/sd/yi-hack/bin/nanotts -l /tmp/sd/yi-hack/usr/share/pico/lang -v $LANG -c > $TMP_FILE
        cat $TMP_FILE > /tmp/audio_in_fifo
        sleep 1
        rm $TMP_FILE

        printf "{\n"
        printf "\"%s\":\"%s\",\\n" "error" "false"
        printf "\"%s\":\"%s\"\\n" "description" "$POST_DATA"
        printf "}"
    else
        printf "{\n"
        printf "\"%s\":\"%s\",\\n" "error" "true"
        printf "\"%s\":\"%s\"\\n" "description" "Speaker busy"
        printf "}"
    fi
else
    if [ ! -f /tmp/sd/yi-hack/bin/speak ]; then
        printf "{\n"
        printf "\"%s\":\"%s\"\\n" "error" "true"
        printf "\"%s\":\"%s\"\\n" "description" "TTS engine not found"
        printf "}"
    elif [ ! -e /tmp/audio_in_fifo ]; then
        printf "{\n"
        printf "\"%s\":\"%s\"\\n" "error" "true"
        printf "\"%s\":\"%s\"\\n" "description" "Audio input disabled"
        printf "}"
    fi
fi

```

roleoroleo, 2021-04-29:
Are you sure you downloaded nanotts for Allwinner?

vin-w, 2021-04-29:
quite sure I downloaded correct one.
I think failed to exec the bin is a problem?

```
root@yicam:/# ls -al /tmp/sd/yi-hack/bin/nanotts
-rwxr-xr-x    1 root     root        193440 Mar 31 19:34 /tmp/sd/yi-hack/bin/nanotts
root@yicam:/# /tmp/sd/yi-hack/bin/nanotts
-ash: /tmp/sd/yi-hack/bin/nanotts: not found
```

```
root@yicam:/tmp/sd/yi-hack# ls -al nanotts.tgz. # the copy i downloaded
-rwxr-xr-x    1 root     root       4805155 Apr 28 18:56 nanotts.tgz
```

```
$ ls -al *.tgz
-rw-r--r--@ 1 vin  staff  4805155 Apr 30 04:34 nanotts_allwinner.tgz
-rw-r--r--@ 1 vin  staff  4805155 Apr 30 04:34 nanotts_allwinner2.tgz
-rw-r--r--@ 1 vin  staff  4805171 Apr 30 04:34 nanotts_mstar.tgz
$ md5 nanotts_allwinner.tgz
MD5 (nanotts_allwinner.tgz) = 3d5a2ec6006ffab094c88bb4831dfefe
$ md5 nanotts_allwinner2.tgz
MD5 (nanotts_allwinner2.tgz) = 3d5a2ec6006ffab094c88bb4831dfefe
```

roleoroleo, 2021-04-29:
Try to download it again.
I updated it.

closed 2021-05-01 by vin-w

antuanix, 2021-05-01:
Can anybody point me to where to get nanotts for Allwinner?

roleoroleo, 2021-05-01:
https://github.com/roleoroleo/yi-hack-utils