roleoroleo/yi-hack-Allwinner #296: RTSP not working after upgrade to 0.2.6

GqSteve, 2021-05-03:
I upgraded two cams:
a. yi-hack-MStar from 0.4.0 to 0.4.1
b. yi-hack-Allwinner from 0.2.5 to 0.2.6

and both have the following same issues:
1. RTSP: streaming is not working
2. Snapshot: snapshot is not working, e.g. http://xx.xx.xx.xx:8080/cgi-bin/snapshot.sh?res=high&watermark=yes

Nevertheless, MQTT snapshots and videos are working.

roleoroleo, 2021-05-07:
Which rtsp client are you using?

GqSteve, 2021-05-08:
Regarding yi-hack-MStar, I found out meanwhile that some configuration items was “lost” after the upgrade. After I recovered the whole configuration, RTSP is working again.

Regarding the yi-hack-Allwinner, RTSP was switched on, but there was no streaming there. I scanned all ports on the cam -> all ports from the services like SSH, HTTP… are active but not the 554 for RTSP. That was a little bit suspicious and I could only thing of the microSD card. In order to avoid bothering you with this, I ordered a SanDisk MAX ENDURANCE microSD card…which should last a little bit longer than the usual microSD cards :)

closed 2021-05-08 by GqSteve