roleoroleo/yi-hack-Allwinner #297: 9FUS on 0.2.5 crashes frequently while AI Human Detection is turned Off

tachyon142, 2021-05-03:
Firmware Version: 0.2.2, 0.2.5 & 0.2.6 (tested on all three)
Base Version:8.3.0.0A_201912271707
Model Suffix: y25ga
Serial Number: 9FUSY32...

Greetings. I'm experiencing issues where my above camera (and 3x more of the same) frequently crash and reboot while AI Human Detection is turned off. Camera is stable otherwise, and has achieved 30days uptime with these settings:

**camera.conf:**
SWITCH_ON=yes
SAVE_VIDEO_ON_MOTION=no  (I like to always save video in case motion is missed)
SENSITIVITY=high
AI_HUMAN_DETECTION=yes
BABY_CRYING_DETECT=undefined
LED=no
ROTATE=no
IR=yes
SOUND_DETECTION=no
SOUND_SENSITIVITY=80

**system.conf**
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=yes
BUSYBOX_FTPD=no
DISABLE_CLOUD=yes
REC_WITHOUT_CLOUD=yes
MQTT=yes
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=yes
ONVIF=yes
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=yes
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=no
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=*****
PASSWORD=*****
TIMEZONE=CST6CDT,M3.2.0,M11.1.0
FREE_SPACE=5
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=*****
CRONTAB=
SPEAKER_AUDIO=yes

My goal is to use the cam as a motion detector inside of HA, and mqtt is configured correctly. When I disable AI Human Detection (in order to get motion_start messages per the other known issues), I do begin seeing motion_start messages as expected, but the camera crashes. Previous diagnoses yielded only a variety of uptimes between crashes, but I'm starting to believe that the crashes coincide with would-be motion_stop events (this is unverified). Crashes continue frequently until AI Human Detection is re-enabled, at which point the cam becomes stable again. Is this behavior indicative of a botched install or problem with the Yi base firmware? I don't see any others reporting this issue, so I assume it must be unique to my devices, but it's odd that all 4 exhibit the same behavior.

Another, cam (older 9FUSY31 on base 8.2....) is working fine with motion messages while AI Human Detection is Off, and no crashes. What's wrong with my others?

Thanks and best regards.

roleoroleo, 2021-05-03:
Probably the cam crashes when it runs a snapshot after a motion detection event.
This happens because the cam ends the free memory and the snapshot requires 3 MB.
Please try to enable  swap file.
Or disable motion_detection_image leaving blank the field "Topic Suffix for jpeg image".

tachyon142, 2021-05-03:
That makes sense - I had noticed separately that the webpage snapshot feature reliably caused crashes. I've enabled the swap file and disabled AI Human Detection, and report that the cam has remained stable thus far. Thank you for your extremely quick response, and excellent suggestion.

Is the current path forward regarding use of AI Human Detection purely reliant on Yi to fix something in their FW? Any hope of cloudless use of their facial recognition features in the future, or is it better to use external solutions like [Frigate](https://github.com/blakeblackshear/frigate)?

closed 2021-05-03 by tachyon142

roleoroleo, 2021-05-04:
AFAIK this feature doesn't work and I think it will never work because Yi does not often release new firmware versions.
If you need it, it's better to use a different solution (or external solution).