roleoroleo/yi-hack-Allwinner #298: Support for YI Dome X FW 8.1.0.0A_2020012111401

fizzi, 2021-05-05:
New Firmware 8.1.0.0A_2020012111401 for new Yi Dome X Camera.

Any plan ?

roleoroleo, 2021-05-05:
Try this:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6427082/y30qa_0.2.6.tar.gz)

antodeilippo, 2021-05-06:
I try to flash the fw. 
Camera try to start (yellow fixed light), but after few seconds the led switch off. Now the camera cannot be reached via internal IP neither via mi home app

FrancescoTesta40, 2021-05-06:
Hi,
In my case seems that the camera loads the file but the led remains in yellow color and blink.
Any Idea
Many Thanks,
Francesco

roleoroleo, 2021-05-06:
Please, don't install this alpha.
I will post a new version asap.

roleoroleo, 2021-05-06:
Here it is:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6437005/y30qa_0.2.6.tar.gz)

fizzi, 2021-05-06:
+1 here !

roleoroleo, 2021-05-06:
@fizzi 
Thanks for your support and thanks for helping me with this model.

roleoroleo, 2021-05-06:
I will add it to the master branch.
Please check if something is wrong and send me a feedback.

FrancescoTesta40, 2021-05-07:
Hi,
Copied all the file extracted from the zip file into the root folder of the
sd card and start the camera.
The firmware seems to be done correctly (see that the file are renamed in
done) but when the camera reboot ask me for connection and I try 3 time to
provide it by the yi app without any success, error from the camera is
login failed invalid password (but the password is correct). If I open the
log in the sd card the wpa_supplicant.conf contains the correct SID and PWD.

Francesco

fizzi, 2021-05-07:
I didn't have this problem ... reset several times and changed several WIFI always worked fine.

FrancescoTesta40, 2021-05-07:
I'll try with another router soon

FrancescoTesta40, 2021-05-07:
I've done the test without any success. Try to update again the cam, no way
to solve the issue.
In the sd card I found this log_pwd_wrong.tar.gz file that contain some
data and cheched the wpa_supplicant.conf with my rooter data and are
correct.
Any Idea? .

antodeilippo, 2021-05-07:
After flashing y30qa_0.2.6.tar.gz I experienced exactly the same behaviour than @FrancescoTesta40 . Camera start but a wi fi password error was always raised and pairing procedure with the yi home app always failed. I've done several test with several WIFi but result is exactly the same.

roleoroleo, 2021-05-08:
Please try this:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6445425/y30qa_0.2.6.tar.gz)

roleoroleo, 2021-05-08:
Did you try other hacks before?

FrancescoTesta40, 2021-05-08:
Yes the One you post at the beginning of this issues

roleoroleo, 2021-05-08:
The same problem also happened to @fizzi.
I changed the hack procedure, please try this file:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6446267/y30qa_0.2.6.tar.gz)

FrancescoTesta40, 2021-05-08:
Now work fine.
beer as well from my side. 
Thanks

FrancescoTesta40, 2021-05-08:
I'm back because now I have another problem, the camera is accessible from http but it is continuously reboot (every 1-2 minute).

roleoroleo, 2021-05-08:
Try to enable swap file.
Reboot occurs because snapshot requires a lot of memory and the cam goes in oom.
Ptz page executes a sequence of snapshots.

roleoroleo, 2021-05-08:
You can control ptz with onvif, with web service or thriugh hass service.

antodeilippo, 2021-05-09:
I'm adding in my Hass-os rapberry through ONVIF integration and testing all the features. 
At a first glance, it seems that camera reboot each time a motion event was detected.
I will updat eth thread when test is completed

roleoroleo, 2021-05-09:
Sorry, but without swap space, the snapshot reboots the cam.
Enable it, please.

If you use ha, try this integration: https://github.com/roleoroleo/yi-hack_ha_integration

Berto93, 2021-05-10:
Hi @roleoroleo , i tried to flash cam my cam 8.1.0.0A_2020012111401 with your last relase "y30qa_0.2.6.tar.gz" but can't access web UI and can't re-add cam to YI app... The FW seems to be flashed because in my wifi list of router i found "yi-hack" near my cam IP address... what could i do? Thanks

roleoroleo, 2021-05-10:
Are you using port 8080?

Berto93, 2021-05-10:
Thank you, yes! i'm just now reached the UI resolved flashing the version of 4 days ago, is there any way to re-add the camera to the YI app? if i scan the QR code with procedure didn't work..

roleoroleo, 2021-05-11:
Is cloud enabled?

roleoroleo, 2021-05-11:
Try to add the cam as onvif.

Berto93, 2021-05-11:
is there any way to add it as onvif in motioneye? not pasting the rtsp link?

roleoroleo, 2021-05-11:
Sorry, I was wrong.
I don't know if motioneye supports onvif.

Berto93, 2021-05-12:
controlling PTZ through hass services in my case is very slow. Anyone have the same issue? thanks

GeorgiDimitrov78, 2021-05-15:
Hi,
I have flashed the last version provided here. The camera was visible in my router hosts list, the light was constantly blue, the GUI was not reachable on http://192....:8080, the camera was not in my YI Home app either.
30mins ago  I used the reset button on the bottom of the camera. Now the device is blinking in yellow and it is no more in my router hosts list.

Please for your help, the beer is in the fridge. :)

roleoroleo, 2021-05-15:
@GeorgiDimitrov78 
Probably you disabled cloud features and now the cam is not able to connect.
Apply the unhack procedure (see the wiki) and restart from the beginning.

GeorgiDimitrov78, 2021-05-15:
I already I hacked this morning and everything is fine with the original FW.
I am trying to read the important things here so not to ask stupid questions, but I am a bit lost.

What I need - to integrate my Dome X with Hass (over MQTT, without Internet preferably). 

Should I have paid cloud service enabled? 
The GUI is present only in the hacked version? How to reach the GUI? 
GUI is possible only with cloud enabled?

thanks for your support

roleoroleo, 2021-05-15:
> I already unhacked this morning and everything is fine with the original FW.

I published the unhack procedure for y30qa 2 hours ago.
Please check if you used this new file.

> Should I have paid cloud service enabled?

No.

> The GUI is present only in the hacked version? How to reach the GUI?

Yes. With your browser ad the ip address of your cam (port 8080).

> GUI is possible only with cloud enabled?

No. Cloud enable is necessary if you want to use the original app too.

GeorgiDimitrov78, 2021-05-15:
No, I used unhack_y25ga, and it helped.

When I apply y30qa_0.2.6 (tried two times) - there is no GUI available. The app doesn’t work either but you said that this is normal.

roleoroleo, 2021-05-15:
Please apply the proper unhack because unhack_y25ga contains wrong files for your cam.
And try again using the last beta published:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6446267/y30qa_0.2.6.tar.gz)

If you are freshly installing the hack to a stock cam, the GUI must work and the yi app must work.
If not, there is a problem.
Please try again as desribed above and let me know what happens.

GeorgiDimitrov78, 2021-05-15:
Just tried.... So!

1. fresh format of SD @4GB FAT32, applied unhack_y30qa - no problems, APP works fine

2. fresh format of SD @4GB FAT32, applied y30qa_0.2.6 - seems to be flashed, blue light ON, there is entry in router's DHCP list, no APP access ("Camera is offline, please reconnect(-3006)", no GUI access (different browsers). I have ping and telnet to the IP. The telnet prompts for login "xiaoyi login:"

RickyTR, 2021-05-17:
I have exactly the same behaviour as @GeorgiDimitrov78. Everything seems to go well but no web access and no APP access. The unhack works perfectly. Already tried with two different SDs (4 and 16 gb).

RickyTR, 2021-05-17:
Some more information... Tried to login with telnet and 'root' login and connected. As far as I can tell looking in /home folder it seems that yihacks folder is completely missing...

roleoroleo, 2021-05-17:
I will check it.

roleoroleo, 2021-05-17:
Please, try this:
[y30qa_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6494720/y30qa_0.2.6.tar.gz)

GeorgiDimitrov78, 2021-05-17:
Successfully hacked! 
I will do some tests and reply if there are issues.

Thanks!

RickyTR, 2021-05-17:
Now it works. 

Thanks a lot!

GeorgiDimitrov78, 2021-05-17:
I have this issued with spontaneous reboots even after I turned the swap file ON. Any suggestions?

roleoroleo, 2021-05-17:
Is it related to a specific command?

GeorgiDimitrov78, 2021-05-17:
Nope, no specific command (via APP) or motion detected (camera faces the wall). Integrated in HASS if it matters.

Berto93, 2021-05-17:
I can get motion through mqtt but can't get sound.. anyway i receive correctly sound in stream, anyone have the same issue? Thanks a lot.

roleoroleo, 2021-05-17:
> Nope, no specific command (via APP) or motion detected (camera faces the wall). Integrated in HASS if it matters.

If you are using a picture-glance, it sends a snapshot command every 10 secs.
Check if swap file is ok running `cat /proc/swaps`

> I can get motion through mqtt but can't get sound.. anyway i receive correctly sound in stream, anyone have the same issue? Thanks a lot.

Do you mean sound detection through mqtt? I noticed a lot of problem with sound detection.
Check if the event arrives to your phone.

RickyTR, 2021-05-17:
> I have this issue with spontaneous reboots even after I turned the swap file ON. Any suggestions?
I had the same problem after enabling integration with Openhab. I solved disabling the cloud connection. May be it’s related to memory?

GeorgiDimitrov78, 2021-05-17:
Issue with reboots solved - cat /proc/swaps showed no swaps. Fixed by reformatting the SD card.
I also have the issue that Berto93 mentions - the entities for Sound detection are always showing Clear state (if I understood him right).

Thanks again for the great job and support @roleoroleo !

roleoroleo, 2021-05-17:
@GeorgiDimitrov78 
Thank you for your donation!

roleoroleo, 2021-05-18:
In the next release I will publish an official image for this cam.

RickyTR, 2021-05-18:
First results after one day of testing.
MQTT works well but if I enable AI Human Detection I don't receive notification anymore from any of the toipcs.

roleoroleo, 2021-05-18:
It's a known bug:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/89

Berto93, 2021-05-18:
@RickyTR Can you detect also sound with mqtt?

roleoroleo, 2021-05-18:
Only if the original fw detects it.
I intercept and I foward it to the mqtt server.

Berto93, 2021-05-18:
If the yi app detects sound I should have it also on mqtt right?

roleoroleo, 2021-05-18:
Yes.

Berto93, 2021-05-25:
@roleoroleo Thanks so much for all your support, i've made some tests on 2 Yi Dome X cameras and work great and stable (Mqtt with sound and movement) with the FW version in this post: "https://github.com/roleoroleo/yi-hack-Allwinner/issues/298#issuecomment-833780375" Your beer is served 👍

[3 more replies]

closed 2024-05-09 by github-actions[bot]