roleoroleo/yi-hack-Allwinner #300: Support for YI Home 3 FW 8.3.0.0A_201912271707

torcuat, 2021-05-08:
Hi. I just bought a Yi Home Camera on amazon.es whose name is IFUSY32VBU and fw 8.3.0.0A_201912271707.

I tried the y25ga_0.2.6 release and I think I may have bricked it (stuck with yellow led on). Interesting to note, the files on the sd card end with *.done extension...

Any clues on how I could fix it and wether it could be supported ?

roleoroleo, 2021-05-08:
Try a previous version.

torcuat, 2021-05-09:
Thanks but I really think I bricked it and I hadn't made any backups.

Before it did something and stayed with the blue blinking led. Then, when I tried to reset it with the button on the back it got bricked and it does nothing but lighting the yellow led, no matter whats in the sd card.

roleoroleo, 2021-05-09:
I will check if I have a backup.

torcuat, 2021-05-09:
Thank you, I would really appreciate that, otherwise my cam is dead.

pablitofernandez, 2021-05-10:
Hello,

On Friday I bought exactly the same camera with the same type of serial number and same firmware. Same as @torcuat I tried the y25ga_0.2.6 and ended up bricking the camera after trying to reset it by hitting the reset button on the back of the camera.

The same Friday I bought exactly the same model of camera, which arrived yesterday. I did a backup correctly and instead of the last one I flashed this version [y25ga_0.1.9.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/5136276/y25ga_0.1.9.tar.gz) (I found it in another issue). Everything ok, RTSP, web server, etc. Then I tried to update to the latest version (y25ga_0.2.6) from the web server ui.The process finished but only telnet is active, no web server, no rtsp or anything else.

The next thing I tried was to install older firmwares, from 0.1.9 upwards. The last version that works correctly is 0.2.3. If I try to flash 0.2.4, 0.2.5 or 0.2.6 I end up with telnet as the only active service.
From telnet I have tried to do a ps and a top, and I don't see neither the httpd nor the ssh process running for example. The /home/yi-hack directory exists, but the /home/yi-hack/www directory does not exist.

I have tried with 3 different MicroSD (Samsung 64Gbs, Samsung 32Gbs and Sandisk 8Gbs) and with the same results.

I hope this information is helpful for debugging and that we manage to install the latest version on this camera model.

Thank you!

roleoroleo, 2021-05-10:
Ok, I will check it.

torcuat, 2021-05-10:
> Hello,
>
> On Friday I bought exactly the same camera with the same type of serial number and same firmware. Same as @torcuat I tried the y25ga_0.2.6 and ended up bricking the camera after trying to [...]

Hi, would you be so kind as to share the backup you made, so I try to unbrick mine?

pablitofernandez, 2021-05-10:
Sure thing, I can upload it in a few hours once I'm at home. @roleoroleo is there anything I have to hide of my backup with sensitive data?

roleoroleo, 2021-05-10:
Share only mtdblock3.
I don't remember exactly (because it depends on the model) but it shouldn't contain sensitive data.

roleoroleo, 2021-05-10:
> On Friday I bought exactly the same camera with the same type of serial number and same firmware. Same as @torcuat I tried the y25ga_0.2.6 and ended up bricking the camera after trying to reset it [...]

Maybe a problem when the update script decompresses the archive.
Probably the archive is too big for /tmp file system.
I need to check it.
From 0.2.3 to 0.2.4 I added 300KB.

> The next thing I tried was to install older firmwares, from 0.1.9 upwards. The last version that works correctly is 0.2.3. If I try to flash 0.2.4, 0.2.5 or 0.2.6 I end up with telnet as the only active service.

pablitofernandez, 2021-05-10:
@torcuat here you have the backup bin file: [mtdblock3.bin](https://file.io/8i1Zr4i8ZcqS)

roleoroleo, 2021-05-11:
@pablitofernandez 
Could you try this version?
[y25ga_0.2.6.tar.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/6459096/y25ga_0.2.6.tar.gz)

roleoroleo, 2021-05-11:
Probably a problem with the size of /tmp.
I added swap memory during setup phase.

To avoid restarts, enable swap file.

pablofernandez-deusto, 2021-05-12:
You are right again, with swap file I have no more restarts. Now the problem I have is I cant have audio enabled in the rtsp stream. I have tried all audio options but I cant see the audio stream being transmited. I don't know if maybe its better to open another issue.

Thanks!

roleoroleo, 2021-05-12:
Please, try it without user and password.

pablofernandez-deusto, 2021-05-12:
I have just tried with no user/password, Same results.

[image]

Also If I try to stream the audio only stream it does not connect to it.

pablitofernandez, 2021-05-12:
Just for testing purposes I have killed the process and launch it manually like I saw in another issue:

```
killall wd_rtsp.sh
killall rRTSPServer
RRTSP_RES=high RRTSP_AUDIO=aac RRTSP_USER=test RRTSP_PWD=test ./rRTSPServer
```

And now I see this in VLC:
[image]

So launching it manually seems to be working 🤷‍♂️

pablitofernandez, 2021-05-12:
Some more information, although I can see the codec of the audio transmitting in VLC I can't actually hear anything coming from the cam. There is no sound.

roleoroleo, 2021-05-12:
Please, post your config.

pablitofernandez, 2021-05-12:
Ok its quite erratic, now it seems to be working (with sound) if I put this command manually:

`RRTSP_RES=high RRTSP_AUDIO=aac RRTSP_USER=test RRTSP_PWD=test ./rRTSPServer`

This is my system.conf

```
HTTPD=yes
TELNETD=no
SSHD=yes
FTPD=no
BUSYBOX_FTPD=no
DISABLE_CLOUD=no
REC_WITHOUT_CLOUD=no
MQTT=no
RTSP=yes
RTSP_STREAM=high
RTSP_AUDIO=aac
SPEAKER_AUDIO=yes
ONVIF=yes
ONVIF_WSDD=yes
ONVIF_PROFILE=high
ONVIF_WM_SNAPSHOT=no
NTPD=yes
NTP_SERVER=pool.ntp.org
PROXYCHAINSNG=no
SWAP_FILE=yes
RTSP_PORT=554
ONVIF_PORT=80
HTTPD_PORT=8080
USERNAME=########
PASSWORD=########
TIMEZONE=CET-1CEST,M3.5.0,M10.5.0/3
FREE_SPACE=0
FTP_UPLOAD=no
FTP_HOST=
FTP_DIR=
FTP_DIR_TREE=no
FTP_USERNAME=
FTP_PASSWORD=
FTP_FILE_DELETE_AFTER_UPLOAD=yes
SSH_PASSWORD=############
CRONTAB=
```

roleoroleo, 2021-05-12:
Could you check the command line when the daemon starts (when audio doesn't work)?
Get the process id with ps and get the command line with:
`cat /proc/PID/cmdline`

pablitofernandez, 2021-05-12:
The output is just:

`rRTSPServer`

roleoroleo, 2021-05-12:
Please try the environment:
`cat /proc/PID/environment`

pablitofernandez, 2021-05-12:
There you go

`USER=rootRRTSP_MODEL=y25gaRRTSP_AUDIO=aacSHLVL=5LD_LIBRARY_PATH=/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib:/home/lib:/home/yi-hack/lib:/tmp/sd/yi-hack/libHOME=/OLDPWD=/partitions=boot@mtdblock1:rootfs@mtdblock2:home@mtdblock3:env@mtdblock4:mfg@mtdblock5:conf@mtdblock6:UDISK@mtdblock7earlyprintk=sunxi-uart,0x05000000boot_type=3cma=0TERM=vt102PATH=/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base:/home/base/tools:/home/yi-hack/bin:/home/yi-hack/sbin:/home/yi-hack/usr/bin:/home/yi-hack/usr/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbinRRTSP_RES=highuboot_message=2018.05-00002-g932b424-dirty(12/13/2019-18:21:53)RRTSP_PWD=*********SHELL=/bin/shRRTSP_USER=*******PWD=/home/appTZ=CET-1CEST,M3.5.0,M10.5.0/3RRTSP_PORT=554`

pablitofernandez, 2021-05-12:
BTW /proc/PID/environment is not found I had to type **/proc/PID/environ**

roleoroleo, 2021-05-12:
The environment is ok.
The problem happens only with aac or even with xlaw?

pablitofernandez, 2021-05-12:
I have tried with all audio modes. Same result.

roleoroleo, 2021-05-12:
Another try please.
If you kill rRTSPServer when it doesn't work and wait for it to restart (by watchdog), it works?

pablitofernandez, 2021-05-12:
> Another try please.
> If you kill rRTSPServer when it doesn't work and wait for it to restart (by watchdog), it works?

Yes, when the watchdog restart the process it's working correctly!

pablitofernandez, 2021-05-12:
BTW which audio codec is the best?

roleoroleo, 2021-05-12:
> BTW which audio codec is the best?

It depends.
Alaw and ulaw are light and require low cpu resouces. But some programs don't support it (eg. ha).
Aac requires high cpu resources.

pablitofernandez, 2021-05-12:
When you say AAC requires high CPU resources you mean YiCamera CPU resources or client CPU resources?
If its YICAM resources should it run better if I disable services like ONVIF, Yi Cloud, etc?

I asking because I would like to integrate the cam with HA, and if AAC is que only compatible codec ...

thanks!!!!!

roleoroleo, 2021-05-13:
I mean cam resources.
If you want to use ha, I advise you to use these custom components:
- https://github.com/roleoroleo/yi-hack_ha_integration
- https://github.com/AlexxIT/WebRTC

with alaw/ulaw audio.

roleoroleo, 2021-05-13:
> Yes, when the watchdog restart the process it's working correctly!

Please try to extend the sleep after rmm starts.
Edit /home/yi-hack/script/system.sh
and change sleep from 6 to 10
```
        LD_LIBRARY_PATH="/home/yi-hack/lib:/lib:/usr/lib:/home/lib:/home/qigan/lib:/home/app/locallib" ./rmm &
        sleep 10
        dd if=/tmp/audio_fifo of=/dev/null bs=1 count=8192
```
at lines 127 and 154.

pablitofernandez, 2021-05-13:
> Please try to extend the sleep after rmm starts.
> Edit /home/yi-hack/script/system.sh
> and change sleep from 6 to 10
>
> ``` [...]

With this little tweak is working correctly, tried with alaw and aac, both working when I restart the camera normally.

Thanks so much @roleoroleo !!

roleoroleo, 2021-05-13:
I will update the repo.

torcuat, 2021-05-24:
> 
> 
> @torcuat here you have the backup bin file: [mtdblock3.bin](https://file.io/8i1Zr4i8ZcqS)

Thank you very much, it worked!

meyomey, 2021-06-05:
Hi,
i have two bricked cameras here (base version 8.3.0.0A_201912271707, model suffix y25ga). After updating to 0.2.6, both no longer boot. I have a backup of another camera of the same type, but I don't know exactly where to enter the commands to copy it back? How is it possible to stop the boot at the U-Boot prompt?  This are my current boot messages:

Camera1:

[38]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[43]BOOT0 commit : 81c18ee
[46]board init start
[48]set pll start
[50]set pll end
[52][pmu]: bus read error
[54]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT `off.`
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[88]spinor id is: c8 40 18, read cmd: 03
[92]Succeed in reading toc file head.
[96]The size of toc is 6c000.
[170]Entry_name        = optee
[173]Entry_name        = u-boot
[178]Entry_name        = dtb
[181]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年                                                                                                                                                                              11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00002-g932b424-dirty (Dec 13 2019 - 18:21:53 +0800) Allwinner Tec                                                                                                                                                                             hnology

[00.249]CPU:   Allwinner Family
[00.252]Model: sun8iw19
I2C:   ready
[00.257]DRAM:  64 MiB
[00.259]Relocation Offset is: f9f6a000
[00.276]secure enable bit: 0
[00.278]pmu_axp2101_probe pmic_bus_read fail
[00.282]drivers/sunxi_power/pmu_axpnull.c pmu_axpnull_probe 26 axpdummy probe
[00.289]bmu_axp2101_probe pmic_bus_read fail
[00.293]BMU: AXPNULL
[00.296]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.302]gic: sec monitor mode
[00.304]flash init start
[00.307]workmode = 0,storage type = 3
SF: Detected gd25q128 with page size 256 Bytes, erase size 64 KiB, total 16 MiB
[00.319]sunxi flash init ok
[00.322]Loading Environment from SUNXI_FLASH... OK
[00.342]sprite_led_gpio start
[00.347]update dts
root_partition is rootfs
set root to /dev/mtdblock2
[00.355]update part info
[00.357]update bootcmd
[00.359]No ethernet found.
[00.361]get mem for descripter OK !

[00.587]Card did not respond to voltage select!
card0 init failed
No sd-card insert,will normal boot
Hit any key to stop autoboot:  0
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2080440 Bytes = 2 MiB
   Load Address: 40008000
[... 188 lines cut ...]
step 3 4
step 4 4
step 5 5
step 6 5
/dev/mmcblk0p1: 15 files, 886/120704 clusters
)

checkdisk.c(do_fat_check-140)[00:00:05.030]:Mount sd ok.

checkdisk.c(check_partion_vfat-156)[00:00:05.030]:TF Card check finished.

checkdisk, main() ret_status(10000)
checkdisk, main() ret_status(10000),g_sysconf_ptr->tf_status.stat(10000)
---update file(home_y20gam) Not exist---
--------------------------insmod sound--------------------------
[    3.500130] --->sunxi_daudio_dev_probe<--- strat.
[    3.507000] --->sunxi_daudio_dev_probe<--- end.
--------------------------insmod sensor--------------------------
[    3.953972] 8189fs: section 3 reloc 9361 sym 'skb_pull': relocation 10 out of range (0xbf875fb0 -> 0xc01a6d2b)
[    3.982759] 8189fs: section 3 reloc 9361 sym 'skb_pull': relocation 10 out of range (0xbf951fb0 -> 0xc01a6d2b)
insmod: can't insert '/home/base/wifi/8189fs.ko': invalid module format
ifconfig: SIOCGIFFLAGS: No such device
/home/app/init.sh: line 147: /home/yi-hack/script/system.sh: not found
hue, start_with_reset = 1 @get_config()->g_sysconf.ssid == 0
>>>>>>>>>>>>>#########is_sd_exit chg to: 10000<<<<<<<<<<<<<<<<<<<

roleoroleo, 2021-06-07:
Press enter after the power on.
The cam should stop.
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam

meyomey, 2021-06-07:
Thanks! it works, but you have to wait for the very exact moment at "Hit any key to stop autoboot"
Now i have written back the file mtdblock3.bin.  Now i get this with inserted SD card:

root@xiaoyi:/#
checkdisk.c(get_tfcard_size_v2-105)[00:00:05.346]:TF card total size : 30436 MB
checkdisk.c(get_fs_type-251)[00:00:05.351]:FS TYPE : VFAT
checkdisk.c(get_cluster_size-313)[00:00:05.369]:cluster_size = 32768 Bytes
checkdisk.c(check_partion_vfat-152)[00:00:05.369]:TF Card  checking ....
checkdisk.c(do_fat_check-117)[00:00:05.370]:Can fat check
checkdisk.c(do_fat_check-121)[00:00:05.737]:retBuf(fsck.fat 3.0.26 (2014-03-07)
step 0 5
0x41: Dirty bit is set. Fs was not properly unmounted and some data may be corrupt.
step 1 5
step 2 5
step 3 5
step 4 5
step 5 5
step 6 5
/dev/mmcblk0p1: 3 files, 4/973584 clusters
)
checkdisk.c(do_fat_check-140)[00:00:05.742]:Mount sd ok.
checkdisk.c(check_partion_vfat-156)[00:00:05.743]:TF Card check finished.
checkdisk, main() ret_status(10000)
checkdisk, main() ret_status(10000),g_sysconf_ptr->tf_status.stat(10000)
---update file(home_y25gam) Not exist---
--------------------------insmod sound--------------------------
[    4.019507] --->sunxi_daudio_dev_probe<--- strat.
[    4.029880] --->sunxi_daudio_dev_probe<--- end.
--------------------------insmod sensor--------------------------

without SD:

root@xiaoyi:/# cat: can't open '/sys/block/mmcblk0/device/cid': No such file or directory
checkdisk, main() ret_status(10012)
checkdisk, main() ret_status(10012),g_sysconf_ptr->tf_status.stat(10012)
umount: can't unmount /tmp/sd: Invalid argument
mount: mounting /dev/mmcblk0p1 on /tmp/sd failed: No such file or directory
---update file(home_y25gam) Not exist---
--------------------------insmod sound--------------------------
[    3.576595] --->sunxi_daudio_dev_probe<--- strat.
[    3.586831] --->sunxi_daudio_dev_probe<--- end.
--------------------------insmod sensor--------------------------

Any idea what can be wrong now?
Thanks meyomey

roleoroleo, 2021-06-07:
What's the problem?
Does the cam start?

meyomey, 2021-06-07:
Hi, the camera starts, but only the yellow LED lights up continuously.
The camera cannot be connected to the app, no "waiting for connect" The reset button does not work.

roleoroleo, 2021-06-08:
Post the whole log starting from power on, if you have serial cable connected.

roleoroleo, 2021-06-08:
There are no problem in the log.
Please run the following commands:
ps
lsmod
mount
ls -laR /

meyomey, 2021-06-08:
Thats the related log: 
[putty.log](https://github.com/roleoroleo/yi-hack-Allwinner/files/6615683/putty.log)

roleoroleo, 2021-07-02:
The output of the ps command is not ok.
Processes don't start.
It's difficult to understand the problem. Probably something wrong in the startup procedure.
Check if /home/app/init.sh is ok and try to run it manually.

Ppatou, 2021-07-08:
Hello,
 I just bought a Yi Home Camera  whose model is 9FUSY32IX2JA4R200611 and fw 8.3.0.0A_201912271707 (same version as torcuat but not same model). 
I tried the y25ga_0.2.6 release and wait for more than 1 hour but nothing append only yellow led constant on.
Reading this post, I tried the y25ga_0.1.9.tar.gz from Pablito but no succes same behavior.
I change 2 times sd card (16 Gb and 32 Gb) and use the original power supply.
Then I tried the proposed version from roleoroleo y25ga_0.2.6.tar.gz from 11 may => idem no boot yellow led stays on for hours.
Is there something I missed ? 
I put the file and the 2 directories on a blank format sd card as in the instructions.
[image]
Question : During the flash process I suppose the yellow led must blink ... I never saw that.
On an older Yi cam I could, without problem, flash it with the mstart hack.
How can I help searching the problem ?
Thanks in advance for your help
Patou

roleoroleo, 2021-07-09:
Don't try to hack the cam with unsupported versions.
Or with a version created for a different model.
You could brick the cam.

Your version is y25ga. If it doesn't work try to change or reformat the sd.
During the hack the cam doesn't blink, only MStar blinks.

Ppatou, 2021-07-09:
Hello Roleo,
Thanks for your answer.
Today, I took a second brand new camera 9 FUSY 32 VPIRFMS 200319 date 2020 06 with  sn 9Y25GAEUW06015044.
- took a 16 Gb new formatted sd card empty
- make the pairing with the Yi app and got the pairing and wifi connected, saw the video on smartphone and pc
- format this same sd card again and load the file and 2 directories of y25ga_0.2.6.tar.gz (from 11 may in this post) in the root of the sd card
-  put this sd in the camera, power on with the original power supply from the camera, green led comes on and after 10 sec the relay clicks, and then nothing changed for more than 1 hour. I cannot see the cam in the Yi app.
-  disconnect power supply, reformat sd card (empty card) and place it again the camera, the camera works again on the Yi app.
- Question : how can I recognize that the new firmware has been flashed ?

roleoroleo, 2021-07-13:
Is the Factory folder rename to Factory.done?
Are you able to connect to the website on port 8080?

Ppatou, 2021-07-13:
Hello 
Thnaks for your reaction
No : the factory folder has still the name factory and no access with "myip":8080

roleoroleo, 2021-07-13:
If the hack didn't run:

1. The cam is not allwinner based.
2. The sd card is not compatible.

If you want to solve definitively this problem you need to connect a serial cable.

Ppatou, 2021-07-13:
What brand or spec of sd card do you recommend ?
Is max size of 32 Gb OK ? it becomes difficult to find smaller cards !!!

roleoroleo, 2021-07-14:
I can't recommend a card.
But small is better.
I have a lot of small cards (1 or 2GB) from old smartphones that work great.

Ppatou, 2021-07-14:
Hello 
Thanks for all your answers ... but still no succes.
I tried 2 other smaller size sd cards 2 Gb and 1 Gb. 
No change in the directory names, no visible Ip to see on the network
So, actually, the  9 FUSY 32 VPIRFMS camera even with the same firmware (fw 8.3.0.0A_201912271707) as IFUSY32VBU is not working. with this hack.
How can I find out the real camera hardware ?
Fortunately I could always come back to the original situation working with the factory app.
Have a nice day
Patou

roleoroleo, 2021-07-15:
Maybe the sd card slot is not working.

> If you want to solve definitively this problem you need to connect a serial cable.

Ppatou, 2021-08-04:
Hello
Return from vacation and format the sd card on the camera and not on the pc and it works
Thanks for your support

bustamantemarlon, 2021-08-11:
> @torcuat here you have the backup bin file: [mtdblock3.bin](https://file.io/8i1Zr4i8ZcqS)

Hi, anyway you can reupload the backup file? I think I bricked my cam. Thanks.

Fluke1975, 2022-05-18:
> @torcuat here you have the backup bin file: [mtdblock3.bin](https://file.io/8i1Zr4i8ZcqS)

Please can you share this backup file to me too? Many thanks

torcuat, 2022-05-19:
> > @torcuat here you have the backup bin file: [mtdblock3.bin](https://file.io/8i1Zr4i8ZcqS)
> 
> Please can you share this backup file to me too? Many thanks

Here you go [mtdblock3.bin](https://file.io/RhdHQH7lCRyG)

[3 more replies]

closed 2024-05-09 by github-actions[bot]