roleoroleo/yi-hack-Allwinner #303: Advertised MQTT settings switches not working as expected in Home Assistant

krisnoble, 2021-05-20:
Hi, I'm having some trouble with the switches exposed by the MQTT camera settings advertisement.

The status LED switch works as expected in Home Assistant, but all others revert back to their previous state after a few seconds and have no effect. That state does accurately reflect the setting on the web portal, though. I have other MQTT stuff like motion and sound detection messages working fine.

From watching the MQTT topics I can see that when turning off, the LED switch sends `yicam/camera_setting/LED/set no` and the status switch sends `yicam/camera_setting/SWITCH_ON/set no`, but only the LED one has any effect on the camera settings.

I'm not seeing anything in the HA logs. I'm using the latest version of the firmware, my camera is a Yi 1080p 9FUS. Sorry if I missed anything, new to all this so learning as I go. Thanks!

krisnoble, 2021-05-20:
Just adding some additional version info:

Attribute | Value
-- | --
Firmware Version | 0.2.6
Base Version | 8.2.0.0A_202012041528
Model Suffix | y20ga
Serial Number | 9FUS

roleoroleo, 2021-05-21:
There is a sync problem between web interface switches and mqtt switched and I can't fix it.
To check if the switch works you have to compare it with the app.
Let me explain with an example:
- Enable the cloud and connect to the cam with the original yi app.
- Switch on/off the switch.
- Check if the corresponding switch in the app changes.

krisnoble, 2021-05-21:
Looks like the camera on/off does sync from app to web but the others don't.

 I found the `ipc_cmd -t` in the code, if I SSH into the camera and run them, turning off/on works as expected, as does the web service at `http://IP:8080/cgi-bin/camera_settings.sh?switch_on=`

So it seems like the `ipc_cmd` should work, and the MQTT is being published correctly, so that suggests the problem is somewhere between receiving the MQTT and executing the command?

Also: not sure if related or not but last night I left the camera without power and today it was doing the "waiting to connect" loop which I fixed with the wiki page method. Now, when I go to test the switches in the app, the power on/off works as expected but the status LED one will turn the light off momentarily before it comes back on whereas it worked as expected yesterday. The HA switch works as expected. Turning the LED on in the app when it's off works as expected.

closed 2024-05-09 by github-actions[bot]