roleoroleo/yi-hack-Allwinner #304: Cam stops working after ??hours, after reboot all fine

mdgarf, 2021-05-20:
Hi, 
Cam should run 24/7, have script to get image (every 20min) and send it out. 
Have 2 cams, both repeatedly stop responding. At different times(10 and more hours), I think. Have to power off, after boot up all is OK. 

yi-hack-Allwinner 0.2.6
y25ga_0.2.6.tgz
9FUSY@8.3.0.0

Any idea how to fix it?

mdgarf, 2021-05-21:
1. cam 8:32h
2. cam 5:10h 
other cams yi, edimax, hik work fine.

roleoroleo, 2021-05-21:
Is the swap file enabled?
The cam runs with little free memory and the snapshot is very expensive.

mdgarf, 2021-05-21:
no
just switched on, will let you know.
thanks for a prompt reply

mdgarf, 2021-05-22:
cams are working, I believe you solved it.
thanks

AJStubbsy, 2021-07-25:
I'm experiencing this problem where the camera keeps turning off. Seems to happen at dusk when the camera is constantly adjusting the light settings. Can you tell me where I can find 'swap file' settings please? I haven't seen this on the web page or camera app.

catatonico2, 2021-12-03:
Around here the same, after a while the camera stops working, firmware 0.27 - BFUS

roleoroleo, 2021-12-03:
What features stop working?

catatonico2, 2021-12-03:
Stop being online, both for local and for app yi, restart returns to online. In the Yi home app it indicates that I have a pending update firmware but it does not take place.

roleoroleo, 2021-12-03:
Maybe a memory leak.
Are you able to connect with ssh when the cam stops working?
If yes, please check the memory and cpu usage.
Run top, df and free commands.

catatonico2, 2021-12-03:
ok right now it is online, as soon as I lose connection I try.

catatonico2, 2021-12-05:
I have lost the connection again, it does not ping the ip, the yi app does not respond or ip: 8080.

roleoroleo, 2021-12-07:
Is swap enabled?

catatonico2, 2021-12-07:
yes,

Staini, 2021-12-12:
I can also confirm this.
I could observe when ipc_cmd is run twice at the same time it occurs more.

The camera is then no longer reachable via the network.

roleoroleo, 2021-12-12:
I don't know how to find the problem...

Staini, 2021-12-12:
Methode 1:
1. open the ptz page.
2. make movements in front of the camera.
3. send in the same time with ssh multiple ipc_cmd commands (e.g. record and stop).

Methode 2:
1. disable the wifi from your router and enable it again.
2. wait 30 seconds.
3. repeat these steps several times.

closed 2024-05-09 by github-actions[bot]