roleoroleo/yi-hack-Allwinner #311: Set username/password and can no longer access web UI

RyanWor, 2021-06-25:
Hi. Thanks for this firmware. This is amazing. I have one issue with a few cameras I setup (9FUS). I set a username/password as well as a root password. Once they rebooted I cannot login to the web interface with that username and password. I can however login via SSH. Is there a way to reconfigure the standard user's credentials from SSH and is there any limitations to the length and characters allowed in the user account for the web interface? Thanks!

roleoroleo, 2021-07-02:
I don't remember exactly but another user found a limit in the size.
You can change the password from SSH editing the file /home/yi-hack/etc/system.conf

RyanWor, 2021-07-02:
Thanks I figured it out eventually. It wasn't actually the length that was a problem for me it was an issue with some symbols. Switched to alphanumeric and it worked fine. Appreciate all your work on this firmware @roleoroleo!

closed 2021-07-02 by RyanWor

roleoroleo, 2021-07-06:
If you use only the suggested special chars, it should work.