roleoroleo/yi-hack-Allwinner #318: restting options should not be a one click thing

pbanj, 2021-07-31:
this is the second time i have reset my camera because there is no confirmation before it does it. at the min at least move it away from reboot

vin-w, 2021-08-16:
I accidentally clicked twice too. So now I always do backup of my config so that I could restore easier.

I hope to have a popup to confirm feature too.

```
<script LANGUAGE="JavaScript">
<!--
function confirmSubmit(text)
{
var agree=confirm("Are you sure to " + text + "?");
if (agree)
 return true ;
else
 return false ;
}
// -->
</script>
 <input class="button-primary" type="button" id="button-reset" value="Reset Camera"  onClick="return confirmSubmit('reset')"/>
```

roleoroleo, 2021-08-17:
:)
I will add it.

pbanj, 2021-08-18:
Thank you so much. Also I finally used the built in wifi settings you added. They worked great 👍

pbanj, 2021-09-08:
@roleoroleo i just updated and the changelog says `Add confirmation to reboot/reset buttons` but reboot doesnt ask. not sure if you made a typo in the change log or if it is supposed to ask. im scared to test the reset option lol

pbanj, 2021-09-08:
nvm it is "working". it's firefox blocking the popup unless you turn off firefoxes enhanced protection. but with the popup blocked the camera still reboots

closed 2024-05-08 by github-actions[bot]