roleoroleo/yi-hack-Allwinner #325: [Question] Know if the camera is currently recording

ivanfor, 2021-09-09:
Hi,
Is it possible to know, from the outside (REST/MQTT), if the camera is currently recording? In other words, if the file tmp.mp4.tmp exists. Alternatively, is it possible to know (from the outside, as well) when the recording ends?
I know I can create a new cgi-bin script, but maybe there's a cleaner way.
Thanks in advance.

roleoroleo, 2021-09-11:
I think cgi is the best way if you want this information on request.
If you want to receive a push message, mqtt is better.
For example, a daemon that, every 1 sec, reads if the file exists and sends a message when the status changes.

ivanfor, 2021-09-13:
Thanks. I didn't notice that mosquitto_pub binary is in place.
A daemon script will to the trick.

closed 2021-09-13 by ivanfor