roleoroleo/yi-hack-Allwinner #329: Video not recorded on SD card

Vormillion, 2021-11-03:
Hi,

I just flashed my camera (BFUS) but I can't figure out why I don't have any recorded files is /tmp/sd/record.

df output:
root@yi-724a:/tmp/sd/record# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     10956       820  93% /home
tmpfs                    32768        84     32684   0% /tmp
/dev/mmcblk0p1           32704        32     32672   0% /tmp/sd

I have following options set:
1) Disable Cloud --> true
2) Recording without cloud --> true
3) Swap --> off (when enabled, then webUI is showing 0% left on SD card)
4) Save video when a motion is detected --> false (I want so far recording all the time, anyway I tested with option enabled and I was walking around camera but nothing in /tmp/sd/record)

Am I'm missing something?

Vormillion, 2021-11-03:
I just tried create 1 GB file using dd on /tmp/sd but I'm getting no space left on device. Windows is seeing 32 GB on this card.

roleoroleo, 2021-11-04:
The only option that affects this feature is "Recording without cloud".
If it's enabled, the cam is configured properly.
About SD card, try to reformat it. The /tmp/sd/ partition must be accessible and writeable.

Vormillion, 2021-11-05:
OK, that was an issue with SD filesystem. After heavy fight with it, I was able to reformat card and now camera sees 32 GB.

closed 2021-11-05 by Vormillion