roleoroleo/yi-hack-Allwinner #330: This firmware is working also with Yi 1080p Home IFUS

huntz, 2021-11-19:
Hi roleoroleo,
I have just flashed a Yi 1080p Home IFUS, with 8.3.0.0A_201912271707 with the y25ga Model Suffix.

I think the camera model **Yi 1080p Home IFUS** should be added to the **Supported cameras** section in the main page, to help other users.

My only hint with this camera is to enable the swap file on SD card to improve stability (without the swap enabled, the camera is experiencing frequent reboots).

Thank you for your work.

roleoroleo, 2021-11-20:
Thank you for your feedback.

essys, 2021-11-27:
I also have the IFUS with serial number starting with IY25GA and would love to you use this hack.
I cannot directly provide the FW version, as I won't install Yi app on my phone.

Nonetheless I created a backup of current FW:
[backup.zip](https://github@roleoroleo /roleoroleo/yi-hack-Allwinner/files/7612497/backup.zip)

Due to this thread I tried with "Yi 1080p Home 9FUS | 8.3.0* | y25ga" and after installing the FW added my Wifi credentials using this procedure https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials.
Looks like I failed or my FW is currently not supported.

@roleoroleo are you able to extract the FW version or provide a lead where to look?
I would love to help out and provide all necessary details to get this hack working.

roleoroleo, 2021-11-28:
Yes.
It's a y25ga 8.3.0.0A_201912271707.
Try again.

essys, 2021-11-29:
Now I can confirm, the FW hack also works on y25ga 8.3.0.0A_201912271707.
It took me a few attempts until I found a supported 16GB SD-Card.

[image]

leo1783, 2022-02-09:
Hello,
I also confirm that the firmware works in a Yi 1080p Home IFUS, with 8.3.0.0A_201912271707 with the y25ga Model Suffix.

I have some issues in the configuration page, as when I change some values (for example: user name, user password, disable TELNET, etc.) when I press "Save Configuration" and I get the message "Saved", then I reboot the camera and the new values are gone.
Because of that I am not able to set user/pass for the rstp stream, something that it is really important for me.
@huntz do you have the same issues?

Thank you for your work!

roleoroleo, 2022-02-10:
Please, try another browser.
This problem is new for me.

leo1783, 2022-02-15:
Hello @roleoroleo,
After some investigation, this is what I found:
[image]

/home partition is full, could that prevent the webpage to save the changes properly?

I have tried to use the console(ssh) to change the file /home/yi-hack/etc/system.conf with vi text editor but I am only able to do  it once or twice, after that the /home partition gets 100% and I am not able to save the changes anymore.

[image]

Also, I don't know if related, sometimes the /etc/shadow partition is not mounted, so the root access is granted without the password stored inside /home/yi-hack/etc/system.conf.

Thank you for the support!
Could I delete some files to get some space? Expand the /home partition?

roleoroleo, 2022-02-15:
You can't expand home partition.
Check if there are"core" files, you can delete it.
Otherwise, run the unbrick procedure and reconfigure the cam from scratch.

Pmachat, 2022-03-01:
Hello,
I had the same problem with the same camera. I temporarily deleted the file /home/yi-hack/bin/onvif_srvd . After reboot, I was able to make the configuration changes to the camera (use ftp upload with login+password)

closed 2024-05-08 by github-actions[bot]