roleoroleo/yi-hack-Allwinner #338: Confirmed unbrick of y20ga using the software-only Allwinner-V2 method.

74ls04, 2022-01-16:
Just wanted to let you know that I was able to unbrick a y20ga camera by using [the method outlined in this ticket](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305) for the other camera models so we can add that procedure to this firmware as well. It was perpetually stuck at the yellow light no matter what I did so it was definitely bricked.

I had another y20ga camera so I used the backup script on that one then copied over the home partition file (mtdblock3.bin) from the backup to my Ubuntu server and followed the above procedure. This is the command I ran:

`mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-home" -d ./mtdblock3.bin home_y20ga`

I then put the **home_y20ga** file on an SD card and inserted it into the camera -- it rebooted after about 30 seconds and I had access to it with the hack still in place. The camera had the hack config of my other camera (since I used it to restore) but I changed everything back using a config backup I had and all was good to go!

The camera got bricked when I tried removing the hack for a firmware update using the unhack package from the Wiki -- not sure what went wrong but it just silently failed and bricked the camera.

roleoroleo, 2022-01-17:
Thank you for your contribution.
Very useful.
I will add a wiki page with this procedure.

roleoroleo, 2022-01-17:
Done:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam

74ls04, 2022-01-17:
Perfect! When you get a chance could you add a suggestion to backup the original firmware first at the beginning of installation to the README? I didn't know that was an option until I started looking for how to unbrick the cam -- thankfully I had a second one to use for recovery. Thanks!

closed 2022-01-17 by 74ls04