roleoroleo/yi-hack-Allwinner #339: Problem after update to 0.2.8

RickyTR, 2022-01-27:
I started the update from the web interface. At first it seemed that everything went well, the cam restarted after sometimes and got the ip. But web server and ssh server never starts and I can access the cam only via telnet. RTSP server also is not starting.

RickyTR, 2022-01-27:
UPDATE: I started the same version with a new SD. The process completed successfully, ssh server is working, RTSP server is working, httpd server still not working.

roleoroleo, 2022-01-27:
Please, try a factory default from the web interface.
And reconfigure the cam.

RickyTR, 2022-01-28:
I cannot access the web interface with 0.2.8 so I cannot try the factory default.

Kaplas80, 2022-01-28:
I've had a similar problem with my IFUS (y25ga) camera. `/home` ran out of space and files in `/home/yi-hack/www` weren't copied during update, so the web interface wasn't available after reboot.

I've fixed it deleting the new rtsp server executable (`/home/yi-hack/bin/rtsp_server_yi`) and copying the `www` directory from a 9FUS camera that upgraded without problems.

roleoroleo, 2022-01-28:
Ok, the problem is the out of space of the home partition.
Maybe y25ga runs with very little free space.

I have to check if  it's necessary to remove something.

roleoroleo, 2022-01-28:
My y20ga, after the update to 0.2.8, is working with 1,4 MB of free space.
```
root@yi-hack-allwinner:~#df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     10372      1404  88% /home
tmpfs                    32768       128     32640   0% /tmp
/dev/mmcblk0p1          956920    436844    520076  46% /tmp/sd
```

The free space should be enough.
Check if there is some "core" file and delete it.

Kaplas80, 2022-01-28:
Free space in 9FUS after update to 0.2.8 (updated ok)
```
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     11188       588  95% /home
tmpfs                    32768        64     32704   0% /tmp
/dev/mmcblk0p1        31149056  29777280   1371776  96% /tmp/sd
```

Free space in IFUS after update to 0.2.8 (after removing `rtsp_server_yi`)
```
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     11560       216  98% /home
tmpfs                    32768        60     32708   0% /tmp
/dev/mmcblk0p1        31242240    209856  31032384   1% /tmp/sd
```

The only "core" file in both cases is `/home/yi-hack/qigan/ko/videobuf2-core.ko` 

It seems that both cameras have the same total space in mtdblock3, but I don't know why IFUS has more space used than 9FUS.

roleoroleo, 2022-01-29:
The jffs2 file system doesn't work well when the free space is not enough.
For example, if you delete a file sometimes it happens that you can't use all the space freed up.
If this is your case, the unbrick procedure is your solution.

k-vq, 2022-02-06:
> I've had a similar problem with my IFUS (y25ga) camera. `/home` ran out of space and files in `/home/yi-hack/www` weren't copied during update, so the web interface wasn't available after reboot. [...]

Could someone share the www directory? I am having the same problem of accessing the web interface after upgrading to 0.2.8

bustamantemarlon, 2022-02-27:
I had the same issue with losing access to the web interface after doing an online upgrade to 0.2.8 on my IFUS (y25ga) cam. I used the unbrick method to go back to stock and then installed 0.2.8 again. 
I checked the space and I only had 44 blocks available after a clean install.
````
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     11732        44 100% /home
tmpfs                    32768        44     32724   0% /tmp
/dev/mmcblk0p1         7767824     70904   7696920   1% /tmp/sd
````
I wasn't able to save any configuration charges.
I did @Kaplas80 said and deleted `/home/yi-hack/bin/rtsp_server_yi`. This gave me 220 blocks free and I was able to save my configuration files.
````Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776     11556       220  98% /home
tmpfs                    32768        44     32724   0% /tmp
/dev/mmcblk0p1         7767824     70904   7696920   1% /tmp/sd
````
I hope this helps and provides more information about this cam version. Also here's my `www` directory, @k-vq. I hope it helps.
[www.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/8149557/www.zip)

k-vq, 2022-02-28:
>Also here's my `www` directory, @k-vq. I hope it helps. [www.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/8149557/www.zip)

Thank you very much @bustamantemarlon

closed 2024-05-08 by github-actions[bot]