roleoroleo/yi-hack-Allwinner #340: motion_stop missing body (after update to 0.2.8)

iupsilon, 2022-02-11:
Hi,
I have several y20ga cams.
After updating to firmware 0.2.8 I saw that mqtt motion_stop message no longer has a body.
I saw from previous mqtt service log that motion_stop message had a body like this:

motion_stop
{
"start":"1970-01-01T01:00:00+0100",
"end":"2022-02-04T08:17:16+0100",
"files":[ "2022Y02M04D08H/17M00S60.mp4", "2022Y02M04D08H/16M14S46.mp4" ]
}

Now I only receive:
motion_stop

without any body.

I confirm that the webcam still records the motion files in the usual folder (/tmp/sd/record) as it did before the upgrade.
I have developed a custom automation triggered by mqtt motion_stop message in order to process recorded files but it no longer works.

Is there something I can check in order to figure out what the problem is?

Many thanks for your great project.

Cheers,
Yari

iupsilon, 2022-02-12:
I found out what was wrong with configuration.

AI Human Detection is enabled.
Once disabled, the mqtt message 'motion_files' now works again as before.

I'm pretty sure this setting was already enabled before upgrade.

Many thanks.

rmaltais, 2022-02-15:
Same problem here with mine. No message detect after anable AI HUMAN DETECTION. 0.2.8

roleoroleo, 2022-02-16:
> Same problem here with mine. No message detect after anable AI HUMAN DETECTION. 0.2.8

This is a know issue.
There are a lot of issues in the yi-hack-Allwinner-v2 repo.
Check this for example:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/333

rmaltais, 2022-02-19:
Hello,

After many attempts at different configurations, I did not find a way to make a good detection of human movement. I always have false detections or not at all.

I removed the hack at the moment since we don't really have a solution that works. Thank you for your development because it is very appreciated to see that you are still doing excellent work on the other features.

Thanks

closed 2024-05-07 by github-actions[bot]