roleoroleo/yi-hack-Allwinner #345: RTSP Audio reverting to DIsabled after save - reboot

amethley, 2022-03-21:
Setting RTSP audio to any value, saving and rebooting, results in RTSP audio setting reverting to "Disabled".

yi-hack-Allwinner
Firmware 0.2.9
Camera Yi 1080p Home BFUS

roleoroleo, 2022-03-21:
When you save, does the label change to "Saved"?

amethley, 2022-03-21:
It does.  Other settings save and persist after reboot fine.  I can confirm there is no audio stream from the camera.  Previously working fine before updating to 0.2.9 (from 0.2.3)

roleoroleo, 2022-03-21:
Maybe they are 2 errors of the same problem.
Post the content of the file /home/yi-hack/etc/system.conf

roleoroleo, 2022-03-21:
The file is ok.
Did you try to clean the cache of the browser (CTRL-F5)?

amethley, 2022-03-21:
Thanks, that has updated the UI.  
However, there is no audio stream from the camera, just the H264 video stream.  I also cannot access the audio-only stream URL.   Is there a setting I've overlooked?

roleoroleo, 2022-03-21:
The conf is ok.
What about "Codec information" in VLC?
Does audio stream appear?
[image]

amethley, 2022-03-21:
No, I see only Stream 0. 

[image]

Note: I have a second camera also updated to latest version today (y21ga on 0.2.1 yi-hack-Allwinner-v2). which is sending Stream 0 and 1.

[image]

roleoroleo, 2022-03-21:
My y20ga works properly.
Maybe the update procedure doesn't complete properly.
Try to create again the sdcard overwriting all files.

amethley, 2022-03-21:
Unfortunately, the camera is hidden in a bird nesting box so is not accessible.

Thanks for your help anyway.

amethley, 2022-03-21:
I have audio!

[image]

cyraxnet, 2022-03-27:
Any chance of new build with this fix, and #347? I can't get to a Linux machine right now to build latest source.

roleoroleo, 2022-03-28:
Yes, but I need a few days.

closed 2024-05-07 by github-actions[bot]