roleoroleo/yi-hack-Allwinner #347: Upload no longer works

Pmachat, 2022-03-23:
Hello
The upload no longer works with my iFUS (y25ga) camera. . I don't know if this is due to the upgrade to version 0.29.
Can you tell me if you have the same problem? 
Can you tell me how to downgrade to 0.28
Best regards 

NOTE: I have another model of YI camera for which the upload works

roleoroleo, 2022-03-23:
Do you mean ftp upload?

Pmachat, 2022-03-23:
Yes, ftp upload

roleoroleo, 2022-03-23:
There is a recent commit.
Try if it fix your problem.
https://github.com/roleoroleo/yi-hack-Allwinner/commit/65d208820ed5b132f9902d5f0de601d8a67cac0f

Pmachat, 2022-03-23:
to test, should I replace the 2 files on the SD card in the directory /home/hi-jack/scripts ?

roleoroleo, 2022-03-23:
Yes.
Use ftp.

Pmachat, 2022-03-24:
it works now after replacing the 2 files by FTP
Thank you

closed 2022-03-24 by Pmachat