roleoroleo/yi-hack-Allwinner #350: Anyone recording streams to NVR or NAS?

cyraxnet, 2022-03-30:
I record the RTSP streams from the yi camera to an NVR.  It's always been temperamental and will record for a period, but then the NVR stops receiving the stream, and thus nothing gets recorded.  Same for viewing it live via the NVR app.

However, when the NVR has stopped recording from yi-hack, almost always I can view the stream from VLC, so it may be a compatibility issue. The NVR is ONVIF and RTSP compatible.  I've tried all versions to date, and on two cameras. Also tried the alternative RTSP in the latest releases.

So does anyone record from yi cameras to an NVR or NAS, and have any issues? Is there anything special you did?

cyraxnet, 2022-04-03:
I tried testing the latest hack 0.29 with [Onvif Device Manager](https://sourceforge.net/projects/onvifdm/). The profiles appear but live stream does not work. I tried the same with allwinner-2 (v9 stock firmware camera) and it also fails.

Googling possible issues hints towards dropped packets, the live555 implementation, or some network issue on my side. Likely, the onvif implementation from onvif_srvd is also not compliant with the standard, hence why my NVR has issues.  Trying a different camera (that supports onvif in stock firmware), it works with ODM and the NVR.

roleoroleo, 2022-04-03:
> Likely, the onvif implementation from onvif_srvd is also not compliant with the standard, hence why my NVR has issues.

Which version of Test Tool are you running?
And which tests fail?
Please let me know the name name/number of the test, I will check it.
I'm using a chinese NVR by Misecu and it works.

roleoroleo, 2022-04-04:
I tested the cam again with ODM and I have no problem with my cam.

cyraxnet, 2022-04-04:
> Which version of Test Tool are you running?
18.06 but the live stream function works here. With ODM (v2.2.250) it does not.  This was true for both v8 camera and v9. Tested in two different networks i.e. different routers to make sure to rule that out.

> And which tests fail?
I'm actually not sure how to best use the Onvif Test Tool. Would there be a test set up you suggest?

> I'm using a chinese NVR by Misecu and it works.
I have a HikVision NVR, but I can fully appreciate there might be an issue with yi-hack and HikVision, and it is ok with Misecu

> I tested the cam again with ODM and I have no problem with my cam.
Is this including the live view? For me I get "No Signal" if I try to connect to live stream

I'm happy to take any suggestions, because it is very confusing how to debug.

cyraxnet, 2022-08-30:
Just an update a few months on.  I couldn't get the hack and NVR to work, and the stream kept dropping for still unknown reason.
I gave up and tried the budget [Tapo cameras](https://www.tp-link.com/uk/home-networking/cloud-camera/tapo-c110/) which support ONVIF out-of-the-box and been working with my NVR from day 1.  No issues.  Would highly recommend them.

closed 2024-05-07 by github-actions[bot]