roleoroleo/yi-hack-Allwinner #353: Issue with only high stream in alt rtsp server

bonuzzz, 2022-04-17:
Seems rtsp doesn't work when using only high stream in alternate rtsp server.
I see two processes in ps:
h264grabber -m y20ga -r high -f
rtsp_server_yi -m y20ga -r high -a no -p 554

with only low stream it works.
y20ga 0.2.9

roleoroleo, 2022-04-17:
What about both streams?

roleoroleo, 2022-04-17:
I tested now your config:
- only high res
- no audio
It seems working, maybe it depends on other settings.

Please share your config (without personal data).

bonuzzz, 2022-05-05:
I switched between live555 and alternate rtsp server on camera several times in gui and seems after that high stream stopped working

bonuzzz, 2022-05-05:
Also I have these lags on vlc with live rtsp server. I don't know whether it is related or not. Video in zip. Wifi strength 90 percent in gui

[2022-05-05 14.35.38.mp4.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/8630807/2022-05-05.14.35.38.mp4.zip)

roleoroleo, 2022-05-06:
Your configs are ok.

> I switched between live555 and alternate rtsp server on camera several times in gui and seems after that high stream stopped working

It stopped working for both rtsp daemons?

About the mp4, you have a lot of lost frames. It's not a normal behavior.
Is your network ok?
I have no other ideas.

bonuzzz, 2022-05-06:
It stopped working only with alternate server. In that time I see recordings on microsd.
I tested again today. When I try to connect to camera with vlc to rtsp stream without audio, I got only error there. If I turn on audio, I hear sounds on vlc but without video. I attached last config
[config.tar.bz2.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/8642574/config.tar.bz2.zip)

bonuzzz, 2022-05-07:
I noticed rmm process not exist and rRTSPserver process has high cpu usage (70-90 percent in top)
It is on live555 rtsp server.. 
I have several mstar cameras with the same config and they work without any issues

roleoroleo, 2022-05-10:
The config is ok.

> I noticed rmm process not exist and rRTSPserver process has high cpu usage (70-90 percent in top)

If rmm doesn't exist it's a big problem.
The cam can't work if rmm is not running.
Probably rRTSPServer increases the cpu usage because the the video buffer is not filled by rmm.

Maybe an error in the whatchdog?
Try to comment the line with wd_rtsp.sh and reboot.
https://github.com/roleoroleo/yi-hack-Allwinner/blob/4f14f09e940bb4a62eaf00752656bd747da22f9a/src/static/static/home/yi-hack/script/system.sh#L282

closed 2024-05-07 by github-actions[bot]