roleoroleo/yi-hack-Allwinner #355: y25ga not updating to 0.3.0 / showing wrong version?

Madaraxa, 2022-04-19:
Hi.
I got two Yicams who need the y25ga firmware.
If I update them through web-if or with files on SD they wont update or still show 0.2.9.
I'm not sure how to check if I am on 0.3.0 and the Web-If only shows 0.2.9 or if I just cannot update...
Anyone else got this issue?

roleoroleo, 2022-04-23:
I checked the archive
https://github.com/roleoroleo/yi-hack-Allwinner/releases/download/0.3.0/y25ga_0.3.0.tgz
and the version is correct.
The update procedure could be bugged but if you create again the sd card it must work.

Try to clean the cache of your browser.

Madaraxa, 2022-04-23:
I'm sorry. But it either doesnt Update or it Shows the wrong Version in the Web-interface.
I took the sd out of the cam, deleted all files, formated the sd. Used your link here to download the files. Put the files (double unpacked) on root of sd, put it in cam and powered up the cam.
It still shows 0.2.9 in the Web interface of the cam...
Maybe its just a visual Bug in Web interface? So the cam is on 0.3.0 but Shows 0.2.9?

roleoroleo, 2022-04-24:
Login with ssh and check the version file:
`cat /home/yi-hack/version`

Madaraxa, 2022-04-24:
It tells me 0.2.9
I really formated the micro sd and only used your files.

roleoroleo, 2022-04-24:
I donwloaded the file again. The file "version" contains 0.3.0.
Try to open the file home_y25gam.stage with 7zip and check the content of version.

Madaraxa, 2022-04-24:
Hmm. You are right. It just dont Update... I dont know what i could try now... I reset cam in Web interface and tried it after as Well...

essys, 2022-04-25:
Same here.
Tried to update a y25ga over the air from 0.27 -> 0.30, seems to work perfectly fine, but after reboot I still have 0.27 installed and it didn't update.

GitHubUser325, 2022-04-25:
Same game for me. Update is neither online via web interface - nor offline via SD card possible.. Model y25ga, Firmware 0.2.7

roleoroleo, 2022-04-25:
The upgrade procedure is complicated.
I will try to improve it.

roleoroleo, 2022-04-25:
After the upgrade, is it present the file "home_y25gam.done" in the sd card?

roleoroleo, 2022-04-25:
Could you please tell me how much free space is there in /home.
Command `df` in shell.

Madaraxa, 2022-04-25:
root@yi-EG:/home# df
Filesystem           1024-blocks    Used Available Use% Moun
ted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528      1824     28704   6% /dev
/dev/mtdblock3           11776      9620      2156  82% /hom
e
tmpfs                    32768        80     32688   0% /tmp
/dev/mmcblk0p1        15613952    444056  15169896   3% /tmp
/sd
/dev/mtdblock3           11776      9620      2156  82% /etc
/passwd
/dev/mtdblock3           11776      9620      2156  82% /etc
/shadow

roleoroleo, 2022-04-25:
I can't understand.
I checked whole procedure and it seems to be ok.
If someone could help me, I would like to make a test.

1 - Download the latest firmware and decompress it.
2 - Copy home_y25gam.stage in your sd card (/tmp/sd) and rename it to home_y25gam.
3 - Run the following commands.

```
root@yi-hack-allwinner:~# mkdir /tmp/update
root@yi-hack-allwinner:~# cp -rf /home/base/tools/extpkg.sh /tmp/update/
root@yi-hack-allwinner:~# killall crond
root@yi-hack-allwinner:~# killall wsdd
root@yi-hack-allwinner:~# killall wd_rtsp.sh
root@yi-hack-allwinner:~# killall rRTSPServer
root@yi-hack-allwinner:~# killall onvif_srvd
root@yi-hack-allwinner:~# killall ntpd
root@yi-hack-allwinner:~# killall ipc_multiplexer
root@yi-hack-allwinner:~# killall mqttv4
root@yi-hack-allwinner:~# killall mqtt-config
root@yi-hack-allwinner:~# killall pure-ftpd
root@yi-hack-allwinner:~# killall httpd
root@yi-hack-allwinner:~# killall dropbear

root@yi-hack-allwinner:~# /tmp/update/extpkg.sh /tmp/sd/home_y20gam
```

Madaraxa, 2022-04-25:
After killall dropbear my connection get canceled by cam (using putty) and i cant connect again

Madaraxa, 2022-04-25:
For better Feedback i use a new Post:
Its updated now to 0.3.0.
Could it be your typo in some updatefile? 
"i try it with /tmp/update/extpkg.sh /tmp/sd/home_y25gam now (with 25 not 20)"

roleoroleo, 2022-05-01:
> After killall dropbear
> 
> my connection get canceled by cam (using putty) and i cant connect again.

You can use telnet instead of ssh.

> EDIT: well. u had a typo :) i try it with /tmp/update/extpkg.sh /tmp/sd/home_y25gam now (with 25 not 20)

Yes, 20 is for my cam.

The update procedure runs exactly this command. I don't know why it doesn't work.

> Could it be your typo in some updatefile?

In the ota procedure the type is not present.

Madaraxa, 2022-05-02:
Well. 
I updated the second y25 like i did with my first, so both are up to date now..
I dont know what it was that failed at the OTA.

GitHubUser325, 2022-05-05:
Since I was on 0.2.7, I tried first to update to 0.2.8 - then to 0.2.9 - and then 0.3.0 (all via SD-Card Method).. but only first two Updates worked, so I'm now on 0.2.9, but 0.3.0 Update still doesn't work - neither online, nor via SD-Card :(

essys, 2022-05-05:
> I can't understand. I checked whole procedure and it seems to be ok. If someone could help me, I would like to make a test.
> 
> 1 - Download the latest firmware and decompress it. 2 - Copy home_y25gam.stage in your sd card (/tmp/sd) and rename it to home_y25gam. 3 - Run the following commands.
> 
> ```
> root@yi-hack-allwinner:~# mkdir /tmp/update
> root@yi-hack-allwinner:~# cp -rf /home/base/tools/extpkg.sh /tmp/update/
> root@yi-hack-allwinner:~# killall crond
> root@yi-hack-allwinner:~# killall wsdd
> root@yi-hack-allwinner:~# killall wd_rtsp.sh
> root@yi-hack-allwinner:~# killall rRTSPServer
> root@yi-hack-allwinner:~# killall onvif_srvd
> root@yi-hack-allwinner:~# killall ntpd
> root@yi-hack-allwinner:~# killall ipc_multiplexer
> root@yi-hack-allwinner:~# killall mqttv4
> root@yi-hack-allwinner:~# killall mqtt-config
> root@yi-hack-allwinner:~# killall pure-ftpd
> root@yi-hack-allwinner:~# killall httpd
> root@yi-hack-allwinner:~# killall dropbear
> 
> root@yi-hack-allwinner:~# /tmp/update/extpkg.sh /tmp/sd/home_y20gam
> ```
> 
> 4 - Post the output.

Basically this worked for me, even though there is the aforementioned typo and also you shouldn't kill the ssh server.
So what I did was the following:

- extract latest FW
- copy home_y25gam.stage to SD root folder and rename it to home_y25gam
- boot camera and login via SSH
- killall crond
- killall wsdd
- killall wd_rtsp.sh
- killall rRTSPServer
- killall onvif_srvd
- killall ntpd
- killall ipc_multiplexer
- killall mqttv4
- killall mqtt-config
- killall pure-ftpd
- killall httpd
- run /tmp/update/extpkg.sh /tmp/sd/home_y2**5**gam

```
------------------------------------home app script update.sh enter------------------------------------
yihackver, newyihackver=0.3.0, curyihackver=0.2.7
update yi-hack
check every file
```

Now I successfully updated to 0.3.0.

GitHubUser325, 2022-05-05:
I've tried your commands, and everything worked, the script was running until:
...
...
home/yi-hack/lib/libasound.so.2.0.0
home/yi-hack/lib/libproxychains4.so
------------------------------------home app script update.sh enter------------------------------------
yihackver, newyihackver=0.3.0, curyihackver=0.2.9
update yi-hack

After that, the camera became unresponsive. I've waited about 10mins, but nothing happened. Rebooted, and couldn't connect SSH (Connection refused), ping was still working. Rebooted several times, no difference. Installed 0.2.9 via SD-Card Method, now its working again.
I think, I skip this 0.3.0 Update :D don't want to brick the cam finally.. thanks anyways, happy enough :)

essys, 2022-05-05:
@GitHubUser325 try to use a different SD card.

GitHubUser325, 2022-05-05:
I have no other SD card lying around here .. sorry, can't test it with a different one :/ but why should the 0.3.0 need another? those updates before were working perfectly (multiple times) with this SD card .. mhh

roleoroleo, 2022-05-07:
Maybe the problem is your record folder.
If the record folder in the sd card is full, the update procedure is very very slow.
Because the procedure (made by yi) checks the hash of every file and it takes long time to calculate it.
It depends on the number of mp4 files, it can take hours...

closed 2024-05-06 by github-actions[bot]