roleoroleo/yi-hack-Allwinner #356: Wifi and other settings lost if device reset or powered off

talkingwalnut, 2022-05-03:
Hi all,

Strange one. I've had my Yi Home 1080p (9FUS) for a few years now and it's been working fine every time I've used it. I started using it again a few weeks ago and all was fine until I plugged it in last night. The orange LED came on and I got the "Welcome to yi home. Waiting to connect" audio message. I set the wifi connection up again in the app and when I connected to the web page is showed the Yi Hack was still active, but some settings had changed. The problem is that if I try and change anything and reboot to activate, I end up back at the "Waiting to connect" step. 
I've tried wiping and scanning the sdcard (h2testw) and no issues were reported. I've tried unhacking and rehacking and still have the same issue. It was working fine for a week or so on this firmware and there's been no network changes or settings changes prior to this issue.
Has anyone any ideas?

roleoroleo, 2022-05-05:
You should try to:
- unhack the cam with the unbrick procedure (stronger than just unhack) using "hacked" option
- reset the cam with the back button
- reconfigure it from scratch
- apply again the hack if you want

talkingwalnut, 2022-05-09:
Trying this now but keep getting the following error when I try to build the unbricked file:
`modprobe: FATAL: Module mtdram not found in directory /lib/modules/5.15.32-v7l+
umount: y20ga/mnt: not mounted.
cp: cannot create regular file 'y20ga/home/app/init.sh': No such file or directory
cp: cannot create regular file 'y20ga/home/app/script/update.sh': No such file or directory
cp: cannot create regular file 'y20ga/home/app/script/wifidhcp.sh': No such file or directory
cp: cannot create regular file 'y20ga/home/base/tools/extpkg.sh': No such file or directory
./create.jffs2: 3: ./mkfs.jffs2: Exec format error
mkimage: Can't open y20ga/mtdblock3_hacked.bin: No such file or directory
`

Tried it on a raspberrypi 5.15.32 and Ubuntu under Windows and got the same issue. Any ideas?

roleoroleo, 2022-05-10:
I think that mtdram module is not available on this platforms.
You should try a standard linux distribution on x86 or amd64.
Or you should modify the script with another jffs2 extractor (jefferson for example).

jojo141185, 2022-09-05:
I had exactly the same Problem with my y20ga camera.
After a reset (with female voice), I can connect the camera to the WiFi via the app using the qr code. But as soon as I restart or reset the camera, it reboots and gets stuck with a flashing blue light and doesn't connect to WiFi.

According to your recommendation I did also create the unbrick image with my initial backup "mtdblock3.bin" :
`./build.sh hacked
23552+0 Datensätze ein
23552+0 Datensätze aus
12058624 Bytes (12 MB, 12 MiB) kopiert, 0,0684203 s, 176 MB/s
Successfully mounted y20ga/mtdblock3.bin on y20ga/mnt
Image Name:   xiaoyi-home
Created:      Mon Sep  5 21:45:43 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    12058624 Bytes = 11776.00 KiB = 11.50 MiB
Load Address: 00000000
Entry Point:  00000000
`
Then extracted the generated "home_y20ga.gz" and copied the "home_y20ga" file to the root of the sdcard. 
After a long reboot, the camera gets stuck with a flashing blue light. After a reset on the small back button (no voice) I can't get it to live again :-(

roleoroleo, 2022-09-06:
With the unbrick method you should be able to restore the cam (if the bootloader is ok).
Try again with the command:
./build.sh factory
In this manner the hash of the image is different and the cam will reload it.

jojo141185, 2022-09-06:
After several attempts to hack or unhack the firmware, I also managed to brick the camera. Only the yellow light was on and the camera no longer booted (no clicking noise, no blue LED). So I opened the case and unbricked the camera with my usb to serial converter according to the wiki entry.
Now the camera is online again, the firmware has been hacked again, but the problem that the camera no longer automatically connects to the wifi persists. After every reboot I have to transmit my wifi credentials again via the app and the QR code of the camera.
@roleoroleo Do you have an idea what could be the reason for this?

jojo141185, 2022-09-08:
For hours I tried various things to get to the bottom of the error. Unfortunately, the points mentioned here did not bring any improvement either... The camera simply did not want to remember the Wifi credentials. After each reboot, the qr code from the yi home app was repeatedly requested :-(
With some research in the various scripts of this repo I understood how and where the wifi access data is stored.
I took a closer look at the /dev/mtdblock6 memory and saw that the SSID and password (in plain text) were repeatedly stored "corruptly".

So I started from scratch and overwrote all MTDblock devices with my backup data:
```
dd if=/tmp/sd/backup/mtdblock0.bin of=/dev/mtdblock0
dd if=/tmp/sd/backup/mtdblock1.bin of=/dev/mtdblock1
dd if=/tmp/sd/backup/mtdblock2.bin of=/dev/mtdblock2
dd if=/tmp/sd/backup/mtdblock3.bin of=/dev/mtdblock3
dd if=/tmp/sd/backup/mtdblock4.bin of=/dev/mtdblock4
dd if=/tmp/sd/backup/mtdblock5.bin of=/dev/mtdblock5
dd if=/tmp/sd/backup/mtdblock6.bin of=/dev/mtdblock6
dd if=/tmp/sd/backup/mtdblock7.bin of=/dev/mtdblock7
reboot
```

After that the firmware was back to its original state. Then I reconnected to wifi via app and QR code. Imported the hack via SDCard and loaded my old config again via the web interface... lo and behold - everything works :D
How it came about... no idea! But I won't be touching the wifi data anytime soon ;-)

In any case, many thanks for all your support here and this great firmware hack!

roleoroleo, 2022-09-09:
Glad to hear that you have solved it.
Next time avoid overwriting mtdblock0, it is a risky operation.

jojo141185, 2022-09-10:
Unfortunately, I got excited too soon!
After the camera was switched off for a longer time, it could not reconnect to WiFi today. Had to connect via QR code from the YiApp again. Then I set the access data via WebIF.

I tried to reproduce the steps of the cgi-script and looked at the content of mtdblock6.

Create /tmp/configure_wifi.cfg with content:
```
wifi_ssid=MYSSID
wifi_psk=MYSECRET
```

Run script: /home/yi-hack/script/configure_wifi.sh

```
...
CURRENT_SSID=$(dd bs=1 skip=28 count=64 if=/dev/mtdblock6 2>/dev/null)
CURRENT_KEY=$(dd bs=1 skip=92 count=64 if=/dev/mtdblock6 2>/dev/null)
CONNECTED_BIT=$(hexdump -s 24 -n 4 -v /dev/mtdblock6 | awk 'FNR <=1' | awk '{print $3$2}')

echo $SSID ${#SSID} - $CURRENT_SSID ${#CURRENT_SSID}
echo $KEY ${#KEY} - $CURRENT_KEY ${#CURRENT_KEY}
....
```

Output:
MYSSID 6 - PA'c'i 6
MYSECRET 8 - 0"!PepUi 8

I can try this multiple times. The output of CURRENT_SSID or CURRENT_KEY is every time different and looks distorted / crypted. Can someone reproduce this behavior? 
Or is the block of my flash memory broken? :-(

PS: I will go with [this](https://github.com/roleoroleo/yi-hack-Allwinner/issues/289) workaround for now.

cchenga, 2023-01-31:
Hi @jojo141185 , I got the same issue. Even worse for me, I cannot find my firmware backup. I connected their support for a firmware, they refused to send me a copy because my devices are out of support.
Could you share your backup with me?

jojo141185, 2023-02-02:
Have a look at the [unbrick howto](https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam) and this [post](https://github.com/roleoroleo/yi-hack-Allwinner/issues/365) for details and (ASK for) backups.

closed 2024-05-06 by github-actions[bot]