roleoroleo/yi-hack-Allwinner #361: High CPU utilization after upgrading to 0.30

MB-Finski, 2022-06-20:
**Issue description**

After upgrading to YiHack 0.30 the CPU load average climbs to  between 2.0 and 3.0 and the rtsp stream stops working. The issue goes away after downgrading to 0.29. With 0.29 the load average stabilizes at around 0.5.

**Steps to reproduce the issue**

Serial/suffix: 9FUS, y20ga
Base version: 8.2.0.0A_202012041528
Upgrade from YiHack 0.29 to 0.30 (absolutely no changes to configuration)

I have 3 cameras and all have the same issue.

**Expected behavior**

Normal function of YiHack

**Current behavior**

Camera stop responding due to heavy CPU load. I've tried disabling all services besides RTSP stream with no effect.

roleoroleo, 2022-06-25:
Maybe the service that creates snapshots is too expensive.
Did you try to enable swap space?

roleoroleo, 2022-06-30:
thumb.sh is the snapshot service that create snapshots of the captured mp4 files.
I added it in the 0.2.9 and at the moment it's not possible to disable it from web interface.
Please comment the line in system.sh:
`echo "* * * * * /home/yi-hack/script/thumb.sh cron" >> /var/spool/cron/crontabs/root`

outofsight, 2022-07-08:
What are too high load averages that requires disabling snapshot services?

roleoroleo, 2022-07-09:
If the swap is enabled, I think that your problem is different.
But you can find this file in /home/yi-hack/script/

MB-Finski, 2022-07-16:
Hmm.. Even after disabling the snapshot service the original issue persists. Load average is now down to about 1.4 (CPU % sits at around 30-40%) but the skipped frames issue is still there.

I even tried resetting one of the cameras and changed the credentials so that nothing else should be connecting to it. Still skipped frames/occasional freezes. Hmm... I'll fire up an old Wifi-router to exclude the possibility of networking issues.

MB-Finski, 2022-07-16:
... seems like it was a networking issue, after all. Well, now I have a reason to setup a separate wifi for the home automation system.

closed 2022-07-16 by MB-Finski