roleoroleo/yi-hack-Allwinner #365: bricked 9fus y25ga

candidotsa, 2022-08-26:
hi. his this camara a paper weight?

[125]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[130]BOOT0 commit : 81c18ee
[133]board init start
[135]set pll start
[138]set pll end
[139][pmu]: bus read error
[142]board init ok
[144]chip id check OK
[146]DRAM BOOT DRIVE INFO: V0.41
[149]DRAM CLK = 528 MHz
[152]DRAM Type = 2 (2:DDR2,3:DDR3)
[155]DRAMC read ODT  off.
[157]DRAM ODT off.
[160]DRAM SIZE =64 M
[167]DRAM simple test OK.
[169]rtc standby flag is 0x0, super standby flag is 0x0
[174]dram size =64
[177]spinor id is: 1c 70 18, read cmd: 03
[181]Succeed in reading toc file head.
[185]toc1 magic error
CCC

this appens after i try to flash the boot partition. thanks

roleoroleo, 2022-08-26:
Sorry but yes...
You can recover it "offline".
Desoldering the spi flash and programming it with an external programmer.

There are a lot of diy programmers.
Search for "SPIPGM" (parallel port) or FT232H (usb port).

candidotsa, 2022-08-27:
hi roleoroleo, thanks for your prompt help.
i use a ftd1232 for flash, this one do the job or has to be the 32H?
this is a strange camera (at least for me) i bricked another one and i was able to restore just copy the files on backup folder to the root of sd card, after power the camera he start upgrade and start asking for connection ( how it was possible? don´t ask me)
[image]
[image]
[image]

roleoroleo, 2022-08-27:
Maybe your adapter could work.
https://www.flashrom.org/FT2232SPI_Programmer

candidotsa, 2022-08-27:
just finish the voltage regulator, tomorrow i will desolder the ic and try to connect to the program
[image]
mer. thanks for your help

candidotsa, 2022-08-30:
hi roleoroleo. everything is ready. 
[image]

candidotsa, 2022-08-30:
i received a new one and i have flash the y25ga_0.3.0. the camera is connected  but i can´t access them, neither in the official application nor in the ip. i won´t try anything or may will happens the same as the previous. i hope you can help. thank you
[backup yi3 3a.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/9455728/backup.yi3.3a.zip)

roleoroleo, 2022-09-02:
Did you restore the old cam?
If not, what kind of help do you need?

About new cam, is the cam connected to the wifi network?
Are you able to get the ip address?
Are you able to login with telnet?

candidotsa, 2022-09-02:
Hi roleo. The old cam I'm trying to see how the flash it's done. Never done
this before. The 3rd camera didn't try with telnet. The led stay blue and
the camera shows connect on the router. Will try telnet in a hour. Thank
you

roleoroleo, 2022-09-02:
About the old cam, you could start following the page above:
https://www.flashrom.org/FT2232SPI_Programmer

First step is to make a backup copy of the flash.
If it works, make another copy and check if are identical.
Just to check if the interface is working properly.

2nd step is to erase the whole flash.

Finally, you need program the bootloader at the starting address of the flash.
If you restore only the bootloader, you will need to apply the unbrick guide to write the other partitions (from 1 to 6) using the sd card.
Otherwise you could prepare a full image of the flash (16MB = 16.777.216 B) but the write procedure is slow.

roleoroleo, 2022-09-02:
With another SD card, make a backup of the cam (after the hack).
I will check if the hack is ok.

candidotsa, 2022-09-03:
> 

[backup1.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/9482466/backup1.zip)
[backup2.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/9482467/backup2.zip)

roleoroleo, 2022-09-06:
The cam is correctly hacked.
The boot script `/home/app/init.sh` contains the following lines:
```
/usr/sbin/telnetd &
/home/yi-hack/script/system.sh
```
so it's ok.
You should be able to login with telnet.

candidotsa, 2022-09-07:
hi. i have done everything from scratch with a different sd and flashed again. the same problem. i can ping the camera but i can´t login with putty or in the browser.

roleoroleo, 2022-09-08:
Please, check the following details:
- Are you really pinging the cam or another device? What happens if you switch off the cam? Does ping still work?
- When you start putty, do you select telnet or ssh?
- When you try to connect to the web interface, are you using port 8080?

candidotsa, 2022-09-08:
hi. its really the camera.
C:\>ping 192.168.1.88

Pinging 192.168.1.88 with 32 bytes of data:
Reply from 192.168.1.88: bytes=32 time=175ms TTL=64
Reply from 192.168.1.88: bytes=32 time=1ms TTL=64
Reply from 192.168.1.88: bytes=32 time=1ms TTL=64
Reply from 192.168.1.88: bytes=32 time=21ms TTL=64

Ping statistics for 192.168.1.88:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 175ms, Average = 49ms

C:\>ping 192.168.1.88

Pinging 192.168.1.88 with 32 bytes of data:
Request timed out.
Request timed out.
Request timed out.
Request timed out.

Ping statistics for 192.168.1.88:
    Packets: Sent = 4, Received = 0, Lost = 4 (100% loss),

C:\>ping 192.168.1.88

Pinging 192.168.1.88 with 32 bytes of data:
Reply from 192.168.1.88: bytes=32 time=2ms TTL=64
Reply from 192.168.1.88: bytes=32 time=8ms TTL=64
Reply from 192.168.1.88: bytes=32 time=1ms TTL=64
Reply from 192.168.1.88: bytes=32 time=1ms TTL=64

Ping statistics for 192.168.1.88:
    Packets: Sent = 4, Received = 4, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 1ms, Maximum = 8ms, Average = 3ms

my bad on putty i was connecting ssh instead telnet.
is request a user and password.
   xiaoyi login:
on the browser:

http://192.168.1.88:8080/
ERR_CONNECTION_REFUSED

roleoroleo, 2022-09-08:
Default is "root" user and blank password.
Try the following command and post the output:
```
ps
mount
df
```

candidotsa, 2022-09-08:
root@xiaoyi:~# ps
  PID USER       VSZ STAT COMMAND
    1 root       980 S    /sbin/init
    2 root         0 SW   [kthreadd]
    3 root         0 SW   [ksoftirqd/0]
    4 root         0 SW   [kworker/0:0]
    5 root         0 SW<  [kworker/0:0H]
    6 root         0 SW   [kworker/u2:0]
    7 root         0 SW   [rcu_preempt]
    8 root         0 SW   [rcu_sched]
    9 root         0 SW   [rcu_bh]
   10 root         0 SW<  [lru-add-drain]
   11 root         0 SW   [kdevtmpfs]
   12 root         0 SW   [kworker/u2:1]
  179 root         0 SW   [oom_reaper]
  180 root         0 SW<  [writeback]
  182 root         0 SW   [kcompactd0]
  183 root         0 SW<  [crypto]
  184 root         0 SW<  [bioset]
  186 root         0 SW<  [kblockd]
  224 root         0 SW   [sys_user]
  231 root         0 SW   [kworker/0:1]
  232 root         0 SW<  [cfg80211]
  239 root         0 SW<  [watchdogd]
  253 root         0 SW<  [spi0]
  263 root         0 SW   [kswapd0]
  348 root         0 SW<  [bioset]
  353 root         0 SW<  [bioset]
  358 root         0 SW<  [bioset]
  363 root         0 SW<  [bioset]
  368 root         0 SW<  [bioset]
  373 root         0 SW<  [bioset]
  378 root         0 SW<  [bioset]
  383 root         0 SW<  [bioset]
  402 root         0 SW   [irq/302-sunxi-m]
  404 root         0 SW   [irq/166-sdc0 cd]
  405 root         0 SW   [irq/303-sunxi-m]
  408 root         0 SW   [kworker/0:2]
  452 root         0 SW   [kworker/u2:2]
  463 root         0 SW<  [kworker/0:1H]
  464 root         0 SW   [kworker/0:3]
  466 root         0 SW<  [bioset]
  467 root         0 SW   [mmcqd/0]
  482 root         0 SWN  [jffs2_gcd_mtd3]
  487 root       980 S    -/bin/sh
  528 root         0 SW   [ksdioirqd/mmc1]
  531 root         0 DW   [RTW_XMIT_THREAD]
  532 root         0 SW   [RTW_CMD_THREAD]
  533 root         0 SW   [RTWHALXT]
  534 root      1072 S    ./dispatch
  537 root       988 S    /usr/sbin/telnetd
  579 root      1400 S    /home/base/tools/wpa_supplicant -c/tmp/wpa_supplican
  610 root       980 S    udhcpc -i wlan0 -b -s /home/app/script/default.scrip
  743 root       980 S    -ash
  796 root       980 R    ps
root@xiaoyi:~# mount
/dev/root on / type squashfs (ro,relatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=30528k,nr_inodes=7632,mode=755)
proc on /proc type proc (rw,nosuid,nodev,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,relatime)
/dev/pts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock3 on /home type jffs2 (rw,relatime)
none on /sys/kernel/debug type debugfs (rw,relatime)
none on /dev/mqueue type mqueue (rw,relatime)
tmpfs on /tmp type tmpfs (rw,relatime,size=32768k)
/dev/mmcblk0p1 on /tmp/sd type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
root@xiaoyi:~# df
Filesystem           1024-blocks    Used Available Use% Mounted on
/dev/root                 1536      1536         0 100% /
devtmpfs                 30528         0     30528   0% /dev
/dev/mtdblock3           11776      6344      5432  54% /home
tmpfs                    32768        28     32740   0% /tmp
/dev/mmcblk0p1          871696     19104    852592   2% /tmp/sd
root@xiaoyi:~#

candidotsa, 2022-09-08:
hi again. i was able to read the bricked cam bios. i use a raspberry with this tutorial https://www.youtube.com/watch?v=KNy-_ZzMnG0 .
 my ftd1232  was not recognized by the flashrom. 
[backup_2.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/9529474/backup_2.zip)
 
[image]

roleoroleo, 2022-09-09:
About the new cam, processes are not running.
Something is wrong on your /home directory, confirmed by your backup2.zip
I don't know why but the yi-hack folder is not present.
I checked the released archive and it's ok.
Decompress the archive and copy the file home_y25gam to your sd card.
Then reboot.

roleoroleo, 2022-09-09:
About the old cam.
The backup you made is good between 0x0 and 0x10000 but is blank (all zeroes) between 0x10000 and 0x80000.
You need to rewrite it with another file.
You could use the mtdblock0.bin from your new cam.

candidotsa, 2022-09-11:
Good morning. New news, the new cam is working with version 2.9 I tried to online update but still on 2.9 later I will try manually extract all the files from the V3 and see if work. The cam with the bios, I dump the bios from the new one check the md5 a lot of time and a was able to get one file from all the reads with the same md5. I write that file in the old bios and I get the same md5 (all good it seems) I solder the booth bios in place the new cam is working and the old is blinking orange. I tried to reset but nothing. Any suggestions? Hardware fault or software bug? Thanks

roleoroleo, 2022-09-11:
Blinking orange is good.
It means that bootloader is ok.
Which partition did you restore?
The whole flash (16 MB)?

candidotsa, 2022-09-11:
yes, the whole flash. after a min the led keeps fixed orange.

[work7a.zip](https://github.com/roleoroleo/yi-hack-Allwinner/files/9542440/work7a.zip)

roleoroleo, 2022-09-11:
Fixed orange means that bootloader is ok.
I don't know what is happening.
But you could try to restore the single partitions.
Starting with rootfs and home.
Check the unbrick guide: https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick
Create the images (starting from your mtdblockX.bin) with these commands:
```
mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-rootfs" -d mtdblock2.bin rootfs_y25
mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-home" -d mtdblock3.bin home_25
```
Pay attention that if you cloned flash of your new cam in the old one, they share the same serial number and the same access key to the cloud. But at the moment we don't care about this problem.

candidotsa, 2022-09-11:
U-Boot 2018.05-00002-g932b424-dirty (Dec 13 2019 - 18:21:53 +0800) Allwinner Technology

[00.249]CPU:   Allwinner Family
[00.252]Model: sun8iw19
I2C:   ready
[00.256]DRAM:  64 MiB
[00.259]Relocation Offset is: f9f6a000
[00.275]secure enable bit: 0
[00.277]pmu_axp2101_probe pmic_bus_read fail
[00.281]drivers/sunxi_power/pmu_axpnull.c pmu_axpnull_probe 26 axpdummy probe
[00.288]bmu_axp2101_probe pmic_bus_read fail
[00.292]BMU: AXPNULL
[00.295]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.301]gic: sec monitor mode
[00.304]flash init start
[00.306]workmode = 0,storage type = 3
SF: Detected en25qh128 with page size 256 Bytes, erase size 64 KiB, total 16 MiB
[00.316]sunxi flash init ok
[00.320]Loading Environment from SUNXI_FLASH... OK
[00.351]sprite_led_gpio start
[00.355]update dts
root_partition is rootfs
set root to /dev/mtdblock2
[00.365]update part info
[00.367]update bootcmd
[00.369]No ethernet found.
[00.371]get mem for descripter OK !

[00.597]Card did not respond to voltage select!
card0 init failed
No sd-card insert,will normal boot
Hit any key to stop autoboot:  0
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2080440 Bytes = 2 MiB
   Load Address: 40008000
   Entry Point:  40008000
[02.114]Starting kernel ...

prefetch abort
pc : [<004e0bf0>]          lr : [<4035d59a>]
reloc pc : [<00440be6>]    lr : [<402bd590>]
sp : 40556ca0  ip : 40557bd8     fp : 40555199
r10: bfff7fff  r9 : 40556dd4     r8 : 00000b44
r7 : 40556d38  r6 : 00000005     r5 : 0000000d  r4 : 40556d68
r3 : c0185ae5  r2 : 004e0bf5     r1 : 00000005  r0 : 40555cd7
Flags: Nzcv  IRQs off  FIQs off  Mode SVC_32
Resetting CPU ...

resetting ...
[38]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[43]BOOT0 commit : 81c18ee
[46]board init start
[48]set pll start
[50]set pll end
[52][pmu]: bus read error
[54]board init ok
[56]chip id check OK
[58]DRAM BOOT DRIVE INFO: V0.41
[61]DRAM CLK = 528 MHz
[63]DRAM Type = 2 (2:DDR2,3:DDR3)
[67]DRAMC read ODT  off.
[69]DRAM ODT off.
[71]DRAM SIZE =64 M
[78]DRAM simple test OK.
[80]rtc standby flag is 0x0, super standby flag is 0x0
[86]dram size =64
[87]key press :
[90]spinor id is: 1c 70 18, read cmd: 03
[94]Succeed in reading toc file head.
[97]The size of toc is 6c000.
[171]Entry_name        = optee
[174]Entry_name        = u-boot
[179]Entry_name        = dtb
[182]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年 11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

roleoroleo, 2022-09-11:
The problem is the kernel.
Use your mtdblock1.bin from the new cam and create the image file with this command:
`mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-kernel" -d mtdblock1.bin kernel_25`
Copy the file in your sd card and restart the cam.

candidotsa, 2022-09-12:
i roleo, do not laugh. 

card_caps:0x3000000a
host_caps:0x3000003f
** Unrecognized filesystem type **
 (22.1 MiB/s)
image[/kernel_y25ga] wrong type
Nothing updated, start normal boot
Hit any key to stop autoboot:  0
=>
=> image[/kernel_y25ga] wrong type
Unknown command 'image[/kernel_y25ga]' - try 'help'
=> Nothing updated, start normal boot

this is why i have my first camera as y20ga after recover with some untrusted restore files that i found somewhere.

roleoroleo, 2022-09-13:
I can assure you that the partition you posted above comes from a y25ga.
At this point, I think that the only solution is the hard way:
https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Unbrick-the-cam
But we need to change something to adapt the procedure to the kernel:

Step 1: ...
Step 2: ...
Step 3: `fatload mmc 0 45000000 mtdblock1.bin`
Step 4: ...
Step 5: starting address of the kernel is `0x00080000` and size should be 2359296, in hex `0x240000`
Step 6: `sf erase 80000 240000`
Step 7: `sf write 45000000 80000 240000`

candidotsa, 2022-09-13:
good morning. ## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2080440 Bytes = 2 MiB
   Load Address: 40008000
   Entry Point:  40008000
[02.257]Starting kernel ...

[    0.000000] ion_mem_reserve: err format for ion reserve list!
[    0.111891] ion_heap_create: Invalid heap type 4
[    0.119491] sunxi_i2c_probe()2209 - [i2c1] warning: failed to get regulator id
[    0.147167] uart uart0: get regulator failed
[    0.174809] sunxi-wlan wlan: request pincrtl handle for device [wlan] failed
[    0.183157] sunxi-wlan wlan: get gpio chip_en failed
[    0.205528] mtd: partition "UDISK" is out of reach -- disabled
[    0.216920] VE: get debugfs_mpp_root is NULL, please check mpp
[    0.216920]
[    0.491640] mmc:failed to get gpios
[    0.539964] sunxi-mmc sdc1: smc 1 p1 err, cmd 52, RTO !!
[    0.546857] sunxi-mmc sdc1: smc 1 p1 err, cmd 52, RTO !!
[    0.557369] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
[    0.564144] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.570928] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.577698] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.584476] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.591268] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.598168] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.605070] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.611975] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.619012] sunxi-mmc sdc1: smc 1 p1 err, cmd 1, RTO !!
[    0.663939] SQUASHFS error: zlib decompression failed, data probably corrupt
[    0.671954] SQUASHFS error: squashfs_read_data failed to read block 0xe5eac
[    0.679813] SQUASHFS error: Unable to read fragment cache entry [e5eac]
[    0.687270] SQUASHFS error: Unable to read page, block e5eac, size 12ac2
[    0.695812] SQUASHFS error: zlib decompression failed, data probably corrupt
[    0.703775] SQUASHFS error: squashfs_read_data failed to read block 0xe5eac
[    0.711625] SQUASHFS error: Unable to read fragment cache entry [e5eac]
[    0.719081] SQUASHFS error: Unable to read page, block e5eac, size 12ac2
[    0.726698] Kernel panic - not syncing: Requested init /pseudo_init failed (error -5).
[    0.735533] CPU: 0 PID: 1 Comm: swapper Not tainted 4.9.118 #58
[    0.742192] Hardware name: sun8iw19
[    0.746148] [<c00118cd>] (unwind_backtrace) from [<c000fdeb>] (show_stack+0xb/0xc)
[    0.754675] [<c000fdeb>] (show_stack) from [<c0050ef1>] (panic+0x71/0x1b4)
[    0.762421] [<c0050ef1>] (panic) from [<c0226c99>] (kernel_init+0x49/0x9c)
[    0.770165] [<c0226c99>] (kernel_init) from [<c000d401>] (ret_from_fork+0x11/0x30)
[    0.778688] ---[ end Kernel panic - not syncing: Requested init /pseudo_init failed (error -5).
[    0.788527] crashdump enter

roleoroleo, 2022-09-13:
Ok.
The cam is trying to load the squasfs file system.
Now you have to load the rootfs and home partition:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/365#issuecomment-1242952232

roleoroleo, 2022-09-13:
Before rootfs and home you need env.

candidotsa, 2022-09-13:
its normal have zeros on load and entry point? 
-T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-env" -d mtdblock4.bin env_25
Image Name:   xiaoyi-env
Created:      Tue Sep 13 09:27:43 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    131072 Bytes = 128.00 KiB = 0.12 MiB
Load Address: 00000000
Entry Point:  00000000

roleoroleo, 2022-09-13:
Yes.

candidotsa, 2022-09-13:
the camera is working :) but now i have block the cloud. the new one dont connect. 
this is the log from the bricked
p2p_tnp.c(state_statistics-6235)[08:48:51.684]:check_login fail 0

roleoroleo, 2022-09-13:
Yes, I don't know exactly how to fix this.

Because the mtd7 partition contains some information that is unique for each cam: cloud ID, serial number, ...
If you have not made a backup copy of the old cam, you cannot enter this data.
And if you clone them from the new cam, you can't access the cloud.

Are you sure the SD card does not contain backup files?

candidotsa, 2022-09-13:
I have the backup of the 3 cameras.

candidotsa, 2022-09-13:
The files that I restore today are from the original backup.

roleoroleo, 2022-09-13:
Then restore mtd6 and mtd7 partitions and reboot.
You can do it directly from a console session with dd.

candidotsa, 2022-09-15:
> 

Hi roleo. How do I calculate to do the same for  mtd6 and 7? Thank you

candidotsa, 2022-09-15:
> I can assure you that the partition you posted above comes from a y25ga. At this point, I think that the only solution is the hard way: [...]

Hi roleo. How do I calculate to do the same for mtd6 and 7? Thank you

roleoroleo, 2022-09-19:
For mtd5:
Step 1: ...
Step 2: ...
Step 3: `fatload mmc 0 45000000 mtdblock5.bin`
Step 4: ...
Step 5: starting address of the kernel is `0x00FE0000` and size is 65536, in hex `0x10000`
Step 6: `sf erase FE0000 10000`
Step 7: `sf write 45000000 FE0000 10000`

For mtd6:
Step 1: ...
Step 2: ...
Step 3: `fatload mmc 0 45000000 mtdblock6.bin`
Step 4: ...
Step 5: starting address of the kernel is `0x00FF0000` and size is 65536, in hex `0x10000`
Step 6: `sf erase FF0000 10000`
Step 7: `sf write 45000000 FF0000 10000`

candidotsa, 2022-09-21:
Who's the boss??? You @roleoroleo !!! 
Thank you, is working fine. Now I understand the partition sizes 😇.

roleoroleo, 2022-09-22:
It was fun.
In one go we tried all the possible ways of restoring a partition.
I will add this issue in the wiki.

candidotsa, 2022-10-06:
The time that we spent I could buy a few cameras. I appreciate your help, I did this because of the challenger, now o know how a bios is recovered and hope never use it again 😂.

roleoroleo, 2022-10-07:
I agree.

candidotsa, 2022-10-11:
For me it was just for the challenger. The time that we spent ( I spent
hours on research.) Could buy at least 3 or 4 cameras😂. If it happens
again in less then an hour is all fixed.
 It's a trick fix, extract the bios (in my case, I don't know if there's
any clamp that fits there or if the ic can be read in place), have another
camera to extract the file. Soldering booth bios in place restore
partitions  etc. 🤯. But at the end, it's wasn't about the camera but
learning something more.
Many thanks to you and all the people like you. The donation was made
before because your prompt help, i valorize that.

roleoroleo, 2022-10-14:
It's the same for me.
I don't carry on these projects for money: probably with the time I spent, I could by a few thousand cameras.
And with the donations, 10-20 cams.
It's only a passion and it allows you to learn new things.

closed 2024-05-06 by github-actions[bot]