roleoroleo/yi-hack-Allwinner #371: Asking to connect

Wall-8, 2022-09-18:
9FUS model, after upgrade from 0.2.9 to 0.3.1
Two issues
1. sound detection doesn't work anymore
2. After a restart it just say's "Waiting to connect" and web interface from yi-hack doensn't work anymore and requires a reflash

roleoroleo, 2022-09-22:
I need time to check this problem.

Rushmed, 2022-10-03:
Also having this issue.

roleoroleo, 2022-10-06:
Only sound? Does motion detection work?

roleoroleo, 2022-10-06:
Please, try this library.
[ipc_multiplex.so.gz](https://github.com/roleoroleo/yi-hack-Allwinner/files/9722663/ipc_multiplex.so.gz)
Uncompress it to /tmp/sd/yi-hack/lib and reboot the cam.

roleoroleo, 2022-10-14:
I tested it again and, with the new ipc_multiplex.so, it works for me.

Rushmed, 2022-10-14:
Just to be sure /home/yi-hack/lib has to look like:
[image]
and sound detection still has its own topic, right?

roleoroleo, 2022-10-16:
> Edit: Switched rights of ipc_multiplex.so to 777, now it works. Is that as intended? Thanks.

It depends on how it was before. But 755 is enough.

> Hmm I cant seem to turn sound detection on in camera settings. If I save and refresh it is off again.

Try to refresh the cache of the browser.

jcPOLO, 2022-10-18:
it works... but detections are not happening always like motion detection does. Sometimes, you do a noise and cam does not send any mqtt packet.

Baby crying never works.

[image]

After rebooting the camera, first sound detected always works. Next ones seems like it is a random thing.

Any idea why this could be happening?

roleoroleo, 2022-10-22:
My cam also behaves like this.
I think it's a problem of the yi software because the original app also does not receive notifications.

closed 2024-05-06 by github-actions[bot]