roleoroleo/yi-hack-Allwinner #372: No more FTP access

Rushmed, 2022-10-08:
V0.3.1

Hi, I can not longer acces my cameras per FTP.
Server is reachable but acces id denied an password ist noct accepted.
Also tried changing teh PW.
[image]
With legecy option it works for Filezilla but WinSCP which I prefer to use the record folder has no contend.

roleoroleo, 2022-10-11:
Version?

roleoroleo, 2022-10-11:
There are no changes in the last version about ftp.
I added sftp-server but it's not related to ftp.
Are you able to login with ssh?
Does sftp-server work?

Rushmed, 2022-10-11:
Oh, I had SSH disabled.
After enabling SSH, SFTP is workig.

Did not know that before thanks a lot!

roleoroleo, 2022-10-11:
I will check ftp server.
There is a problem with the authentication.

closed 2024-04-01 by Rushmed