roleoroleo/yi-hack-Allwinner #374: Differences between AllWinner and MStar SoCs?

iointerrupt, 2022-10-15:
Can anyone she some light on the differences in the cams based on the AllWinner and MStar chipsets?  Does one have better performance (stability) than the other?

I ask because I have a couple of Yi Home and Dome 720p cameras (47US and 43US) that are using the Hi3518e* chipsets.  I was hesitant a while back to upgrade to the 1080p's because they were using same chipset - Yi upgraded the cameras but lowered the fps on the 1080ps.  Since there are now MStar and AllWinner SoC variations, wondering which would yield the best performance/stability in this matter.

roleoroleo, 2022-10-16:
IMHO MStar is more stable than Allwinner but also less powerful.
But probably MStar is no longer available, it depends on the model (home, dome, outdoor, etc...).
From the perspective of the fw, the big difference is the structure of the file system.
MStar has 2 r/w partitions.
Allwinner has 2 or 3 partitions, mixed read only and r/w.

iointerrupt, 2022-10-17:
Thanks for the clarifications.  Just ordered a 9FUS (AllWinner) on ebay to test.  They appear to be easier to find than the MStar, which like you mentioned probably is no longer available.

closed 2024-05-06 by github-actions[bot]