roleoroleo/yi-hack-Allwinner #375: Wi-Fi issues (multiple SSID?)

andriej, 2022-10-17:
I have latest firmware on my cams and strange situation.

In garage and base floor when they were placed - cameras are not connecting to WiFi anymore.
They are working as I can hear IR switching on/off depending on light.

When I took them away for debugging - they are connecting with no issues.

Both of cameras are near 2 new APs which are WiFi6/AX (but with WPA2, no WPA3 enabled).
They are connecting to 3rd new AP which is also WiFi6/AX (same WPA2 pass).
All AP have same SSID and 802.11r turned on.

Has anyone else got same issues? How to avoid it? 
I did not change SSID or password when migrating from WiFi 5, all other IoTs connect with no issues (no single device have issues except cameras lost).

andriej, 2022-10-17:
100%, most of my esphome devices run on that network (and only mostly phones and laptop on 5GHz)

roleoroleo, 2022-10-18:
Never seen anything like this.

andriej, 2022-10-18:
Do you have any ideas how could I debug this or at least get some more info on what's happening? 
Except for the AP logs which I will try to gather later

andriej, 2022-10-19:
Do you have any ideas how could I debug this or at least get some more info on what's happening?
Except for the AP logs which I will try to gather later

So far on AP I see only this in log on openwrt AP:

```
Wed Oct 19 10:37:17 2022 daemon.info hostapd: phy1-ap0: STA 38:00:25:xx:xx:xx IEEE 802.11: authenticated
Wed Oct 19 10:37:17 2022 daemon.info hostapd: phy1-ap0: STA 38:00:25:xx:xx:xx IEEE 802.11: associated (aid 4)
Wed Oct 19 10:37:17 2022 daemon.notice hostapd: phy1-ap0: STA-OPMODE-SMPS-MODE-CHANGED 38:00:25:xx:xx:xx off
Wed Oct 19 10:37:26 2022 daemon.info hostapd: phy1-ap0: STA 38:00:25:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
```

roleoroleo, 2022-10-22:
You should try to increase debug log using "-dd" option.
In wifi.sh and wificonnect.sh.

andriej, 2022-10-24:
Is there any place that BSSID of AP is saved within?
I can see from logs that camera tries to connect to the AP that far-far-away instead of the one that is 1 meter away...

roleoroleo, 2022-10-26:
> Is there any place that BSSID of AP is saved within?

Yi saves these configs in a raw flash partition.
```
CURRENT_SSID=$(dd bs=1 skip=28 count=64 if=/dev/mtdblock6 2>/dev/null)
CURRENT_KEY=$(dd bs=1 skip=92 count=64 if=/dev/mtdblock6 2>/dev/null)
```

andriej, 2022-11-03:
Those are SSID and KEY, proper ones. Can't still figure out why it's trying to connect to 'old' BSSID (MAC of AP) instead of the one Yi has next to it

roleoroleo, 2022-11-03:
AFAIK the cam doesn't save the mac address of the AP.
Only SSID and key.

Maybe a wrong channel...
Are you using a channel not licensed for your country?

andriej, 2023-02-21:
As I had to repair SD card (not able to upgrade), I've taken some steps after the upgrade to investigate more.
Camera was in same room as AP although having only 37% WiFi signal strength. I went to the panel and I can see two networks detected with same SSID.

I've selected the first from the top, saved the password and rebooted camera - it has reconnected to the 'closer' AP.

What I would recommend then is to add BSSID/MAC address of AP near to the SSID name or at least it's signal strength.
It would allow to fix that and choose proper network if few with same SSID are detected.

roleoroleo, 2023-02-22:
Unfortunately this process is not under my control: Yi manages the wifi connection.
And the firmware saves only SSID and password.

closed 2024-05-05 by github-actions[bot]