roleoroleo/yi-hack-Allwinner #378: Yi 1080p Home IFUS not working with y25ga firmware

miquelrosell99, 2022-11-18:
I am trying to flash the firmware on a Yi 1080p Home IFUS, but it is not working. The camera boots with a yellow led and then changes to flashing blue and then blue. Firmware on the app still appears to be version 11.something.
Is there something I'm missing? I formatted the sd from the camera and copied all the files from y25ga inside.
After rebooting the camera and waiting, I connected the sd to the pc to see the files, and they were changed to:
[image]
Any help?
Thanks!

roleoroleo, 2022-11-18:
> Firmware on the app still appears to be version 11.something.

Where are you reading version 11?

If the file home_y25gam exists, the cam didn't process the update.
So, probably your cam is not a y25ga.
Check the fw version with the app.

agustinlare, 2022-11-28:
+1,  same scenario

Here's my app fw version if it's helps.

[image]

roleoroleo, 2022-11-29:
Your cam is a y211ga, not a y25ga.
Please check the 1st step in the procedure https://github.com/roleoroleo/yi-hack-Allwinner#install-procedure:
1. Check if your cam is supported in the "Supported cameras" section and note the file prefix.

closed 2024-05-05 by github-actions[bot]