roleoroleo/yi-hack-Allwinner #379: Cam connects to wifi for a few seconds before disconnecting and flashing blue light

rikrdo89, 2022-11-18:
I was able to flash my YI 1080p home AI+, but as soon as I change the wifi SSID and password using either configure_wifi.cfg (https://github.com/roleoroleo/yi-hack-Allwinner/wiki/Change-WiFi-credentials) or wpa_supplicant.conf (https://github.com/roleoroleo/yi-hack-Allwinner/files/6323138/wifi_on_sd.zip), the camera is able to connect to the new network but only temporally (for a few seconds, or couple of minutes) before disconnecting and flashing blue light. When the camera is temporally connected, I do have access to the web server and to the rtsp feed.  I am not sure why this keeps happening, and now I cannot longer add the camera to a network using the Yi Home app on my mobile.

Firmware Version | 8.2.0.0A_201912270941
Model Suffix | y20ga_0.3.1
Serial Number | 9FUS

roleoroleo, 2022-11-19:
After a reboot what happens?
Did you try with another access point?

rikrdo89, 2022-11-24:
I think it might have been the camera... I tried a different camera and it connected just fine. Thanks!

closed 2022-11-24 by rikrdo89