roleoroleo/yi-hack-Allwinner #382: BFUS y20ga firmware 8.2.0.0A_202012041528 --> 12.2.0.2_202212191129 - yihack works

divadiow, 2022-12-20:
Saw from the app there was an update to v12.2. removed yihack. reset cam to factory. re-paired to app, updated firmware. performed fresh yihack and all seems to be well. not tested much though

[image]
[image]

divadiow, 2022-12-20:
success with 9FUS too 
[image]
[image]

although the serial number is now prefixed with 0000

divadiow, 2022-12-20:
sadly still no timestamp option for video. watermark has changed from Yi to Kami

[image]

roleoroleo, 2022-12-21:
Please check if motion detection works.

divadiow, 2022-12-21:
yes. I had both cams on overnight and it was detecting motion and uploading videos to my ftp as it always has. behaviour seems the same

bonuzzz, 2022-12-23:
@divadiow is something changed in video quality with new firmware?

divadiow, 2022-12-23:
annoyingly I didn't do a before and after and now I've updated all 3 of my cams :(

I *think* it's better, but it could be my imagination. Please post an exact comparison if you have one not yet flashed.

LZsolt75, 2022-12-27:
I have also updated the firmware. At first start the camera offers to use the kami cloud as a Smart AI detection (plan required).  With usual motion detection option I think the motion detection is different. My experience is that the camera misses recording of movements even in the high video detection sensitivity setting. It used to record 3 minutes after an event, now it is 2 minutes 30 secs. And I cannot switch off the notification (or i can not find it).

divadiow, 2022-12-27:
I think you're right. I caught a mouse the other day and there is no footage of the trap going off or the mouse running to the trap. I would have expected a video. Other movements are detected and saved though

roleoroleo, 2022-12-28:
The AI algorithm is completely new.
Now the cam uses Pilot AI so I think there are differences from the previous detection (sensitivity, errors, etc...).
Now you can choose between no motion, "generic" motion and AI motion.
With the last choice you can activate human detection (I think for free) and vehicle/animal detection with a Kami cloud plan.
I'm working on a new hack that supports this new feature but a downgrade is always possible restoring mtdblock3.

LZsolt75, 2022-12-31:
It would be great if you could integrate the new features. Is there any guide on how to go back to previous firmware, how to flash only the mtdblock3 partition?

roleoroleo, 2023-01-03:
Check the unbrick guide, the procedure is the same.

murigny, 2023-01-17:
Hello
Y20GA BFUSY31, i'm trying to hack with firmware 12.2.0.2: bootloop.
So, I downgrade to 8.2.0.0, and then, hack with out issues

roleoroleo, 2023-01-21:
Please wait before upgrading to version 12.

cchenga, 2023-01-31:
Hi @murigny, could you let me know how would downgrade to 8.2.0? Do you have the firmware? If yes, could you share with me?

cchenga, 2023-01-31:
> Hello Y20GA BFUSY31, i'm trying to hack with firmware 12.2.0.2: bootloop. So, I downgrade to 8.2.0.0, and then, hack with out issues

Hi @murigny, could you let me know how would downgrade to 8.2.0? Do you have the firmware? If yes, could you share with me?

murigny, 2023-02-01:
you can download here: https://www.kamihome.com/firmware/

and choose 
YI Home Camera 1080P AI+

y20ga_8.2.0.0A_202012041528

Raeprice, 2023-02-11:
Im having 0 luck with this at all at the moment and really unsure whats going wrong. I have 1080p Yi Home on FW v. y20ga_8.2.0.0A_202012041528 after the thankful suggestion from @murigny andits back in my Yi App

I remove my SD Card from the camera
Insert to Windows PC - remove all files (SD Was originally formatted in Yi App)
Download latest y20ga_0.3.2.tgz (SN is BFUS)
Put files directly on root of SD with nothing else.

[image]

Insert and power up camera
light blinks orange, Light flashes blue, then solid orange, then blue flash - solid.
Camera still all working as normal in Yi App after
Cannot conncet to http://IPADDRESS:8080/ or http://ip-cam:8080/
Seems like nothing has happened.

When i remove SD Card again to check filesi see the files have changed so assume its done

[image]

I cannot connect via web UI despite it appearing to be done
This is a first for me, so really hoping to get some suggestions!? TIA

roleoroleo, 2023-02-11:
You have to check which ip address is assigned to your cam and use it:
http://DISCOVERED_IP:8080/

Raeprice, 2023-02-11:
> You have to check which ip address is assigned to your cam and use it in the place of DISCOVERED_IP: http://DISCOVERED_IP:8080/

Hi @roleoroleo I have tried this - sorry I probably wasn't so clear when I referred to HTTP://IPADDRESS:8080/ this was using the cameras IP, which ican successfully ping.

Can you confirm that the file changes confirm the flash was complete? Also should it work In the app post flash?

Raeprice, 2023-02-11:
@roleoroleo ignore me I got it from your suggestion! I did not put http! Stupid me! Thanks alot

roleoroleo, 2023-02-13:
> Can you confirm that the file changes confirm the flash was complete? Also should it work In the app post flash?

Yes.
.done extension means that the hack completed successfully.

LZsolt75, 2023-02-21:
I have tried to flash the original mtdblock3.bin partition as it is  described in the unbrick wiki.
it looks like the flashing is succesfull, in the yi app it showa the earlier version, but it is frequently freezing now, and make a click noise when restarting. The camera body is also quite warm touching by hand.
Any suggestion how to solve this problem?

roleoroleo, 2023-02-21:
Restoring the mtdblock3 restores the cam to its initial state.
The last try could be a reset with the button.

LZsolt75, 2024-04-03:
My problem with the freezing issue and with the heating is gone with switching off the snapshot and activate the swap file. 
@roleoroleo  Any progress with implementing the new functions with the new firmware?

roleoroleo, 2024-04-06:
If you mean the features related to AI detections, they are implemented.

Jaragar, 2024-07-28:
> you can download here: https://www.kamihome.com/firmware/
> 
> and choose YI Home Camera 1080P AI+
> 
> y20ga_8.2.0.0A_202012041528

Hello, could you give me the firmware please, I can't find it anywhere

przemyslaw-latoch, 2024-10-08:
Has anyone tried it out on the new 12.2.0.3_202409241615 version yet? Any changes?

roleoroleo, 2024-11-30:
Please, share your backup. I'll check if the hack is compatible.

konqueror81, 2024-11-30:
> Please, share your backup. I'll check if the hack is compatible.

How can I share? I have to attach all the files?

roleoroleo, 2024-12-11:
I need mtdblocks 2 and 3 only.
You can attach the files here.

divadiow, 2024-12-22:
anyone got the **mtdblock3.bin** for y20ga version **12.2.0.2_202212191129**?

the file here https://github.com/roleoroleo/yi-hack-Allwinner/issues/415#issuecomment-1761506178 is for 
 **y20ga_8.2.0.0A_202012041528**  which doesn't seem to offer me the update :/

roleoroleo, 2024-12-23:
Here it is:
[mtdblock3.bin.gz](https://github.com/user-attachments/files/18232225/mtdblock3.bin.gz)

closed 2025-04-23 by github-actions[bot]