roleoroleo/yi-hack-Allwinner #383: BFUS y20ga - Bricked (help needed)

nicmorais, 2022-12-26:
Hello!
So, I recently tried to update the hack in my Yi Home 1080p camera (y20ga), but something seems wrong. The device will not boot anymore, and the orange LED just keeps blinking.

I also tried flashing the "unhack" stuff for y20ga, and also tried flashing the hack again. Didn't help. When I try flashing it again, the blue LED starts blinking, the camera reboots itself, and then it justs blinks the orange LED...

What can I do? I'm skilled with soldering and connecting the UART pins... Will provide logs soon, if needed.

nicmorais, 2022-12-27:
Here's the bootlog: https://pastebin.com/PqGqn92b

roleoroleo, 2022-12-27:
Do you have a backup copy of the flash?
You should try to restore mtdblock3.

nicmorais, 2022-12-27:
Hello @roleoroleo. I hope you are fine.

Yes, I have the backup. I tried to do so with that unbrick.sh script. But did not solve the issue.
I have two BFUS cameras, which one is currently "bricked". The orange LED has became solid (does not blink) and the bootlog is that one I've just provided. Thankfully, I managed to get into the command line.

Do you know if getting the mtdblock3 from the working camera and writing it down in the bricked one could solve this issue? If so, how do I do it?

I also copied the .ko files from the working (and hacked) camera and ran insmod with then into the bricked camera, and it seems to successfully install them. Except the 8189fs.ko module. It says: "can't insert '8189fs.ko': unknown symbol in module, or unknown parameter".

(Thanks in advance for your great work! I surely will donate.)

roleoroleo, 2022-12-28:
> Yes, I have the backup. I tried to do so with that unbrick.sh script. But did not solve the issue. I have two BFUS cameras, which one is currently "bricked". The orange LED has became solid (does not blink) and the bootlog is that one I've just provided. Thankfully, I managed to get into the command line on both devices.

If the backup is ok you should be able to restore the cam.
In this model, it's very very uncommon to damage other partitions. Only mtdblock3 has a chance to break.

> Do you know if getting the mtdblock3 from the working camera and writing it down in the bricked one could solve this issue? If so, how do I do it?

Yes, if the other cam is the same model (y20ga) mtdblock3 is the same.

> I also copied the .ko files from the working (and hacked) camera and ran insmod with then into the bricked camera, and it seems to successfully install them. Except the 8189fs.ko module. It says: "can't insert '8189fs.ko': unknown symbol in module, or unknown parameter".

This is not the right way. Try to restore the whole partition.
If this procedure https://github.com/roleoroleo/yi-hack-Allwinner/tree/master/unbrick does not work, try to repeat it changing the type of the partition created. I mean:
- Create the partition to restore with `./build.sh hacked`
- Try to load it to the cam.
- Create the partition to restore with `./build.sh factory`
- Try to load it to the cam.

The bootloader stores the hash of the partition loaded, so you can't load the same partition 2 times.

nicmorais, 2022-12-28:
Thanks a lot! The camera was definitively saved.
After flashing the unbrick, which was built with `build.sh factory`, the device was trying to run the system.sh script from the hack, but could not find it.
It seemed like the init.sh script was still the hacked version. So I flashed the hack again, and the issue was solved. :smile: 

I hope you have a great 2023

closed 2022-12-28 by nicmorais

roleoroleo, 2022-12-28:
> I hope you have a great 2023. Enjoy my donation!

Thank you very much and happy new year!