roleoroleo/yi-hack-Allwinner #385: y20ga BFUS - Wrong time and date on files

nicmorais, 2022-12-30:
My issue now is related to the timezone of the camera (0.32 version).

Even after setting the timezone to BRT3 in the web page and rebooting the camera, the files it records in the SD card still in UTC timezone.
This does not happens with yi-hack-v5: in this version of the hack, the timezone shows as UTC in the command line, but the files are recorded in the correct time.

Also, the timezone for America/Sao Paulo in the timezones page does not seem right, because [Brazil abolished it in 2019](https://en.wikipedia.org/wiki/Daylight_saving_time_in_Brazil#:~:text=Brazil%20abolished%20DST%20in%202019). Now, the correct timezone string for America/Sao Paulo is BRT3.

roleoroleo, 2023-01-06:
Which fw do you have?
I mean Yi firmware.

nicmorais, 2023-01-09:
Hello. The version is 8.2.0A_201912270941

nicmorais, 2023-01-11:
Also, should I upgrade the firmware to 12.2.0.2_202212191129?

roleoroleo, 2023-01-16:
With version 9, the hack works in GMT. So the records are in GMT.
You should check if the "Motion Events" web page translates the dates correctly. You should see "Directory name" column in UTC and "Date & time" column in your local time.

> Also, the timezone for America/Sao Paulo in the timezones page does not seem right, because [Brazil abolished daylight saving time in 2019](https://en.wikipedia.org/wiki/Daylight_saving_time_in_Brazil#:~:text=Brazil%20abolished%20DST%20in%202019). Now, the correct timezone string for America/Sao Paulo is BRT3.

I fixed it.

> Also, should I upgrade the firmware to 12.2.0.2_202212191129?

Please, wait a few days to update.
The hack is not fully compatible with v12 at the moment.

nicmorais, 2023-01-16:
Hello!
Alright, them.
This does not seem a major issue, when browsing the recorded videos in the YI app, it does not snap forward the timezone offset. The cameras I own that record video in my timezone makes it kinda annoying to watch previous recordings in the YI Home app, because of such snapping in the timeline bar. I'm not sure if I explained this well enough, I just want you to know that recording the files in GMT is the best option so far.
But an option to upload the files to the FTP server with the user's timezone would be great. What are your thoughts?

roleoroleo, 2023-01-21:
I am very confused between the various fw versions.
Without the hack, does the app works? Is the timeline ok?

> But an option to upload the files to the FTP server with the user's timezone would be great. What are your thoughts?

I already developed this feature, but it's not released.

But at the moment I'm working on the new version 12, where the time zone managemente is changed.
So I need to update all these scripts.

nicmorais, 2023-01-21:
Hello again, @roleoroleo
Yes, without the hack the app used to work well, and the timeline in the YI app also used to work as expected.

Anyway, I'm closing this issue. Thanks a lot for your attention.

closed 2023-01-21 by nicmorais