roleoroleo/yi-hack-Allwinner #388: Issues with MQTT

ibanmda, 2023-01-19:
Hi again @roleoroleo !

I saw this issues in my registry of Home Assistant. Always appears when I switch on/off the camera.  How can I repair it?

```
Payload ' { "SWITCH_ON":"no", "SAVE_VIDEO_ON_MOTION":"yes", "SENSITIVITY":"low", "AI_HUMAN_DETECTION":"no", "SOUND_DETECTION":"no", "SOUND_SENSITIVITY":"50", "LED":"no", "ROTATE":"no", "IR":"yes", "CRUISE":"off", "NULL":"NULL" }' is not a Number

Error parsing value: 'dict object' has no attribute 'PTZ_PRESET' (value: { "SWITCH_ON":"no", "SAVE_VIDEO_ON_MOTION":"yes", "SENSITIVITY":"low", "AI_HUMAN_DETECTION":"no", "SOUND_DETECTION":"no", "SOUND_SENSITIVITY":"50", "LED":"no", "ROTATE":"no", "IR":"yes", "CRUISE":"off", "NULL":"NULL" }, template: {{ value_json.PTZ_PRESET | int }})
```

Firmware Version | 0.3.2
-- | --
Base Version | 8.3.0.0A_201912271707
Model Suffix | y25ga

Thank You!

roleoroleo, 2023-01-21:
Are you using my integration or mqtt discovery?

ibanmda, 2023-01-23:
> Are you using my integration or mqtt discovery?

Mqtt discovery

roleoroleo, 2023-01-23:
This part comes from a pr and the code is not mine.
I need time to check it.

ibanmda, 2023-01-23:
This problem is recent, i didn´t have it before. Maybe due an upgrade of Home Assistant...

A lot of thanks @roleoroleo

closed 2024-05-05 by github-actions[bot]