roleoroleo/yi-hack-Allwinner #390: Can't save modifications in camera settings tab

Berto93, 2023-02-22:
My setup:

- YI Dome X BFUSY291J9TOLZ200725 - y30qa
- Firmware Version 0.3.3

After update to version 0.3.3 I can't save any modifications in the camera settings tab. Do you have any tips?

Thanks in advance.

roleoroleo, 2023-02-23:
Please, clear the cache of the browser.

Berto93, 2023-02-24:
@roleoroleo Thanks for your reply, i tried, also with a different browser but didn't work, the camera is ON but the switch is OFF. 

When i try to get camera informations using web service "http://ip:8080/cgi-bin/get_configs.sh?conf=camera" returns camera ="OFF" but the camera is ON.

I also tried to use "http://ip:8080/cgi-bin/camera_settings.sh?switch_on=yes" but gives me the following output:

{
"error":"false"
}

roleoroleo, 2023-02-24:
Error false means ok, so the command completed successfully.
Does the problem refers only to switch_on or to all the settings?
If it's only switch_on there is a fixed bug:
https://github.com/roleoroleo/yi-hack-Allwinner/issues/391#issuecomment-1443963887

snakuzzo, 2023-02-24:
I tried webGUI and web services... and every settings in camera settings are not saved and showed correctly.
It's not just a view problem.
I have `"SWITCH_ON":"no"` and I can't change to `"SWITCH_ON":"yes"` after upgrade to latest release. Trying to view RTSP stream it's impossible to open the stream.

Berto93, 2023-02-24:
@roleoroleo it’s exactly the same issue of @snakuzzo 

https://github.com/roleoroleo/yi-hack-Allwinner/issues/390#issuecomment-1444439707

roleoroleo, 2023-02-25:
@snakuzzo 
Probably the hack is not correctly installed.
If you want to fix it, the best way is to restart from scratch: apply the unbrick procedure and reinstall the hack.
If you prefer the soft approach, open a ftp connection with the cam, and compare the yi-hack folder with the release version.
And try to fix missing files.

snakuzzo, 2023-02-25:
Solved.
- Enabled Cloud
- Removed hack
- Upgraded official fw
- Hacked again

Berto93, 2023-02-26:
for me still not wortking, tried to remove hack and reinstall on my y30qa but gives me "error while saving" and every time that I try to change any setting camera reboots

roleoroleo, 2023-02-26:
Try to run the command with a terminal.
Open a ssh session and run:
```
cd /home/yi-hack/www/cgi-bin/
QUERY_STRING="save_video_on_motion=yes&motion_detection=yes&sensitivity=low&ai_human_detection=no&ai_vehicle_detection=no&ai_animal_detection=no&sound_detection=yes&sound_sensitivity=50&led=no&ir=yes&rotate=no&cruise=no&switch_on=yes" ./camera_settings.sh
```
and check the output.

closed 2024-05-05 by github-actions[bot]